ev3dev/ev3dev #898: send files to ev3 from desktop Mac

legoparrot, 2017-06-06:
I'm trying to send my python file to my ev3dev using scp from my desktop.
I'v tried using "scp test.py ev3dev.ev3" or "scp /Users/legoparrot/Desktop/ev3/test.py ev3dev.ev3" but it can't find my file.
I'm using Mac OS X Sierra(10.12.5)

jabrena, 2017-06-06:
Hi @legoparrot, try this syntax:

```
cd /Users/legoparrot/Desktop/ev3/
scp test.py ev3dev.ev3:/home/robot
```

Juan Antonio

legoparrot, 2017-06-06:
Didn't work. Says "-bash: cd: /Users/legoparrot/Desktop/ev3/test.py: Not a directory"

dlech, 2017-06-06:
It sounds like you added `test.py` to the `cd` command.

legoparrot, 2017-06-06:
tried it again and got:
"ssh: connect to host ev3dev.ev3 port 22: Operation timed out
lost connection"

jabrena, 2017-06-06:
Hi @legoparrot can you use the ip?

``` 
ssh robot@192.168.1.206
```

Review the IP in the top of the display and replace in the ssh call:
[image]

legoparrot, 2017-06-06:
Works great. Thanks for the help!

closed 2017-06-06 by legoparrot

mob41, 2017-06-07:
@legoparrot Just for a reference, you might want to use <code>ev3dev.local</code> instead of <code>ev3dev.ev3</code>

FRNetwork, 2017-07-17:
OK, I'm just not quite understanding the responses here, and I am having the same problem. I can SSH into my ev3 just fine and have been running all sorts of commands to it such as moving motors, talking etc, but now I am trying to upload a .py file to run a program and I'm stuck.
So:
1. I open a new terminal shell window 
2. I ssh into it with ssh robot@192.168.1.75
3. once I'm in it, all is well, no errors.
4. I try running a scp command there and it just sits and waits forever.
5. I open a new terminal shell window and I cd to where the file is locally on my mac
6. then I try to run the scp command to push it from my mac to the ev3 and I get asked for passwords for an account that doesnt exist, my local user name from my mac on the robot. Since this account doesnt exist (the robot account exists), I cant move the file. I tried sudo -u robot but then my mac says that account doesnt exist locally.
Command run: scp RobotProgram.py 192.168.1.75:/home/robot

I'm missing something fundamental and I need a nudge, please help.