ev3dev/ev3dev #899: Question about PID

cho934, 2017-06-06:
Hi,
I'm trying to update the last version of ev3dev and to use internal motor regulation instead of create a thread to do it with bad latency.

Here are my question:
- What is the frequency of this internal PID loop ?
- I don't understand why both parameters count_per_rot and count_per_m ?
ok count_per_rot is defined because we know that we have 360 pulse per rotation
but for count_per_m, it depends the diameter of my wheel ? isn't it ?
- Do you think, we can add the calculation of the odometry in that loop to avoid to create another thread for that ?
- If I use position regulation with "run-to-" functions, do the speed regulation is activated in the same time ? Normally yes because before incate position with position_sp, I need to indicate speed_sp, I'm right ?

Christophe.

WasabiFan, 2017-06-06:
> I don't understand why both parameters count_per_rot and count_per_m ?
ok count_per_rot is defined because we know that we have 360 pulse per rotation
but for count_per_m, it depends the diameter of my wheel ? isn't it ?

`count_per_m` is only defined for "linear actuators", which extend and retract their length instead of rotating. Using the rotation measurement is correct.

> Do you think, we can add the calculation of the odometry in that loop to avoid to create another thread for that ?

You can periodically read the distance measurements in your own code, but to my knowledge there is no other means of tracking that data without modifying the kernel/ev3dev kernel modules.

> If I use position regulation with "run-to-" functions, do the speed regulation is activated in the same time ? Normally yes because before incate position with position_sp, I need to indicate speed_sp, I'm right ?

I believe the current implementation is relatively simplistic: If you ask to run at 300 speed for 1000 counts, it will run at 300 speed without regard for position until it hits the target speed and then stop and hold position. It does not slow down as it approaches, as far as I know.

dlech, 2017-06-06:
> What is the frequency of this internal PID loop ?

It is [set to 2 msec](https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-stretch/ev3/legoev3_motor.c#L43). If the kernel can actually keep up with this, I'm not sure.

> I don't understand why both parameters count_per_rot and count_per_m ?
> ok count_per_rot is defined because we know that we have 360 pulse per rotation
> but for count_per_m, it depends the diameter of my wheel ? isn't it ?

`count_per_m` is [for linear actuators only](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/motors.html#sysfs). It does not mean anything for regular motors.

> Do you think, we can add the calculation of the odometry in that loop to avoid to create another thread for that ?

Not sure what you are asking for here. The motor driver keeps track of position already. How is odometry different from this?

> If I use position regulation with "run-to-" functions, do the speed regulation is activated in the same time ? Normally yes because before indicate position with position_sp, I need to indicate speed_sp, I'm right ?

Yes, this is correct.

dlech, 2017-06-06:
> if you have a non-zero deceleration time

Translation: if you set `ramp_down_sp`.

cho934, 2017-06-07:
> What is the frequency of this internal PID loop ?
> It is set to 2 msec. If the kernel can actually keep up with this, I'm not sure.
I make (with an old release of ev3dev) a loop using a dedicated thread to calculate the odometry = position of the robot (x, y, theta), and to calculate just after a position PID.
But the EV3 is very slow and I was obliged to decrease this timer to every 50msec. In conclusion, the regulation was very bad !

My goal is to use the up-to-date version of ev3dev with what already exists.

So with the last version of ev3dev, I just need to create a thread to calculate the position of the robot (odometry) with my parameters (wheel diameter, the 2-wheel spacing, etc) and that's all. The kernel manages both speed PID and position PID loops when the run commands are set. If I optimize cosinus calculation, I think I would not have overflows...

I'm going to test that ;)

closed 2020-08-17 by dlech