ev3dev/ev3dev #900: Problems with Some sensors on PiStorms v2

jabrena, 2017-06-07:
Hi,

I am releasing the new version of EV3Dev-lang-java and I noticed that some sensors doesn´t run with PiStormsV2. 

+ EV3USSensor
+ EV3IRSensor
+ EV3GyroSensor

I attach the logs for every example:

**EV3USSensor**

``` java
package ev3dev.sensors.ev3;

import lejos.hardware.port.SensorPort;
import lejos.robotics.SampleProvider;
import lejos.utility.Delay;

public class USSensorDemo {
	
	private static EV3UltrasonicSensor us1 = new EV3UltrasonicSensor(SensorPort.S1);

	public static void main(String[] args) {

		final SampleProvider sp = us1.getDistanceMode();
		int distanceValue = 0;

        final int iteration_threshold = 100;
        for(int i = 0; i <= iteration_threshold; i++) {

        	float [] sample = new float[sp.sampleSize()];
            sp.fetchSample(sample, 0);
            distanceValue = (int)sample[0];

			System.out.println("Iteration: " + i + ", Distance: " + distanceValue);

			Delay.msDelay(500);
        }
		
	}

}
```

```
java -server -jar /home/robot/ev3dev-lang-java-all-0.6.1-RELEASE.jar
ev3dev#7|2017-06-07 19:08:09 [main] TRACE ev3dev.hardware.EV3DevPlatforms - /sys/class/power_supply/legoev3-battery
ev3dev#7|2017-06-07 19:08:09 [main] TRACE ev3dev.hardware.EV3DevPlatforms - Detected platform: PISTORMS
ev3dev#7|2017-06-07 19:08:09 [main] TRACE ev3dev.hardware.EV3DevPlatforms - /sys/class/power_supply/legoev3-battery
ev3dev#7|2017-06-07 19:08:09 [main] TRACE ev3dev.hardware.EV3DevPlatforms - Detected platform: PISTORMS
ev3dev#7|2017-06-07 19:08:09 [main] TRACE ev3dev.hardware.EV3DevPlatforms - /sys/class/power_supply/legoev3-battery
ev3dev#7|2017-06-07 19:08:09 [main] TRACE ev3dev.hardware.EV3DevPlatforms - Detected platform: PISTORMS
ev3dev#7|2017-06-07 19:08:09 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: pistorms:BBS2
ev3dev#7|2017-06-07 19:08:09 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/lego-port
ev3dev#7|2017-06-07 19:08:09 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port0/address
ev3dev#7|2017-06-07 19:08:09 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-port/port1/address
ev3dev#7|2017-06-07 19:08:09 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-port/port1/address
ev3dev#7|2017-06-07 19:08:09 [main] INFO  ev3dev.hardware.EV3DevSensorDevice - detected lego port: /sys/class/lego-port/port1
ev3dev#7|2017-06-07 19:08:09 [main] TRACE ev3dev.utils.Sysfs - echo ev3-uart > /sys/class/lego-port/port1/mode
ev3dev#7|2017-06-07 19:08:09 [main] TRACE ev3dev.utils.Sysfs - echo lego-ev3-us > /sys/class/lego-port/port1/set_device
ev3dev#7|2017-06-07 19:08:10 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: pistorms:BBS2
ev3dev#7|2017-06-07 19:08:10 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/lego-sensor
ev3dev#7|2017-06-07 19:08:10 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor16/address
ev3dev#7|2017-06-07 19:08:10 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-sensor/sensor16/address
ev3dev#7|2017-06-07 19:08:10 [main] INFO  ev3dev.hardware.EV3DevSensorDevice - detected lego sensor: /sys/class/lego-sensor/sensor16
ev3dev#7|2017-06-07 19:08:10 [main] TRACE ev3dev.utils.Sysfs - echo US-DIST-CM > /sys/class/lego-sensor/sensor16/mode
ev3dev#7|2017-06-07 19:08:11 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor16/value0
ev3dev#7|Iteration: 0, Distance: 255
ev3dev#7|2017-06-07 19:08:11 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor16/value0
[... 226 lines cut ...]
ev3dev#1|2017-06-07 19:17:10 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor20/value0
ev3dev#1|Iteration: 8
ev3dev#1|Gyro angle: -2550
ev3dev#1|2017-06-07 19:17:10 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor20/value0
ev3dev#1|Iteration: 9
ev3dev#1|Gyro angle: -2550
ev3dev#1|2017-06-07 19:17:11 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor20/value0
ev3dev#1|Iteration: 10
```

Anyone has reported this fact?

```
Image file:         ev3dev-jessie-rpi2-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-rpi2
Board:              Raspberry Pi 3 Model B Rev 1.2
Revision:           a02082
Brickman:           0.8.1
ev3devKit:          0.4.3
```

Juan Antonio

Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/301

dlech, 2017-06-07:
Can you provide examples in bash or python that can be run on ev3dev without installing additional software or compiling?

Are you setting up the port first? PiStorms is like BrickPi in that it cannot automatically detect sensors, so you have to set the port to ev3-uart mode and then write to the `set_device` attribute. Then you can use the lego-sensor device.

jabrena, 2017-06-07:
Yes, for the last case you have the bash traces. I will explain.

```
ev3dev#1|2017-06-07 19:17:04 [main] TRACE ev3dev.utils.Sysfs - echo ev3-uart > /sys/class/lego-port/port1/mode
ev3dev#1|2017-06-07 19:17:04 [main] TRACE ev3dev.utils.Sysfs - echo lego-ev3-gyro > /sys/class/lego-port/port1/set_device
ev3dev#1|2017-06-07 19:17:05 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detecting device on port: pistorms:BBS2
ev3dev#1|2017-06-07 19:17:05 [main] TRACE ev3dev.utils.Sysfs - ls /sys/class/lego-sensor
ev3dev#1|2017-06-07 19:17:05 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor20/address
ev3dev#1|2017-06-07 19:17:05 [main] DEBUG ev3dev.hardware.EV3DevDevice - Detected port on path: /sys/class/lego-sensor/sensor20/address
ev3dev#1|2017-06-07 19:17:05 [main] INFO  ev3dev.hardware.EV3DevSensorDevice - detected lego sensor: /sys/class/lego-sensor/sensor20
ev3dev#1|2017-06-07 19:17:05 [main] TRACE ev3dev.utils.Sysfs - echo GYRO-ANG > /sys/class/lego-sensor/sensor20/mode
ev3dev#1|2017-06-07 19:17:06 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor20/value0
ev3dev#1|Iteration: 0
ev3dev#1|Gyro angle: -2550
```

As you say, for BrickPi+ & PiStorms, it is necessary to set the port to operate with the sensor:

``` 
echo ev3-uart > /sys/class/lego-port/port1/mode
echo lego-ev3-gyro > /sys/class/lego-port/port1/set_device
``` 

Later, once you have the port in the right way, it is possible to set the mode and later view the values:

```
echo GYRO-ANG > /sys/class/lego-sensor/sensor20/mode
cat /sys/class/lego-sensor/sensor20/value0
-2550
````

If you observe, the logs from `ev3dev-lang-java` try to explain a bash session. :)
If you observe, the value is rare. The same idea is for other cases.

Juan Antonio

dlech, 2017-10-03:
Is this still a problem?