ev3dev/ev3dev #901: Problems with Some sensors on BrickPi+

jabrena, 2017-06-07:
Hi,

I am releasing the new version of EV3Dev-lang-java and I noticed that some sensors doesn´t run with BrickPi+ in the right way.

+ GyroSensor with Angle & Rate mode
+ IRSensor on IRSeek
+ IRSensor on IRRemote

Details:

**GyroSensor with Angle & Rate mode**

``` 
package ev3dev.sensors.ev3;

import lejos.hardware.port.SensorPort;
import lejos.robotics.SampleProvider;
import lejos.utility.Delay;
import org.slf4j.Logger;

public class GyroSensorDemo3 {

	private static final Logger log = org.slf4j.LoggerFactory.getLogger(GyroSensorDemo3.class);
	//Robot Configuration
	private static EV3GyroSensor gyroSensor = new EV3GyroSensor(SensorPort.S1);

	//Configuration
	private static int HALF_SECOND = 500;

	public static void main(String[] args) {

		final SampleProvider sp = gyroSensor.getAngleAndRateMode();
		int value = 0;

		int iterationCounter = 0;

        //Control loop
		while(true){
			float [] sample = new float[sp.sampleSize()];
            sp.fetchSample(sample, 0);
            value = (int)sample[0];

            log.info("Gyro angle/rate: {}", value);

			if(value >= 90){
				//Sound.getInstance().beep();
				log.info("Rotated 90 degrees");
				break;
			}

			iterationCounter++;
			if(iterationCounter >= 100){
				break;
			}

			Delay.msDelay(HALF_SECOND);
		}

	}

}
```

``` 
java -server -jar /home/robot/ev3dev-lang-java-all-0.6.1-RELEASE.jar
ev3dev#11|2017-06-07 19:34:31 [main] TRACE ev3dev.hardware.EV3DevPlatforms - /sys/class/power_supply/legoev3-battery
ev3dev#11|2017-06-07 19:34:31 [main] TRACE ev3dev.hardware.EV3DevPlatforms - Detected platform: BRICKPI
ev3dev#11|2017-06-07 19:34:31 [main] TRACE ev3dev.hardware.EV3DevPlatforms - /sys/class/power_supply/legoev3-battery
ev3dev#11|2017-06-07 19:34:31 [main] TRACE ev3dev.hardware.EV3DevPlatforms - Detected platform: BRICKPI
ev3dev#11|2017-06-07 19:34:31 [main] TRACE ev3dev.hardware.EV3DevPlatforms - /sys/class/power_supply/legoev3-battery
[... 337 lines cut ...]
ev3dev#3|2017-06-07 19:43:46 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor12/value2
ev3dev#3|2017-06-07 19:43:46 [main] TRACE ev3dev.utils.Sysfs - cat /sys/class/lego-sensor/sensor12/value3
ev3dev#3|Iteration: {}4
ev3dev#3|Beacon Channel 1: Remote: {}0
ev3dev#3|Beacon Channel 2: Remote: {}0
ev3dev#3|Beacon Channel 3: Remote: {}0
ev3dev#3|Beacon Channel 4: Remote: {}0
``` 

Anyone has reported this fact?

```
Image file:         ev3dev-jessie-rpi2-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-rpi2
Board:              Raspberry Pi 3 Model B Rev 1.2
Revision:           a02082
Brickman:           0.8.1
ev3devKit:          0.4.3
```

Juan Antonio
https://github.com/ev3dev-lang-java/ev3dev-lang-java

Related issue:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/302

dlech, 2017-06-07:
duplicate of #542.

dlech, 2017-06-07:
This is a known problem with the BrickPi firmware and can't be fixed in ev3dev without fixing the brickpi firmware.

closed 2017-06-07 by dlech