ev3dev/ev3dev #902: EVB on stretch, motors are jittery

dwalton76, 2017-06-09:
I installed https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2017-05-15/snapshot-ev3dev-stretch-bone-generic-2017-05-15.img.xz but it is not happy, when this runs

```
   def init_motors(self):

        for x in self.motors:
            if not x.connected:
                log.error("%s is not connected" % x) 
                sys.exit(1)
            x.reset()

        # Lower all the way down, then raise a bit, then lower back down.
        # We do this to make sure it is in the same starting spot each time.
        log.info("Initialize elevator %s" % self.elevator)
        self.elevator.run_forever(speed_sp=40, stop_action='brake')
        self.elevator.wait_until('running')
        self.elevator.wait_until('stalled')
        self.elevator.run_to_rel_pos(speed_sp=200, position_sp=-50)
        self.elevator.wait_while('running')
        self.elevator.run_forever(speed_sp=40, stop_action='hold')
        self.elevator.wait_until('running')
        self.elevator.wait_until('stalled')
        self.elevator.stop()
        self.elevator.reset()
        self.elevator.stop(stop_action='brake')

        if self.shutdown_event.is_set():
            return

        log.info("Initialize flipper %s" % self.flipper)
        self.flipper.run_forever(speed_sp=150, stop_action='hold')
        self.flipper.wait_until('running')
        self.flipper.wait_until('stalled')
        self.flipper.stop()
        self.flipper.reset()
        self.flipper.stop(stop_action='hold')
        self.flipper_at_init = True 

        if self.shutdown_event.is_set():
            return
```

The motors are really jittery
https://www.youtube.com/watch?v=ihOFbrRN9S8

dlech, 2017-08-12:
Does this still happen with the `ev3dev-1.4.0` kernel?

dwalton76, 2017-08-12:
Will give it a try

dwalton76, 2017-08-12:
@dlech am trying to test it but am hitting an earlier failure when I try to capture an image via fswebcam.  I'm using a standard lego webcam, it bombs out on
```
robot@evb[lego-crane-cuber]# fswebcam -d /dev/video0 -r 320x240 foo.png
--- Opening /dev/video0...
Trying source module v4l2...
/dev/video0 opened.
No input was specified, using the first.
Adjusting resolution from 320x240 to 356x292.
--- Capturing frame...
Timed out waiting for frame!
No frames captured.
robot@evb[lego-crane-cuber]#
```

We see the webcam
```
robot@evb[lego-crane-cuber]# lsusb
Bus 001 Device 004: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 001 Device 003: ID 046d:0850 Logitech, Inc. QuickCam Web
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
robot@evb[lego-crane-cuber]# 
robot@evb[lego-crane-cuber]# ls -l /dev/video0 
crw-rw---- 1 root video 81, 0 Aug 12 23:29 /dev/video0
robot@evb[lego-crane-cuber]# 
```

I was trying some of the ideas in this thread https://forum.arduino.cc/index.php?topic=253571.0
It doesn't recognize it as a v4l2 device?
```
robot@evb[lego-crane-cuber]# sudo fswebcam -d /dev/bus/usb/001/003 foo.png
--- Opening /dev/bus/usb/001/003...
Trying source module v4l2...
/dev/bus/usb/001/003 opened.
/dev/bus/usb/001/003: Not a V4L2 device?
Trying source module v4l1...
/dev/bus/usb/001/003 opened.
/dev/bus/usb/001/003: Not a V4L device?
VIDIOCGCAP: Inappropriate ioctl for device
/dev/bus/usb/001/003 closed.
Unable to find a source module that can read /dev/bus/usb/001/003.
robot@evb[lego-crane-cuber]# 
```
Can you do a quick test and see if you can capture an image via fswebcam with a lego webcam?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-bone-generic-2017-05-15
Kernel version:     4.9.41-ev3dev-1.4.0-bb.org
/usr/bin/ev3dev-sysinfo: line 68: warning: command substitution: ignored null byte in input
Board:              TI AM335x BeagleBone Black
Revision:           00c0
Brickman:           0.9.0
ev3devKit:          0.5.1
```

dlech, 2017-08-13:
`fswebcam -d /dev/video0 -r 320x240 foo.png` works for me...

[image]

Granted, I am using a BeagleBone Green instead of Black, but there shouldn't be a difference there with regards to USB.

```
robot@ev3dev-bbg:~$ lsusb
Bus 001 Device 002: ID 046d:0850 Logitech, Inc. QuickCam Web
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Could the hub/wifi be causing the problem?

dwalton76, 2017-08-13:
Swapped the hub, camera and even the beaglebone black but still have the same issue.  Let me re-install from latest image on http://www.ev3dev.org/downloads/ and see if that makes any difference

dwalton76, 2017-08-13:
re-imaged and still the same issue (tried before and after a dist-upgrade).  If you do `sudo fswebcam -d /dev/bus/usb/001/002 foo.png` does it show your camera as a V4L2 device.

dlech, 2017-08-13:
`sudo fswebcam -d /dev/bus/usb/001/002 foo.png` doesn't work for me. And I would not expect it to either. It is looking for a V4L device node, not a USB device node.

Have you tried connecting the camera directly without a hub?

dwalton76, 2017-08-13:
Hmm it does indeed work if I remove the hub from the equation. I could try a powered usb hub that shouldn't be necessary.

dwalton76, 2017-08-13:
crap I just noticed I did not download the stretch image when I re-installed....re-install take II

dwalton76, 2017-08-13:
ok re-imaged with stretch and am still having the fswebcam issue whenever I use a usb hub but let's ignore that for now.  Getting back to the issue of "are the motors are jittery?", I cannot say for certain because it hits #896 almost immediately.  Will comment there with an update.

dwalton76, 2017-09-09:
FYI I am no longer seeing this on the following...will close this one
```
robot@evb[ev3dev-lang-python]# ev3dev-sysinfo 
Image file:         ev3dev-stretch-bone-generic-2017-09-01
Kernel version:     4.9.41-ev3dev-1.4.0-bb.org
/usr/bin/ev3dev-sysinfo: line 68: warning: command substitution: ignored null byte in input
Board:              TI AM335x BeagleBone Black
Revision:           000c
Brickman:           0.10.0
ev3devKit:          0.5.2
robot@evb[ev3dev-lang-python]# 
```

closed 2017-09-09 by dwalton76

dlech, 2017-09-09:
Cool.

One or more of these probably fixed it:

* 26bd24375d13b8801169ee78e0a23874c9b5d2ac
* b13d8229a1a6f6bebc82d8126398879ed260f13a
* e30d15336f47abc7b630dd9771913db6d85716bb