ev3dev/ev3dev #904: Cyrillic characters on EV3 screen?

AXKuhta, 2017-06-13:
### What are you trying to do?

Trying to display russian text on EV3 screen (in console mode)
I installed and configured console-cyrillic package and also generated ru_RU.UTF-8 locale.

### What did you expect to happen?

To see russian text when outputting it in console on EV3 screen.

### What actually happened?

Only those characters that have non-cyrillic analogues (Like capital T, a and o) outputted correctly, characters like Ж and Я looked like rectangle symbols.
It works as intended in SSH.

### What hardware and software are you using (including version numbers)?

EV3 brick, 2017-06-09 ev3dev image

dlech, 2017-06-13:
Did you select the font using the `setfont` command?

Some relevant man pages:
* https://manpages.debian.org/jessie/kbd/setfont.8.en.html
* https://manpages.debian.org/jessie/console-setup-mini/console-setup.5.en.html
* https://manpages.debian.org/jessie/manpages/locale.1.en.html

AXKuhta, 2017-06-14:
Setting font to one of the unicode fonts in /usr/share/consolefonts/ fixed the cyrillic characters but it also resulted in empty spaces being displayed as zeros (i.e. `clear` fills screen with zeros) or other symbols, depending on font

dlech, 2017-06-14:
I'm not sure what could be causing that.

dlech, 2017-07-27:
What kernel version are you using? There seems to be a problem in ev3dev-stretch (4.9.x kernel).

dlech, 2017-07-28:
Meh... It looks like the in-kernel fbcon doesn't support unicode. There was a patch a while back, but it looks like it was never taken to completion.

https://lkml.org/lkml/2010/11/26/50

dlech, 2017-07-28:
:tada: :ru: :sparkles: 

I finally figured it out. Install the `console-cyrillic` package...

```
sudo apt update
sudo apt install console-cyrillic
```

Set the locale to a non-UTF8 locale. I tested with `ru_RU.ISO-8859-5`, but there is also `ru_RU.CP1251` and `ru_RU.KOI8-R`. Alternatively, you can just generate one of these locales and leave the default as UTF8, in which case you will need to `export LANG=ru_RU.ISO-8859-5` in the virtual console.

```
sudo dpkg-reconfigure locales
```

Then, in the virtual console...

```
unicode_stop
setfont /usr/share/console-cyrillic/alt_8x14.psf -m /usr/share/console-cyrillic/iso-8859-5.screenmap
```

The font can be any of the `.psf` files in `/usr/share/console-cyrillic` and the `-m` argument should match the locale you chose.

dlech, 2017-07-28:
It looks like there are some fonts that ship with ev3dev as well, so installing the `console-cyrillic` package is not necessary.

```
unicode_stop
setfontCyrSlav-Fixed13 -m /usr/share/consoletrans/ISO-8859-5.acm.gz
```

AXKuhta, 2017-07-28:
Can confirm that this worked, thanks!

closed 2017-07-28 by AXKuhta