ev3dev/ev3dev #905: Strange behavior after setting up NFS file share and rebooting ev3

ivankeller, 2017-06-13:
I connect to ev3 via ssh using internet USB sharing connection (I followed the well explained tutorial http://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/). 
Everything work like a charm. When rebooting ev3 the connection start again.

Then I set up an NFS file share as explained on http://www.ev3dev.org/docs/tutorials/setting-up-an-nfs-file-share/ 
I had to slightly modify the procedure by adding a sudo in front of the command for the setup on the ev3 (otherwise there is a Permission denied error):
```
sudo systemctl enable nfs-common.service
sudo systemctl start nfs-common.service
sudo systemctl enable rpcbind.service
sudo systemctl start rpcbind.service
```
Everything seems ok so far. I could successfully test some python code that was on the NFS share on my laptop. Great!

The problem came when I rebooted the ev3: I could notice a bunch of FAILED and SKIP lines during the ev3dev booting  process. After booting I am unable to connect to the ev3. 
Selecting Wireless and Networks entry in the ev3dev menu displays an endless 'Loading...'.
Since I am not able to connect to ev3 I cannot lok to the boot log to try to diagnose the issue.

I could reproduce this behavior repeating the whole procedure (a bit tedious since I had to flash again the microSD and reconfigure the connection).  Each time I do the NFS step the same issue appears but if I skip it there is no connecting issue. But I really would like to use NFS file sharing.

Running  macOS 10.12.5 on a Mac Book Pro and booting EV3 with image ev3dev-jessie-ev3-generic-2017-06-09.zip

dlech, 2017-06-13:
Hmm... it sounds like NFS breaks networking at boot.

A simple workaround would be to skip the `enable` commands. This is what causes these services to start during boot up. You would have to manually run the `start` commands after each boot though.

If you want to try to troubleshoot farther, you could use a serial console on Input port 1 or you could attach a keyboard directly to the EV3 and press <kbd>ALT</kbd>+<kbd>CTRL</kbd>+<kbd>F6</kbd> to get a login prompt. Then log in and run `sudo journalctl -b | sudo tee log.txt`. You could also disable the NFS services this way so you don't have to re-flash the SD card.

Also, if the tutorial page needs to be changed, could you submit a pull request for that?

manifolded, 2017-07-04:
I am having exactly the same problem that @ivankeller describes.  I have subscribed to this post as I am very eager for NFS to be part of my and my boy's playflow.  We have only started messing with ev3dev as of a few days ago.

manifolded, 2017-07-04:
Using a magnifying glass I was able to see in the rolling startup log that the message
"FAILED to start connection <garbled>" appears repeatedly.  

Unfortunately the keyboard I have for my Raspberry Pi 3 doesn't have function keys, and whatever the serial console is that plugs in to port 1, I don't think I have one.  I'll try to get my hands on an actual full size non-bluetooth keyboard.  I haven't seen one in ages.

dlech, 2017-07-04:
FYI, NFS alternative for macOS: https://github.com/ev3dev/ev3dev/wiki/Setting-up-Apple-File-Sharing

ivankeller, 2017-07-17:
Hi David, 
I tried to install netatalk as you suggested but apt-get cannot find the package:
``` 
robot@ev3dev:~$ sudo apt-get install netatalk
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package netatalk
```
Of course I did a `sudo apt-get update` before and everything seemed to go well.

dlech, 2017-07-17:
Hmm... the `netatalk` package is not in Jessie for some reason: https://packages.debian.org/stretch/netatalk

HikloLime, 2017-07-17:
I am also having the exact same issue as stated above. Bluetooth won't boot either (as that was how i connected to the ev3)

oknystad, 2017-10-01:
I'm experiencing the same issue as described above on my EV3 Brick. Current kernel: `4.4.87-22-ev3dev-ev3`.

> or you could attach a keyboard directly to the EV3 and press ALT+CTRL+F6 to get a login prompt. Then log in and run sudo journalctl -b | sudo tee log.txt. You could also disable the NFS services this way so you don't have to re-flash the SD card.

Doing this, I am able to reboot normally after running commands:
```
sudo systemctl disable nfs-common.service
sudo systemctl disable rpcbind.service
```
Running `sudo journalctl -b | sudo tee log.txt` gave the result:
```
robot@ev3dev:~$ cat log.txt
-- Logs begin at Thu 1970-01-01 00:00:09 UTC, end at Sun 2017-10-01 13:08:02 UTC. --
Oct 01 13:01:52 ev3dev systemd-journal[89]: Runtime journal is using 4.0M (max allowed 8.0M, trying to leave 4.2M free of 24.5M available → current limit 8.0M).
Oct 01 13:01:52 ev3dev systemd-journal[89]: Runtime journal is using 4.0M (max allowed 8.0M, trying to leave 4.2M free of 24.5M available → current limit 8.0M).
Oct 01 13:01:52 ev3dev systemd-journal[89]: Missed 30 kernel messages
Oct 01 13:01:52 ev3dev kernel: Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Oct 01 13:01:52 ev3dev kernel: Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Oct 01 13:01:52 ev3dev kernel: CPU: Testing write buffer coherency: ok
Oct 01 13:01:52 ev3dev kernel: Setting up static identity map for 0xc0008400 - 0xc000843c
Oct 01 13:01:52 ev3dev kernel: devtmpfs: initialized
Oct 01 13:01:52 ev3dev kernel: clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Oct 01 13:01:52 ev3dev kernel: futex hash table entries: 256 (order: -1, 3072 bytes)
Oct 01 13:01:52 ev3dev kernel: NET: Registered protocol family 16
Oct 01 13:01:52 ev3dev kernel: DMA: preallocated 256 KiB pool for atomic coherent allocations
Oct 01 13:01:52 ev3dev kernel: cpuidle: using governor ladder
Oct 01 13:01:52 ev3dev kernel: cpuidle: using governor menu
Oct 01 13:01:52 ev3dev kernel: legoev3_init: global_cursor_default: 0, default_screen_mode: -1
Oct 01 13:01:52 ev3dev kernel: edma edma.0: Legacy memcpy is enabled, things might not work
Oct 01 13:01:52 ev3dev kernel: edma edma.0: TI EDMA DMA engine driver
Oct 01 13:01:52 ev3dev kernel: edma edma.1: Legacy memcpy is enabled, things might not work
Oct 01 13:01:52 ev3dev kernel: edma edma.1: TI EDMA DMA engine driver
Oct 01 13:01:52 ev3dev kernel: SCSI subsystem initialized
Oct 01 13:01:52 ev3dev kernel: usbcore: registered new interface driver usbfs
Oct 01 13:01:52 ev3dev kernel: usbcore: registered new interface driver hub
Oct 01 13:01:52 ev3dev kernel: usbcore: registered new device driver usb
Oct 01 13:01:52 ev3dev kernel: Bluetooth: Core ver 2.21
Oct 01 13:01:52 ev3dev kernel: NET: Registered protocol family 31
Oct 01 13:01:52 ev3dev kernel: Bluetooth: HCI device and connection manager initialized
Oct 01 13:01:52 ev3dev kernel: Bluetooth: HCI socket layer initialized
Oct 01 13:01:52 ev3dev kernel: Bluetooth: L2CAP socket layer initialized
Oct 01 13:01:52 ev3dev kernel: Bluetooth: SCO socket layer initialized
Oct 01 13:01:52 ev3dev kernel: clocksource: Switched to clocksource timer0_1
Oct 01 13:01:52 ev3dev kernel: NET: Registered protocol family 2
Oct 01 13:01:52 ev3dev kernel: TCP established hash table entries: 1024 (order: 0, 4096 bytes)
Oct 01 13:01:52 ev3dev kernel: TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
Oct 01 13:01:52 ev3dev kernel: TCP: Hash tables configured (established 1024 bind 1024)
Oct 01 13:01:52 ev3dev kernel: UDP hash table entries: 256 (order: 0, 4096 bytes)
Oct 01 13:01:52 ev3dev kernel: UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
Oct 01 13:01:52 ev3dev kernel: NET: Registered protocol family 1
Oct 01 13:01:52 ev3dev kernel: RPC: Registered named UNIX socket transport module.
Oct 01 13:01:52 ev3dev kernel: RPC: Registered udp transport module.
Oct 01 13:01:52 ev3dev kernel: RPC: Registered tcp transport module.
Oct 01 13:01:52 ev3dev kernel: RPC: Registered tcp NFSv4.1 backchannel transport module.
Oct 01 13:01:52 ev3dev kernel: NFS: Registering the id_resolver key type
Oct 01 13:01:52 ev3dev kernel: Key type id_resolver registered
Oct 01 13:01:52 ev3dev kernel: Key type id_legacy registered
Oct 01 13:01:52 ev3dev kernel: nfs4filelayout_init: NFSv4 File Layout Driver Registering...
Oct 01 13:01:52 ev3dev kernel: io scheduler noop registered (default)
Oct 01 13:01:52 ev3dev kernel: Serial: 8250/16550 driver, 3 ports, IRQ sharing disabled
Oct 01 13:01:52 ev3dev kernel: serial8250.0: ttyS0 at MMIO 0x1c42000 (irq = 25, base_baud = 9375000) is a 16550A
Oct 01 13:01:52 ev3dev kernel: serial8250.1: ttyS1 at MMIO 0x1d0c000 (irq = 53, base_baud = 8250000) is a 16550A
Oct 01 13:01:52 ev3dev kernel: serial8250.2: ttyS2 at MMIO 0x1d0d000 (irq = 61, base_baud = 8250000) is a TL16C550
Oct 01 13:01:52 ev3dev kernel: loop: module loaded
Oct 01 13:01:52 ev3dev kernel: zram: Added device: zram0
Oct 01 13:01:52 ev3dev kernel: at24 1-0050: 16384 byte 24c128 EEPROM, read-only, 0 bytes/write
Oct 01 13:01:52 ev3dev kernel: spi_davinci spi_davinci.1: DMA: supported
Oct 01 13:01:52 ev3dev kernel: spi_davinci spi_davinci.1: DMA: RX channel: 0x00000012, TX channel: 0x00000013, event queue: 0
Oct 01 13:01:53 ev3dev kernel: spi_davinci spi_davinci.1: Controller at 0xfef0e000
Oct 01 13:01:53 ev3dev kernel: spi_davinci spi_davinci.0: DMA: supported
[... 557 lines cut ...]
[... cut ...]
Oct 01 13:03:53 ev3dev sudo[519]: robot : TTY=tty6 ; PWD=/home/robot ; USER=root ; COMMAND=/usr/bin/tee log.txt
Oct 01 13:03:53 ev3dev sudo[519]: pam_unix(sudo:session): session opened for user root by robot(uid=0)
Oct 01 13:07:54 ev3dev sudo[518]: pam_unix(sudo:auth): authentication failure; logname=robot uid=1000 euid=0 tty=/dev/tty6 ruser=robot rhost=  user=robot
Oct 01 13:08:02 ev3dev sudo[518]: robot : TTY=tty6 ; PWD=/home/robot ; USER=root ; COMMAND=/bin/journalctl -b
Oct 01 13:08:02 ev3dev sudo[518]: pam_unix(sudo:session): session opened for user root by robot(uid=0)

```

dlech, 2017-10-01:
Does this also happen on the ev3dev-stretch snapshot images?

What if you create a systemd mount unit file instead of following the tutorial on the ev3dev website?

https://cloudnull.io/2017/05/nfs-mount-via-systemd/

JesApp, 2017-11-17:
Just tried your systemd suggestion, @dlech and it worked just fine. I'll try and come up with a pull request for a new tutorial over the weekend.

JesApp, 2017-11-17:
Just created a [pull request](https://github.com/ev3dev/ev3dev.github.io/pull/386) and asked two questions there.

dlech, 2017-11-19:
Please follow the updated tutorial using a systemd unit file instead of fstab.

http://www.ev3dev.org/docs/tutorials/setting-up-an-nfs-file-share/

closed 2017-11-19 by dlech

sgalzin, 2021-01-12:
I have followed what I believe to be the new set of instructions, using the .mount file, but I get the "Job for home-robot-nfsshare.mount failed" error as well:
`robot@ev3dev:~$ sudo systemctl start home-robot-nfsshare.mount
Job for home-robot-nfsshare.mount failed.
See "systemctl status home-robot-nfsshare.mount" and "journalctl -xe" for details.
`
Note that this does not prevent ev3dev from booting, but the NFS share does not work. Running the suggested command to know more, I get:
`robot@ev3dev:~$ systemctl status home-robot-nfsshare.mount
● home-robot-nfsshare.mount - Mount an nfs share
   Loaded: loaded (/etc/systemd/system/home-robot-nfsshare.mount; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-04-10 19:08:53 UTC; 11s ago
    Where: /home/robot/nfsshare
     What: 192.168.0.111:/home/steph/nfs/ev3dev
  Process: 604 ExecMount=/bin/mount 192.168.0.111:/home/steph/nfs/ev3dev /home/robot/nfsshare -t nfs (code=exited, status=32)
`

As someone did above, I am including the result of the "sudo journalctl -b | sudo tee log.txt" command, in case this is what is necessary to understand more:
`robot@ev3dev:~$ sudo journalctl -b | sudo tee log.txt
-- Logs begin at Fri 2020-04-10 18:22:19 UTC, end at Fri 2020-04-10 19:10:29 UTC. --
Apr 10 19:01:34 ev3dev kernel: Booting Linux on physical CPU 0x0
Apr 10 19:01:34 ev3dev kernel: Linux version 4.14.117-ev3dev-2.3.5-ev3 (david@freyr) (gcc version 6.4.1 20171012 (Linaro GCC 6.4-2017.11)) #1 PREEMPT Sat Mar 7 12:54:39 CST 2020
Apr 10 19:01:34 ev3dev kernel: CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
Apr 10 19:01:34 ev3dev kernel: CPU: VIVT data cache, VIVT instruction cache
Apr 10 19:01:34 ev3dev kernel: OF: fdt: Machine model: LEGO MINDSTORMS EV3
Apr 10 19:01:34 ev3dev kernel: Memory policy: Data cache writeback
Apr 10 19:01:34 ev3dev kernel: Reserved memory: created DMA memory pool at 0x80000000, size 0 MiB
Apr 10 19:01:34 ev3dev kernel: OF: reserved mem: initialized node serial@80000000, compatible id shared-dma-pool
Apr 10 19:01:34 ev3dev kernel: DaVinci da850/omap-l138/am18x variant 0x1
Apr 10 19:01:34 ev3dev kernel: On node 0 totalpages: 16384
Apr 10 19:01:34 ev3dev kernel: free_area_init_node: node 0, pgdat c07081e0, node_mem_map c3f79000
Apr 10 19:01:34 ev3dev kernel:   DMA zone: 128 pages used for memmap
Apr 10 19:01:34 ev3dev kernel:   DMA zone: 0 pages reserved
Apr 10 19:01:34 ev3dev kernel:   DMA zone: 16384 pages, LIFO batch:3
Apr 10 19:01:34 ev3dev kernel: pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Apr 10 19:01:34 ev3dev kernel: pcpu-alloc: [0] 0 
Apr 10 19:01:34 ev3dev kernel: Built 1 zonelists, mobility grouping on.  Total pages: 16256
Apr 10 19:01:34 ev3dev kernel: Kernel command line: console=tty1
Apr 10 19:01:34 ev3dev kernel: PID hash table entries: 256 (order: -2, 1024 bytes)
Apr 10 19:01:34 ev3dev kernel: Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Apr 10 19:01:34 ev3dev kernel: Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Apr 10 19:01:34 ev3dev kernel: Memory: 53552K/65536K available (5071K kernel code, 259K rwdata, 1660K rodata, 172K init, 158K bss, 11984K reserved, 0K cma-reserved)
Apr 10 19:01:34 ev3dev kernel: Virtual kernel memory layout:
                                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                                   vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
                                   lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
                                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                                     .text : 0xc0008000 - 0xc04fc3c8   (5073 kB)
                                     .init : 0xc069d000 - 0xc06c8000   ( 172 kB)
                                     .data : 0xc06c8000 - 0xc0708ca0   ( 260 kB)
                                      .bss : 0xc070d854 - 0xc073542c   ( 159 kB)
Apr 10 19:01:34 ev3dev kernel: SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Apr 10 19:01:35 ev3dev kernel: Preemptible hierarchical RCU implementation.
Apr 10 19:01:35 ev3dev kernel:         Tasks RCU enabled.
Apr 10 19:01:35 ev3dev kernel: NR_IRQS: 325
Apr 10 19:01:35 ev3dev kernel: clocksource: timer0_1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
Apr 10 19:01:35 ev3dev kernel: sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
Apr 10 19:01:35 ev3dev kernel: Console: colour dummy device 80x30
Apr 10 19:01:35 ev3dev kernel: console [tty1] enabled
Apr 10 19:01:35 ev3dev kernel: Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
Apr 10 19:01:35 ev3dev kernel: pid_max: default: 32768 minimum: 301
Apr 10 19:01:35 ev3dev kernel: Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Apr 10 19:01:35 ev3dev kernel: Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Apr 10 19:01:35 ev3dev kernel: CPU: Testing write buffer coherency: ok
Apr 10 19:01:35 ev3dev kernel: Setting up static identity map for 0xc0008400 - 0xc000843c
Apr 10 19:01:35 ev3dev kernel: Hierarchical SRCU implementation.
Apr 10 19:01:35 ev3dev kernel: devtmpfs: initialized
Apr 10 19:01:35 ev3dev kernel: random: get_random_u32 called from bucket_table_alloc+0x14c/0x198 with crng_init=0
Apr 10 19:01:35 ev3dev kernel: clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... cut ...]
Apr 10 19:09:51 ev3dev ntpd[546]: error resolving pool 3.debian.pool.ntp.org: Name or service not known (-2)
Apr 10 19:10:29 ev3dev sudo[614]:    robot : TTY=pts/0 ; PWD=/home/robot ; USER=root ; COMMAND=/bin/journalctl -b
Apr 10 19:10:29 ev3dev sudo[615]:    robot : TTY=pts/0 ; PWD=/home/robot ; USER=root ; COMMAND=/usr/bin/tee log.txt
Apr 10 19:10:29 ev3dev sudo[614]: pam_unix(sudo:session): session opened for user root by robot(uid=0)
Apr 10 19:10:29 ev3dev sudo[615]: pam_unix(sudo:session): session opened for user root by robot(uid=0)
`