ev3dev/ev3dev #906: ev3dev PyCharm Tutorial Problem

Rosstb, 2017-06-14:
I have started using ev3dev recently. I have been following the tutorial which shows how to set up PyCharm to program the EV3. The link to the tutorial is here: http://www.ev3dev.org/docs/tutorials/setting-up-python-pycharm/

My goal was to run some test programs on the EV3, then move on to a project.

I followed the tutorial, but when I got to the step named "Configuring synchronization in PyCharm", I ran into some problems.

The first step is:

What we are going to do now is make a clone of the project that we created on the ev3dev machine so we can work on it on the development PC. In the Welcome dialog of PyCharm, choose: Checkout from version control > Git. Now type the hostname of the ev3dev machine (which is usually going to be ev3dev or ev3dev.local), followed by a colon and then the project name. In the other fields choose a nice parent and project directory.

I followed these directions, then proceeded to the next step, which is:

If all goes well you should end up with a new PyCharm window open to your new empty project. If the ‘testing’ dialog stays on screen for a long time, it might be that your PyCharm master password is needed for your PyCharm password storage. Cancel the checkout, type the master password and try again.

I did not end up with the new window it is talking about. I found out that if I make a new folder then use that folders name for the directory name, it works fine. Then, when I put in the test code from the tutorial, I tried to find the "Commit Change" button under the VCS tab, but it is not there. Are there some steps that I am not aware of that I need to execute?

Note: I am using Windows 10 laptop, running PyCharm Community edition. 

Any help would be greatly appreciated, and thanks in advance!

WasabiFan, 2017-06-15:
Mentioning @fisherds @antonvh

They were the original authors of the tutorial, so I'd say they are your best bet.

P.S. Thanks for including detailed repro steps and  a link to the specific tutorial! It makes solving your problem much easier.

fisherds, 2017-06-15:
I only did the area from [Additional features for the PyCharm Professional version](http://www.ev3dev.org/docs/tutorials/setting-up-python-pycharm/#additional-features-for-the-pycharm-professional-version) on so I'm only useful at troubleshooting the PyCharm Professional version of the tutorial sorry.  It's super slick (free only to educators though). 

Unrelated sidenote:
  BTW I used that setup for a bunch of Middle School students at a recent programming camp and it worked great.  With PyCharm Professional even the 7th graders had no issues using ev3dev on the robots.  They complained to no end about the slowness to start running a program, but it all worked slick.

antonvh, 2017-06-15:
Hm. I have a Mac, git is installed on it. Not sure if windows computers come with git.

Also for my last project I reverted to a more cumbersome but robust way of getting the code the ev3 machine. I just check in my code on github and pull it from the brick. Then I run the code either from brickman or the command line. This way of working is dependent on the laptop and brick being on the same wifi and on basic git knowledge.

I experimented with an sftp mount of ev3 home folder, but couldn't get that stable. Wasn't there a stabler version of SMB on the newer ev3dev images? I'll try that again and rework the tutorial to make it simpler.

Rosstb, 2017-06-15:
antonvh, could you let me know when it's updated? Thanks!

WasabiFan, 2017-06-15:
@Rosstb If you'd like to follow along with the existing tutorial, I'd suggest trying to install [Git for Windows](https://git-scm.com/). My guess is that installing Git will solve the problem; install that and restart PyCharm to test it.

Rosstb, 2017-06-28:
I've just encountered another problem. This one is less related to the tutorial, but is still relevant. I have been able to write and run programs on the EV3 brick using the SSH terminal, and so far it has gone well. However, in the future I would like to be able to run programs without using a PC. That is where I ran into problems. When I first tried to start a program using the file browser, it displayed a message saying the file was not executable. Then I looked around and found out that I had to add the "#!/usr/bin/env python" as the first line of code, and also add the "chmod +x program.py" line to the SSH. However, when I select the program in the file browser, the screen goes blank for about 4 seconds, and the LED's go orange, and then it returns to the file browser. I am having trouble figuring out if the problem is with my code or how I set up the project. I have read all the issues here on GitHub, and I can't find anything that helps me. Here is the code I am using:
#!/usr/bin/env python
from ev3dev.auto import *

motorA = Motor(OUTPUT_A)
motorA.run_timed(time_sp = 3000,speed_sp = motorA.max_speed)

Thanks!

fisherds, 2017-06-28:
#!/usr/bin/env python3

Unless you have setup python to route to python3 you need to add the 3 onto that line.

fisherds, 2017-06-28:
BTW if you do want to setup python to default to python3 here are some commands I used once to do that for my students:

(full steps)
```
which python
ls -al /usr/bin/python*
cd /usr/bin
ls -al python*
sudo rm python
ls -al python*
sudo ln -s python3 python
ls -al python*
```

(explanation of each line)
determine that python2 is the default (known to be true don't check)
find where python is at (you'll know the anwser)
change directory
see what's there
remove the symlink
show that it's gone
make a symlink from python to python3
see that it worked

(shorter version that does only the necessary steps)
```
cd /usr/bin
sudo rm python
sudo ln -s python3 python

```

Rosstb, 2017-06-28:
Thanks! BTW, is there a way to keep from having to remark the file as executable every time I commit and push some changes to the file?

dlech, 2017-06-28:
@fisherds, touching files in `/usr/bin/` is not a good idea. These files are "owned" by packages, so, for example, if there is ever a security update for python, this will break because the package will write over your change.

A simpler solution would be...

    sudo ln -s -r $(which python3) /usr/local/bin/python

The local administrator "owns" `/usr/local/`and `/usr/local/bin/` takes precedence over `/usr/bin/`.

Or, if you want a fancy way to switch back and forth between python 2 and 3, see [this page] about using the "alternatives" feature in Debian.

dlech, 2017-06-28:
> Thanks! BTW, is there a way to keep from having to remark the file as executable every time I commit and push some changes to the file?

If you are using git to transfer the files, mark the file as executable it git.

https://stackoverflow.com/questions/21691202/how-to-create-file-execute-mode-permissions-in-git-on-windows

Rosstb, 2017-06-28:
@dlech , I am using Putty for the SSH, and transferring the files in PyCharm.

dlech, 2017-06-28:
Isn't PyCharm using git to transfer the files?

Rosstb, 2017-06-28:
I suppose it is, but I am not seeing a way to mark the file as executable in git. I have been marking it as executable in Putty, but I have to redo it every time I change the code. My apologies if I'm missing something here. This may seem obvious to some people but I believe I'm slightly in over my head.

dlech, 2017-06-28:
> but I am not seeing a way to mark the file as executable in git

That is what the link I posted is about. :wink: Depending on how git was installed, you might have a program in your start menu called "Git Bash". You can use that to browse to the folder where your code is stored and follow the steps. The key command there is `git update-index --chmod=+x myprogram.py`.

samfish74, 2017-06-30:
Hi, I have had similar problems to Rosstb, I followed same tutorial and have been using moboXTerm and trying to configure the ev3 to run using the ps3 remote
I used the following tutorial and i have had same issue when running on the brickman, screen goes blank for some time and then in the tiniest possible writing it says something like "No such file or directory".
https://sites.google.com/site/ev3python/system/app/pages/search?scope=search-site&q=ps3
http://www.ev3dev.org/docs/tutorials/using-ps3-sixaxis/
and the code from
https://gist.github.com/artpoz/083777cb1f8c9efe6418c75d18bdd32b
Please could someone help advise please?

WasabiFan, 2017-06-30:
@samfish I suspect that's a line endings issue. See [http://python-ev3dev.readthedocs.io/en/latest/faq.html#my-script-works-when-launched-as-python3-script-py-but-exits-immediately-or-throws-an-error-when-launched-from-brickman-or-as-script-py](here).

samfish74, 2017-06-30:
Hi Thanks for your help
I've jsut tried the link but it doesn't seem to take me anywhere? It just says error 404 on Github and that there isn't anything there?
Cheers

WasabiFan, 2017-06-30:
Sorry, mis-formatted the link. It should work now.

Rosstb, 2017-07-17:
Thanks for your help, @dlech! I just got around to trying this out, and I still seem to be having a problem with the file becoming non-executable every time I commit and push changes to that file. I will try to give an in-depth description of what I am doing.
First, I create a python file in pycharm, add some code, then commit and push the file to the EV3. Most importantly, the first line of that code is #!/usr/bin/env python3.
Second, I make changes to the file until I am satisfied with it, then I work on making the file executable.
If I understood correctly, I navigate to the correct file on the computer with the code stored in it. I am supposed to type "git update-index --chmod=+x myprogram.py" into Git Bash, then I press enter. At that point the file is still not executable. Then in PUTTY I typed "chmod +x myprogram.py". At that point the file is executable. But, when I make changes in pycharm, commit, and push those changes, the file ceases to be executable. Am I missing any steps somewhere?

dlech, 2017-07-17:
did you commit the changes after `git update-index --chmod=+x myprogram.py`?

    git commit -a -m "set executable bit"

Rosstb, 2017-07-18:
Thanks, that worked!

dlech, 2017-09-05:
can this issue be closed?

samfish74, 2017-09-05:
Yep

closed 2017-09-05 by dlech