ev3dev/ev3dev #908: university wifi connection fails

d-e-t, 2017-06-16:
I've been stumped trying to connect to our university wifi.  Connecting to my home network with a wpa passcode works fine, but the university setup is more complicated.  The main wifi network requires login with username and password, and the guest wifi network does not complete the connection until a browser screen is opened and the usage agreement acceptance button clicked.

How can I access advanced network options, aside from what is accessible through brickman?

I just downloaded the latest image for the SD card today, and the wifi dongle is an EP-N8508GS.

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-06-09
Kernel version:     4.4.68-20-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2017-06-17:
You can use `connmanctl` to set up network connections from the command line.

    connmanctl
    agent on
    connect wifi_...
    
You could also install `lynx`, which is a text based web browser.

    sudo apt update
    sudo apt install lynx
    lynx

WasabiFan, 2017-06-17:
To clarify: you generally would do this over a remote terminal via SSH, but given that you don't have an internet connection at this point, you can also plug a keyboard into the EV3 -- or through a hub if necessary -- and press <kbd>CTRL</kbd>+<kbd>ALT</kbd>+<kbd>F6</kbd> to get to a terminal on the brick itself. The same key combo but with <kbd>F1</kbd> (or maybe it's <kbd>F2</kbd>) will get you back to Brickman.

d-e-t, 2017-06-17:
Thanks for the suggestions!  I'll try these out Monday.

d-e-t, 2017-06-23:
Following up on this:  I didn't have much luck with either option.  (1) I installed lynx, but that did not allow me to accept the usage agreement and complete the guest network connection.  (2) I set up a config file with ConnMan, but was not able to connect to the main WPA2 Enterprise (PEAP) connection.  This will probably have to be a job for our IT department to figure out if I do end up going with the EV3 hardware; but I'm currently leaning towards using a Raspberry Pi and BrickPi.

This is all part of evaluating options for projects to use in a Fall semester programming course for mechanical engineering students.  I like the idea of using lego for controls/sensors applications, without having to spend a lot of time building the hardware.  This site has been very helpful as I've evaluated the options and got started.

closed 2017-06-23 by d-e-t

WasabiFan, 2017-06-23:
@fisherds teaches a course with EV3s on a campus with restricted Wi-Fi, and I believe what he does is get the MAC address of the Wi-Fi dongle and have it whitelisted by the network administrator. That's probably the easiest option given that you would need to do many of them. The MAC address is generally on the label on the dongle.

P.S. There are some musings on the subject of teaching with ev3dev [here](https://github.com/ev3dev/ev3dev/issues/783) and in particular @fisherds has shared his curriculum [here](https://github.com/Rosebotics/ev3dev-curriculum).

fisherds, 2017-06-26:
Our IT group already had a way to whitelist the dongle by MAC address.  It worked out slick for me, but every network is different I guess.  Nice I had a friend in IT to set it up.