ev3dev/ev3dev #909: ev3 try to build BLE success

YuHanChen04, 2017-06-19:
### What are you trying to do?
I try to build iBeacon on EV3

### What did you expect to happen?
I follow this github https://github.com/danny-source/iBeaconForLinux,and finally it's success!

And , I try to make EV3 can scan the iBeacon.
But it's very hard !
I hope it can be success from someone!

dlech, 2017-06-19:
The built-in Bluetooth on the EV3 is Bluetooth "Classic" only. It does not support Low Energy. If you want to use Bluetooth Low Energy with the EV3, you will need to get a USB Bluetooth dongle that supports Bluetooth Low Energy.

JorgePe, 2017-06-21:
You can also use python with gatttlib (ev3dev aleady includes gatttlib), there is an example here for emulating a beacon:
https://bitbucket.org/OscarAcena/pygattlib/src/a858e8626a93cb9b4ad56f3fb980a6517a0702c6/examples/beacon.py?at=default&fileviewer=file-view-default

[not sure if it works with current kernel]

closed 2020-08-17 by dlech