ev3dev/ev3dev #910: How to use the new stacking features included with BrickPi3?

jabrena, 2017-06-19:
Hi,

Recently, the kernel was updated to add support for BrickPi3 and the board has new features to stack multiple BrickPi3 together.

The file: `config.txt` allow enable 4 BrickPi together:

```
#####################################
# Dexter Industries BrickPi3 config #
#####################################

# uncomment if you are using BrickPi3
dtoverlay=brickpi3

# uncomment only if you are stacking multiple BrickPi3s
# replace 0123... with actual id number of each BrickPi3
#dtparam=id1=0123456789ABCDEF0123456789ABCDEF
#dtparam=id2=0123456789ABCDEF0123456789ABCDEF
#dtparam=id3=0123456789ABCDEF0123456789ABCDEF
#dtparam=id4=0123456789ABCDEF0123456789ABCDEF
```

Once, you have connected a second BrickPi3, how to read the Motor.A from the first BrickPi3 and later call, the Motor.B from the Second BrickPi3 for example?

I tried to read the technical notes about BrickPi3, but I could find any info.

http://www.ev3dev.org/news/2017/04/15/kernel-release-cycle-20/
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/brickpi3.html

Cheers

Juan Antonio

dlech, 2017-06-20:
You can find the ID of a BrickPi3 with `dmesg | grep brickpi`. Then, like the instructions say, add the ID to `config.txt` and uncomment those lines.

I believe that the ports for the BrickPi3 with `id2` will appear as address ...S5-S8 and ...ME-MH, and so on.

dlech, 2017-10-28:
http://docs.ev3dev.org/en/ev3dev-stretch/platforms/brickpi3.html#setup

closed 2017-10-28 by dlech