ev3dev/ev3dev #911: How to use a Grove Sensor port included on BrickPi3?

jabrena, 2017-06-19:
Hi,

Recently, the kernel was updated to add support for BrickPi3 and I would like to know, how to use the Grove Sensor Port included with the board. I suppose that it is necessary to send/receive I2C messages, but I don´t know if exist some example or technical documentation.

Many thanks in advance.

Juan Antonio

dlech, 2017-06-20:
There is nothing special about ev3dev here. Try <https://github.com/DexterInd/GrovePi>.

JorgePe, 2017-06-21:
Not BrickPi but in case someone searching for it gets here, I used the Mindsensors Grove adapter with ev3dev on EV3
http://ofalcao.pt/blog/2017/using-grove-devices-to-the-ev3

It's really just a matter of using the proper I2C commands.

jabrena, 2017-06-21:
Hi,

Many thanks for the article. The question was about the i2c commands to read values from the new grove port added on the BrickPi3 board, so when I user connect a grove sensor into the grove input port, it is possible to follow the commands described in the article from @JorgePe ?

Juan Antonio

JorgePe, 2017-06-21:
The concepts look similar...
BrickPi3 seems to have a GrovePi, an Arduino-based board that uses i2C at address 0x04 - at least as I understand from a quick reading of this library:
https://github.com/DexterInd/GrovePi/blob/master/Software/Python/grovepi.py
You should read that file to understand how to set the board to INPUT or OUTPUT and how to send commands to the Grove devices through it.

They also have code in C, perhaps it's better (you're trying to port to Java?)

jabrena, 2017-06-21:
Many many thanks @dlech & @JorgePe for the technical support.

This weekend, I will to connect with a Grove Sensor.

Cheers

Juan Antonio

JorgePe, 2017-06-21:
@jabrena are you going to try stacking? ( #910 )
I wonder if the I2C address of GrovePi is also changeable. If it isn't, you will have to (somehow) disable one of the Grove adapters.
Have to get time this summer holidays to test the BrickPi3 myself.

jabrena, 2017-06-21:
Hi @JorgePe, 

Yes, yes I will try this summer but, now the focus is with ROSJava nodes:
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/210
https://github.com/ev3dev-lang-java/ev3dev-lang-java/issues/313
:)

Later, I will try with Stacking for BrickPi3 but I have to review the concept about MotorPort/SensorPort because until BrickPi3, every platform only has 4 ports and with this feature, the library should  handle until 16 Ports so I have to review it in detail.

Cheers

Juan Antonio

dlech, 2017-06-21:
The Grove port on the BrickPi3 is connected directly to the I2C pins on the RPi. GrovePi is a totally separate board.

JorgePe, 2017-06-21:
really? Then it is necessary something like an i2C hub to allow using more than one grove device.

jabrena, 2017-06-22:
Hi @JorgePe,
Maybe this item could be pretty interesting...
http://www.robotshop.com/uk/grove-i2c-hub-extension-module.html

new sensors!!! :D

JorgePe, 2017-06-22:
I have the first. Do I really need a 3V3-5V converter?

jabrena, 2017-06-22:
I don´t know. It is another hub. 
The first one is coming. :D

JorgePe, 2017-06-22:
I'll start my summer holidays in a few days... hope I have time to test something, but kids deserve some attention. Will keep an eye on you to ;)

jabrena, 2017-06-22:
ROS first but maybe some i2c commands in the middle :p

jabrena, 2017-08-01:
Hi @JorgePe, 

A Grove electromagnet sensor is coming to follow your Article and learn the idea.

Cheers

JorgePe, 2017-08-01:
:) That reminds me that my BrickPi 3 is still waiting for some of my attention.

jabrena, 2017-08-01:
Yes yes, I will start with the Mindsensors adapter for EV3 and later, I will try to jump to BrickPi3.

Cheers

jabrena, 2017-09-03:
Hi @dlech,

I try to find the technical documentation for the Grove adapter but I couldn´t find it.
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensors.html#other-i2c-sensors
The unique reference is this post:
http://www.ev3dev.org/news/2017/02/11/ev3dev-jessie-2017-02-11-release/

Do you know the link?

Juan Antonio

dlech, 2017-09-03:
To be clear, you are talking about the Grove I2C port on the BrickPi3 and not the Mindsensors Grove sensor adapter?

The link you gave is about BrickPi3. There is no documentation because it is just an I2C port. There are no special ev3dev features about it.

jabrena, 2017-09-03:
Sorry, I talked about `Mindsensors Grove sensor adapter`.

dlech, 2017-09-03:
There is no support in ev3dev for this sensor.

jabrena, 2017-09-03:
Any reason? Complexity?

dlech, 2017-09-03:
No one has asked about this sensor until now, so I suspect there is not much interest.

jabrena, 2017-09-03:
I am not sure @dlech. 

Do you like If I could collaborate in the project trying to create that Driver? For me, the process will be a bit hard but I will learn in the process. :)

dlech, 2017-09-03:
I would recommend starting with something much simpler and just use direct I2C commands for this sensor.

jabrena, 2017-09-03:
Oki, 

In the next opportunity, I will try. :)

JorgePe, 2017-09-03:
I believe me and @jabrena are the only ones using Mindsensors Grove Adapter with ev3dev :)
For me, direct I2C commands are fine.
But if someone is willing to write a driver, I can at least help with the tests.

jabrena, 2017-09-03:
The `Iberia` team!!! :D

Nyceane, 2019-12-10:
@JorgePe, I am using it, and can't find much resources...  I am having a difficult enough time just reading the data...

closed 2020-08-17 by dlech