ev3dev/ev3dev #914: Error when transfering python file to /home/robot

lucky962, 2017-07-11:
I am wondering how I would be able to transfer a file from my pc to /home/robot on my ev3dev.
I tried using FileZilla as recommended by someone on a forum. Although when I drag it into /home/robot this comes up:
Error:	/home/robot/First.py: open for write: received failure with description 'Failure'
Error:	File transfer failed
I then look into the "failed transfers" section and see the reason saying "Could not start transfer"

I'm not sure if my hardware/software info matters but here it is anyways:
brickman v0.8.1
Kernel: 4.4.68-20-ev3dev-ev3
Model: LEGO MINDSTORMS EV3 Programmable Brick

Thanks in advance for anyone's help!

lucky962, 2017-07-13:
I have just changed my SD card and now everything is working fine!
Thank you @dlech for suggesting this for me!

closed 2017-07-13 by lucky962

dlech, 2017-07-13:
Awsome!