ev3dev/ev3dev #915: No Sharing Tab in Wi-Fi Properties

lucky962, 2017-07-12:
### What are you trying to do?
I am trying to connect my EV3 brick to the internet.

### What did you expect to happen?
I expected to see a sharing tab in my Wi-Fi properties to be able to share my internet.

### What actually happened?
There was no sharing tab in Wi-Fi properties.

### What hardware and software are you using (including version numbers)?
brickman v0.8.1
Kernel: 4.4.68-20-ev3dev-ev3
Model: LEGO MINDSTORMS EV3 Programmable Brick

###Other things that might be important.
I searched this problem up and some people said that I need more than 1 internet connections. How do I do that?

dwalton76, 2017-07-12:
@lucky962 when you say `see a sharing tab (as I read in many tutorials)` do you mean a sharing tab on your windows machine? 

What is the problem you are trying to solve...is it just to get your ev3dev on wifi?  If so there are steps in the "With a Wi-Fi dongle" at http://www.ev3dev.org/docs/networking/

dlech, 2017-07-12:
Given the fact that you have had two unusual problems I'm starting to wonder if you have a bad SD card. Do you have another SD card you can try?

lucky962, 2017-07-12:
@dwalton76 Yes sorry, I forgot to say that I was on Windows 10. Yes, I was hoping to get it connected with Bluetooth but I guess I will just have to buy a Wi-Fi dongle.
@dlech I feel like the File Transfer didn't work because I didn't have internet but I'll try another SD card.

Thank you guys for the help!

closed 2017-07-13 by lucky962

WasabiFan, 2017-07-13:
Glad to hear that everything worked out! If you have suggestions for how our docs could be improved, don't hesitate to tell us.