ev3dev/ev3dev #916: USB Tethering (No Internet) for WRO Use

LaiYanKai, 2017-07-14:
Hi,
How do I tether using USB or bluetooth quickly to the EV3 from a windows 7 or 10 PC? This tether must not connect to the internet because the wifi and bluetooth have to be turned off during the World Robotics Olympaid competition. 

Im quite new to SSH and networking, but I have followed the tutorials and done both wifi and usb ssh tethering. However, they are all done via an active internet connection and wifi. If this can be done offline without turning on any wireless connection, and quickly (if im not wrong, can the file be compiled on the PC before downloading into the robot? This saves a lot of time as the EV3 is much slower in compilation), it is a really fantastic and versatile platform to use, even more so than RobotC and other platforms.

Please kindly reply as the competition is happening in about a month and half... it is a really great platform and I really do not want to miss this opportunity.

dlech, 2017-07-14:
You can manually assign IP addresses to both the host computer and the EV3.

LaiYanKai, 2017-07-17:
Hi I managed to get it connected via usb by checking the tethering gadget option on the ev3 and then looking for the ip address in the tethering network info, with and without the wifi enabled. The one on the top is for the internet connection and cannot be used for offline tethering. Thanks.

I am now looking to implement some speed regulation and syncing of the motors.

LaiYanKai, 2017-07-22:
Hi,
I was able to execute python files directly on the ev3. However, it always compiles the py files first before it runs and this takes a very long time. Considering that my teams need to time their runs, this is undesirable. Running straight from the ssh is faster but we cannot do that. I have also tried the docker tutorial but it cannot compile python files, it works only for C.

1) so, is there a way to save the compiled python files on the ev3 and access  it in the file browser? Where can I find the compiled files after executing from the ev3 and from ssh?

2) Is there a way for the docker to compile python files? I might have to install packages but I dont really know how.

3) Is there ultimately a way to start the program immediately upon executing the compiled (pyc?) files, just like in ev3g, so there is no delay as mentioned?

dlech, 2017-07-22:
You can compile python programs with `python3 -m py_compile my_program.py`. This should work on the EV3 or in Docker. But, unless you have a really huge program, there is no advantage to using Docker. The compiled files will be in a `__pycache__/` subdirectory in the same directory as your `.py` file, but you still launch the program using the `.py` file and it will find the cached compiled file.

This may improve startup time somewhat, but python programs will still not start instantaneously like EV3-G programs because it takes a noticeable amount of time for the python runtime to start up before your code actually runs.

One trick that was done for the Open Roberta project is that they created a service written in python that runs in the background. This service can then dynamically load and run other python programs. This eliminates the startup delay from the runtime since python is already running. [[code](https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/roberta/lab.py)]

LaiYanKai, 2017-07-23:
Hi thanks for the tip for py_compile.
Can this background service be ran automatically once the ev3 has booted? If so, how can I implement it?

dlech, 2017-07-23:
[Here is how OpenRoberta does it](https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/debian/openrobertalab.service). If you create a service file like this and save it in `/etc/systemd/system/` and then run `sudo systemctl daemon-reload`, you will be able to use it like any systemd service.

More info in [systemd.unit man page](https://manpages.debian.org/jessie/systemd/systemd.unit.5.en.html).

LaiYanKai, 2017-07-24:
Thanks for the help but I have really no clue how to do it. However I just noticed that we can start the robot using the centre button after the program finished its startup. This means the compilation time wont eat into the challenge time.... still, it will be great if ev3dev have this feature, so we can test the program without waiting for too long...

Really grateful for the help. Ev3dev imo is a helpful step towards rpi or arduino programming, so I'm introducing this into my high school. Thanks.

LaiYanKai, 2017-12-26:
Hi,
May I kindly know if there is a faster way to upload files onto the EV3 using MobaXTerm? The current SSH is a quite slow...

dlech, 2017-12-26:
You could setup a FTP server on the EV3. It will be *slightly* faster because it does not have to encrypt/decrypt files. But, be aware that if you are using Wi-Fi, people will be able to see your password.

https://wiki.debian.org/FTP#FTP_Servers

LaiYanKai, 2017-12-28:
Thank you. I would look into it.

closed 2018-02-02 by LaiYanKai