ev3dev/ev3dev #917: BrickPi ports issue

yash-1337, 2017-07-14:
### What are you trying to do?
I just got started with my raspberry pi 2, brickpi+, and ev3dev.
I was trying to use python to run a ev3 large motor connected to port A.
this is the code:
`import ev3dev.ev3 as ev3`
`m = ev3.LargeMotor('outA')`
`m.run_timed(time_sp=3000, speed_sp=500)`

### What did you expect to happen?
I expected the motor to run.

### What actually happened?
The motor did not run and said device is not connected.
I removed the 'outA' parameter and then reran the program and it worked.
I tried plugging the motor into a different port and changed the code but still it did not work.
For some reason it doesn't recognize the port names or something....
### What hardware and software are you using (including version numbers)?
I am using the raspberry pi 2 model B
Brickpi+
I checked and I am running the latest ev3dev system :

**System info (from `ev3dev-sysinfo`)**

Image file:         ev3dev-jessie-rpi2-generic-2017-06-09
Kernel version:     4.4.68-20-ev3dev-rpi2
Board:              Raspberry Pi 2 Model B Rev 1.1
Revision:           a21041
Brickman:           0.8.1
ev3devKit:          0.4.3

dlech, 2017-07-14:
The port names are different on BrickPi. See http://www.ev3dev.org/docs/platform-comparison/

yash-1337, 2017-07-14:
Oh wow... I am so stupid. I had seen that page and I KNEW I was forgetting something. Thanks for the quick reply 😄

closed 2017-07-14 by yash-1337