ev3dev/ev3dev #918: cannot send files to ev3 from desktop Mac using SCP

FRNetwork, 2017-07-17:
OK, I'm just not quite understanding the responses here to "send files to ev3 from desktop Mac #898", and I am having the same problem. I can SSH into my ev3 just fine and have been running all sorts of commands to it such as moving motors, talking etc, but now I am trying to upload a .py file to run a program and I'm stuck.
So:

I open a new terminal shell window
I ssh into it with ssh robot@192.168.1.75
once I'm in it, all is well, no errors.
I try running a scp command there and it just sits and waits forever.
I open a new terminal shell window and I cd to where the file is locally on my mac
then I try to run the scp command to push it from my mac to the ev3 and I get asked for passwords for an account that doesnt exist, my local user name from my mac on the robot. Since this account doesnt exist (the robot account exists), I cant move the file. I tried sudo -u robot but then my mac says that account doesnt exist locally.
Commands run: 
scp RobotProgram.py 192.168.1.75:/home/robot 
or
scp test.py ev3dev.ev3:/home/robot

I'm missing something fundamental and I need a nudge, please help.

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-06-09
Kernel version:     4.4.68-20-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2017-07-17:
You need to add `robot@` before the hostname/IP address. And it is `ev3dev.local`, not `ev3dev.ev3` (and `~` is a shortcut for `/home/robot`).

    scp test.py robot@ev3dev.local:~

dlech, 2017-07-17:
Oops, the above is to send from the host computer to the EV3. To send from the EV3 to the host computer, replace `robot` with your username on the host computer and `ev3dev.local` with the name or IP address of your host computer. For me, this would be...

    scp test.py david@gerda.local:~

dlech, 2017-07-17:
Or you can do it the other way around. On the host computer, run...

    scp robot@ev3dev.local:~/test.py .

This will copy `/home/robot/test.py` from the EV3 to the current directory (`.`) on the host computer.

FRNetwork, 2017-07-17:
Thank you, and now that it has been added the to the known hosts this will be easier in the future. I'm trying to teach myself Python and some of the SSH ins and outs are new to me.

closed 2017-07-17 by FRNetwork