ev3dev/ev3dev #919: ev3dev-buildscripts : help needed

JorgePe, 2017-07-18:
Sorry if this is trivial but I'm stuck following ev3dev-buildscripts guidelines at
https://github.com/ev3dev/ev3dev-buildscripts

I'm running Ubuntu 16.10
`4.10.0-26-lowlatency #30-Ubuntu SMP PREEMPT Tue Jun 27 10:37:28 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`

At step 2, after the 2 git commands complete and I change to my work directory I get a fatal error trying to manually pull driver submodule:

```
~/Documents/Personal/LEGO/ev3dev/work/ev3-kernel/drivers/lego$ git pull origin master
fatal: Couldn't find remote ref master
```

I've no experience with git but it seems I'm missing the definition of master:
```
git branch -a
* (HEAD detached at 63dbce8)
  ev3dev-stretch
  remotes/origin/HEAD -> origin/ev3dev-stretch
  remotes/origin/ev3dev-jessie
  remotes/origin/ev3dev-stretch
  remotes/origin/issue-661
  remotes/origin/ms-pps58-nx
  remotes/origin/wedo-names
  remotes/origin/wedo-servo
```

dlech, 2017-07-18:
Hmm... maybe I forgot to push something... let me check...

dlech, 2017-07-18:
Wait a minute. the problem is `master` there is no branch name `master`.
 
just do 

    git checkout ev3dev-stretch
    git pull

JorgePe, 2017-07-18:
Thanks, it worked.
I'm suprised, just 8 minutes to compile a kernel, I was afraid it would take much longer. Now I can afford to make much more mistakes :)

closed 2020-08-17 by dlech