ev3dev/ev3dev #920: Hi! I didn't find ability to put Hi-Technic Color Sensor V2 in passive mode (sensor led turns off) seen in EV3-G. Is there a way to do this?

nikmachines, 2017-07-21:
### What are you trying to do?

### What did you expect to happen?

### What actually happened?

### What hardware and software are you using (including version numbers)?

dlech, 2017-07-21:
Can you post a picture of the EV3 block? I don't see anything at http://www.hitechnic.com/cgi-bin/commerce.cgi?preadd=action&key=NCO1038 that indicates that you can turn the LED off. However, https://github.com/botbench/robotcdriversuite/blob/master/include/hitechnic-colour-v2.h does mention a "passive mode" command, but it is not clear that this turns off the LED.

nikmachines, 2017-07-24:
Here it is. In EV3-G passive mode turns led off.
[image]

dlech, 2017-07-24:
When you select passive, what are the outputs? Just red, green, and blue or something else?

dlech, 2017-07-25:
I added a `PASSIVE` mode in the `21-ev3dev` kernel. I don't have this sensors, so I have no idea if it works or not, but you can try it out.

nikmachines, 2017-07-28:
PASSIVE mode does not power leds off...

dlech, 2017-07-28:
As far as I can tell, we are doing exactly what the EV3-G block does. Here are the internals:

[image]

[image]

It looks like I did make a mistake with the data type though. It should be 16-bit instead of 8-bit.

If you are absolutely sure that passive mode does indeed turn off the LED, the someone with a logic analyzer will need to take a look at what is actually going over the wires. If someone wants to send me one of these sensors (for keeps), I can do it.

nikmachines, 2017-08-04:
Sorry, i was wrong - it seems to be working! Led is definitely off in "PASSIVE" mode :) But i have strange readings for value(0) and value(1): either "0" or "1"

closed 2017-08-04 by nikmachines