ev3dev/ev3dev #921: Bluetooth Audio and brickman

JorgePe, 2017-07-23:
Hi again.
I am using a BT speaker with ev3dev (snapshot-ev3dev-stretch-ev3-generic-2017-06-27.img) and it is working fine from the shell (SSH) after doing these steps (based on #198):

- sudo apt-get install --no-install-recommends pulseaudio pulseaudio-module-bluetooth
- Enable BT on brickman
- pair with BT speaker (brickman works but usually use bluetoothctl)

Audio works fine from the shell:
`paplay -d bluez_sink.C7_B5_42_B4_72_EC.a2dp_sink /usr/share/sounds/alsa/Front_Center.wav`

Now I want to use python but if possible without writing my own subprocess functions to call shell scripts.
So I created a /etc/asound.conf

```
pcm.pulse {
    type pulse
}

ctl.pulse {
    type pulse
}
```

and  this allows me to pass ALSA to PulseAudio so I can run

`aplay -Dpulse /usr/share/sounds/alsa/Front_Center.wav`

this also allows me use python and the ev3.Sound methods 'speak' and 'play'

```
#!/usr/bin/env python3
from time import sleep
import ev3dev.ev3 as ev3
ev3.Sound.speak('Hello').wait()
ev3.Sound.play('/usr/share/sounds/alsa/Front_Center.wav')
```

but not tone nor beep because they bypass ALSA and go straight to the internal speaker.

Sometimes after a reboot this doesn't work because the default sink changes between EV3 speaker and BT speaker so I fix that with

`pactl set-default-sink bluez_sink.C7_B5_42_B4_72_EC.a2dp_sink`

Now my problem: if I execute the same python script from brickman it uses the EV3 speaker, I think because brickman runs as root and root sees only one sink (the EV3 speaker).

I forgot something?

dlech, 2017-07-24:
According to https://wiki.archlinux.org/index.php/PulseAudio/Examples#Set_the_default_output_source

> To set it as the system wide default, add the following to /etc/pulse/default.pa
>
>     set-default-sink alsa_output.pci-0000_04_01.0.analog-stereo

JorgePe, 2017-07-24:
Thanks, already had that.
I think I must have missed (or messed) something as pulseaudio was not running at startup so it couldn't connect to a non-existing default sink.

After repeating 'systemctl enable pulseaudio.service' it now runs. It doesn't autoconnect to BT speaker at startup but after I manual connect with bluetoothctl I can I now see the BT sink if I login as root.

Strange now I have to use sudo for the pactl command to work with robot and my python script also needs to be run with sudo. And from brickman it no longer uses the internal speaker but it also doesn't use the BT speaker but now I see error messages "connection refused"

So I must have messed up the permissions, it's better to start all over.

JorgePe, 2017-07-24:
Yeap, after adding root and robot again to pulse-acess and audio I can now run my python script without sudo through SSH and finally through brickman as well.

Need to review my process.

Thanks again.

closed 2017-07-24 by JorgePe