ev3dev/ev3dev #922: Plug-in for V-REP (LDraw/LeoCAD + ev3dev)

Symbian9, 2017-07-23:
### What did you expect to happen?

Import/Export LDraw models + simulating EV3 robots in V-REP PRO EDU
* https://github.com/leozide/leocad/issues/86

### What actually happened?

No any 3D simulators of programmed EV3 robots for Linux-based desktop PC. V-REP is good free (partially open-source) 3D robotics simulator, that support scripting using Lua, C++, etc.)

### What hardware and software are you using (including version numbers)?

* LeoCAD 17.07 and above
  * LDraw library 16.01 with updates
* V-REP PRO EDU 3.4.0 (Linux, amd64)

dlech, 2017-10-24:
How does ev3dev fit into this?

Symbian9, 2017-10-25:
ev3dev fit into this because for realize this idea need make scripts for V-REP, that will include logic and commands of ev3dev bindings