ev3dev/ev3dev #924: missing USB net driver dm9601 for ethernet wired USB device on EV3 image

BertLindeman, 2017-07-28:
### What are you trying to do?
Trying to use an USB - Ethernet dongle on my EV3

### What did you expect to happen?
Automagically load driver and work.
From my debian laptop at debian 16.4.2:
uname -a
Linux mumba-ubuntu 4.4.0-87-generic #110-Ubuntu SMP Tue Jul 18 12:55:35 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
From debian 16.4.2:
```
[vr jul 28 10:25:44 2017] usb 3-1: new full-speed USB device number 3 using xhci_hcd
[vr jul 28 10:25:45 2017] usb 3-1: New USB device found, idVendor=0fe6, idProduct=9700
[vr jul 28 10:25:45 2017] usb 3-1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[vr jul 28 10:25:45 2017] usb 3-1: Product: USB 2.0 10/100M Ethernet Adaptor
[vr jul 28 10:25:45 2017] dm9601 3-1:1.0 eth1: register 'dm9601' at usb-0000:04:00.0-1, Davicom DM96xx USB 10/100 Ethernet, 00:e0:4c:53:44:58
[vr jul 28 10:25:45 2017] usbcore: registered new interface driver sr9700
[vr jul 28 10:25:45 2017] dm9601 3-1:1.0 enx00e04c534458: renamed from eth1
[vr jul 28 10:25:45 2017] IPv6: ADDRCONF(NETDEV_UP): enx00e04c534458: link is not ready
[vr jul 28 10:25:45 2017] dm9601 3-1:1.0 enx00e04c534458: link up, 100Mbps, full-duplex, lpa 0xFFFF
```
So I assume the driver would be dm9601

Some of the modules on deb 16.4.2:
```
# lsmod | grep -e "dm9" -e "sr97" -e usbnet
sr9700                 16384  0
dm9601                 16384  0
usbnet                 45056  2 dm9601,sr9700
mii                    16384  4 r8169,dm9601,sr9700,usbnet
```

### What actually happened?
USB device recongnized, but driver module dm9601 not loaded nor found
```
# modprobe dm9601
modprobe: FATAL: Module dm9601 not found in directory /lib/modules/4.9.39-ev3dev-1.3.0-ev3
# modprobe sr9700
modprobe: FATAL: Module sr9700 not found in directory /lib/modules/4.9.39-ev3dev-1.3.0-ev3
```

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2017-06-27
Kernel version:     4.9.39-ev3dev-1.3.0-ev3
Board:              LEGO MINDSTORMS EV3
Revision:           0000
Brickman:           0.9.0
ev3devKit:          0.5.1
```
Removed this line from the output above:
```/usr/bin/ev3dev-sysinfo: line 68: warning: command substitution: ignored null byte in input```

I do see dm9601 on other ev3dev images.
Would it be possible (easily) to include dm9601 and it requisites in a future ev3 image?

dlech, 2017-07-28:
Sure, we can add them.

BertLindeman, 2017-07-28:
Great! thanks.

BertLindeman, 2017-08-12:
Updated my strech image to 
```
Linux ev3dev 4.9.42-ev3dev-1.4.0-ev3 #1 PREEMPT Fri Aug 11 11:33:00 CDT 2017 armv5tejl GNU/Linux
```
Reboot nicely provides the ethernet connection 👍 
Thanks

closed 2017-08-12 by BertLindeman