ev3dev/ev3dev #925: Stretch issues with BT

JorgePe, 2017-07-28:
Using stretch image 2017-07-26 for EV3
(to test Visual Studio Code extension)

After power up, BT is "not available in Brickman". After reboot it is available.

Also happened a few times with a previous version (at least one week ago) when trying to use BT speaker. At that time I didn't pay attention because I was messing with pulseaudio and permissions.

When this happens...
connmanctl technologies doesn't show bluetooth
hciconfig -a shows nothing
dmesg has some "Bluetooth:" messages

dlech, 2017-07-28:
There have been a number of other issues about similar sorts of problems. The best I can tell is that there is some sort of crosstalk between UART lines or some sort of blocking in the kernel between tty devices. Along with the bluetooth messages, there are usually UART sensor error messages (if you have UART sensors attached during boot).

I was noticing this this week as well. It seems there is a better chance that bluetooth will start correctly if you don't have any UART sensors attached.

JorgePe, 2017-07-28:
I only had the Codatex RFID sensor attached (thats how I found that I2C modes were unavailable). And a USB hub with the Edimax Wi-Fi. Not even my usual BT4 USB dongle.

dlech, 2017-09-02:
https://github.com/ev3dev/ev3-systemd/commit/e9bc0cffb3ddb4e30e60ce72e4df88d3f086deb4

dlech, 2017-09-02:
This should be better in the 2017-09-01 snapshot image. Although currently just using a workaround and not really fixing the problem (so don't close the issue yet event if it seems to be working).

https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2017-09-02/

obo, 2018-04-24:
Just confirming that snapshot-ev3dev-stretch-ev3-generic-2018-04-22.img.xz still has this problem.
After a fresh flashing of my SD with this build, brickman Bluetooth says not available.

Sadly, neither reboot, nor a cold start helped.

I also detached all the motors and all the sensors and did a cold start but I still have Bluetooth not available.

Any clues? (I don't have other ways of connecting.)

dlech, 2018-04-24:
I don't have an immediate solution unfortunately. I do have a potential solution, but it involves backporting about 50 patches from the mainline kernel or waiting for the next LTS kernel (which hasn't been announced yet).

dlech, 2018-04-24:
Please open a new issue for the USB connection problems and include screenshots of the errors.

dlech, 2018-04-27:
It turns out @obo 's bluetooth problem was a new one. Some required kernel modules got turned off when we updated to the 4.14.y kernel. Fix will be in the next kernel release.

https://github.com/ev3dev/ev3-kernel/commit/cc7b1c2339af1d1b167bcab94f6ec25986f352b3

Leaving this issue open for intermittent issues with Bluetooth not starting on boot.

dlech, 2018-07-15:
There were some CPU usage improvements made in the `ev3dev-2.2.0` kernel release that have hopefully made this not a problem anymore.

JorgePe, 2018-10-06:
I think I have bad news... been using stretch with Microsoft Xbox Adaptive Controller, no problems at all with internal bluetooth and USB wifi Edimax... till yesterday when I decided to test it with a rover with 2 large motors. Now I have have to reboot perhaps 1 in 3 or 2 in 5 because I don't have bluetooth after boot.

ev3dev-sysinfo 
Image file:         ev3dev-stretch-ev3-generic-2018-05-24
Kernel version:     4.14.71-ev3dev-2.3.0-ev3
Brickman:           0.10.0
BogoMIPS:           148.88
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534F4DD3
BOARD_INFO_TYPE=main

The system is updated and have also installed python-ev3dev2 to use MoveSteering.

dlech, 2018-10-06:
So if you disconnect motors during boot, is Bluetooth more reliable?

staadecker, 2019-04-14:
I'm having the same issue as @JorgePe . Any updates?

prithvimoses, 2019-04-19:
Hi have the same issue with 2 large motors and a few other devices connected, Bluetooth shows up as unavailable on boot and reboot. After multiple reboots, it works

dlech, 2019-07-04:
Should be fixed, or at least greatly improved in `ev3-systemd 5.4.2`.

I also noticed that Bluetooth doesn't work if the batteries are low (< 5V).

closed 2019-07-04 by dlech

JanderII, 2019-11-07:
Requoting from the FB ev3dev page:

"I'm having no joy at all with getting bluetooth active - it just says "not available".

I'm using ev3dev-stretch-ev3-generic-2019-10-23, and the only thing plugged in is an Edimax Nano. I've tried with two EV3s, and both show the same issue. Even after multiple reboots."

As requested, here's my ev3-sysinfo and the bluetooth systemd status:

root@ev3dev:/home/robot# ev3dev-sysinfo
Image file:         ev3dev-stretch-ev3-generic-2019-10-23
Kernel version:     4.14.117-ev3dev-2.3.4-ev3
Brickman:           0.10.2
BogoMIPS:           148.88
Bluetooth:
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534E2A13
BOARD_INFO_TYPE=main
root@ev3dev:/home/robot# systemctl status ev3-bluetooth | cat
● ev3-bluetooth.service - LEGO MINDSTORMS EV3 built-in Bluetooth
   Loaded: loaded (/lib/systemd/system/ev3-bluetooth.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2019-11-06 22:06:25 UTC; 11h ago
  Process: 461 ExecStart=/usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 flow nosleep (code=exited, status=1/FAILURE)
  Process: 342 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)

Nov 06 22:06:13 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 built-in Bluetooth...
Nov 06 22:06:23 ev3dev hciattach[461]: Found a Texas Instruments' chip!
Nov 06 22:06:23 ev3dev hciattach[461]: Firmware file : /lib/firmware/ti-connectivity/TIInit_6.6.15.bts
Nov 06 22:06:23 ev3dev hciattach[461]: Loaded BTS script version 1
Nov 06 22:06:23 ev3dev hciattach[461]: texas: changing baud rate to 2000000, flow control to 1
Nov 06 22:06:25 ev3dev hciattach[461]: Can't init device hci0: Operation not possible due to RF-kill (132)Can't initialize device: Operation not possible due to RF-kill
Nov 06 22:06:25 ev3dev systemd[1]: ev3-bluetooth.service: Control process exited, code=exited status=1
Nov 06 22:06:25 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 built-in Bluetooth.
Nov 06 22:06:25 ev3dev systemd[1]: ev3-bluetooth.service: Unit entered failed state.
Nov 06 22:06:25 ev3dev systemd[1]: ev3-bluetooth.service: Failed with result 'exit-code'.

I piped to cat due to it systemctl chopping the line length in my terminal.

I've installed rfkill just to check and I get:

root@ev3dev:/home/robot# rfkill list all
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no

There's no apparent hci device, so I think the RF-Kill part may be a red herring?

Jerry

JanderII, 2019-11-07:
Ok. After a lot of tinkering and many, many, reboots I have it working. The sequence of events that seems to have got it going are related to that RF-kill that I thought was irrelevant. I did the following:

1) Installed rfkill
2) Ran "usr/bin/hciattach -t 30 /dev/ttyS2 any 2000000 flow nosleep" - note the "any", not "texas"
3) Ran "rfkill list all" ->

# rfkill list all
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: yes
	Hard blocked: no

4) Unblocked the hci0 device:
# rfkill unblock 1
# rfkill list all
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no

5) Rebooted the EV3

After the reboot I checked "systemctl status ev3-bluetooth" and it came up with:

● ev3-bluetooth.service - LEGO MINDSTORMS EV3 built-in Bluetooth
   Loaded: loaded (/lib/systemd/system/ev3-bluetooth.service; static; vendor preset: enabled)
   Active: active (running) since Thu 2019-11-07 10:43:33 UTC; 2min 13s ago
  Process: 459 ExecStart=/usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 flow nosleep (code=exited, status=0/SUCCESS)
  Process: 348 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
 Main PID: 465 (hciattach)
   CGroup: /system.slice/ev3-bluetooth.service
           └─465 /usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 flow nosleep

Nov 07 10:43:22 ev3dev systemd[1]: Starting LEGO MINDSTORMS EV3 built-in Bluetooth...
Nov 07 10:43:32 ev3dev hciattach[459]: Found a Texas Instruments' chip!
Nov 07 10:43:32 ev3dev hciattach[459]: Firmware file : /lib/firmware/ti-connectivity/TIInit_6.6.15.bts
Nov 07 10:43:32 ev3dev hciattach[459]: Loaded BTS script version 1
Nov 07 10:43:32 ev3dev hciattach[459]: texas: changing baud rate to 2000000, flow control to 1
Nov 07 10:43:33 ev3dev hciattach[459]: Device setup complete
Nov 07 10:43:33 ev3dev systemd[1]: Started LEGO MINDSTORMS EV3 built-in Bluetooth.
root@ev3dev:/home/robot#

It's now available in the Brickman menus.

Now, this may be totally coincidental, but the fact that it came back up after an rfkill unblock on the hci device does look like it may be related.

Since then I've done a reboot, and a power down + remove the batteries, and it's still working.

dlech, 2019-11-07:
Thanks for the detailed info. This issue is actually closed because the underlying problem was fixed. We have #1314 for ongoing Bluetooth not available issues that have a different, yet to be determined, root cause.

ashindea, 2020-01-27:
RE-inserting battery pack fixed it
Using the latest ev3dev download ( Dec 27 2019). I re-booted 7 times without any input or output ports connected. It kept saying on BT " Not available".
The used an old trick from pre-smart phone days and removed and re-inserted the battery.
BT worked immediately on next reboot