ev3dev/ev3dev #928: rpob

jabrena, 2017-07-31:
Hi, 

I am trying to connect with a Brick with `LeJOS` using ssh but I receive the following error:

```
robot@brick1:/usr/lib/jni$ ssh -oKexAlgorithms=+diffie-hellman-group1-sha1 root@192.168.1.85
Unsupported KEX algorithm "+diffie-hellman-group1-sha1"
command-line line 0: Bad SSH2 KexAlgorithms '+diffie-hellman-group1-sha1'.
```

From my OSX, I don´t have any problem but from the brick, I have that problem. Any way to fix the issue?

Juan Antonio

dlech, 2017-07-31:
Why are you specifying an algorithm? Why not let ssh figure out which one to use?

You could always read the [man page](https://manpages.debian.org/stretch/openssh-client/ssh.1.en.html).

```
KexAlgorithms
    Specifies the available KEX (Key Exchange) algorithms. Multiple algorithms must be comma-separated. Alternately if the specified value begins with a ‘+’ character, then the specified methods will be appended to the default set instead of replacing them. The default is:

    curve25519-sha256,curve25519-sha256@libssh.org, 
    ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521, 
    diffie-hellman-group-exchange-sha256, 
    diffie-hellman-group-exchange-sha1, 
    diffie-hellman-group14-sha1
        

    The list of available key exchange algorithms may also be obtained using “ssh -Q kex”.
```

I'm guessing `diffie-hellman-group1-sha1` won't be listed if you run this command.

dlech, 2017-07-31:
you should probably leave out the `+`.

jabrena, 2017-07-31:
Fixed :D 

```
robot@brick1:/usr/lib/jni$ ssh -oKexAlgorithms=diffie-hellman-group1-sha1 root@192.168.1.85
The authenticity of host '192.168.1.85 (192.168.1.85)' can't be established.
RSA key fingerprint is 6c:59:d2:e4:e6:78:fd:51:5d:be:85:a8:ea:a1:e5:7c.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added '192.168.1.85' (RSA) to the list of known hosts.
root@192.168.1.85's password: 
root@EV3:~# 
```

closed 2017-07-31 by jabrena

jabrena, 2017-07-31:
Sorry @dlech, 

I didn´t answer your question:

Yes I specify the `ssh` algorithm because, it is the unique way to connect.
https://lejos.sourceforge.io/forum/viewtopic.php?t=8188
https://sourceforge.net/p/lejos/wiki/Connecting%20to%20the%20EV3%20with%20ssh/

Cheers

Juan Antonio

epicn1337, 2021-01-30:
thanks for sharing this it worked for me :)

epicn1337, 2021-01-30:
> Why are you specifying an algorithm? Why not let ssh figure out which one to use?
>
> You could always read the [man page](https://manpages.debian.org/stretch/openssh-client/ssh.1.en.html).
>
> ``` [...]

thanks for sharing this it helped me root the machine i'm working on LOL