ev3dev/ev3dev #929: NXTCamv5 Stream images

jabrena, 2017-07-31:
Hi @dlech, 

I am testing NXTCam v5 and I would to know if you observed some way to stream the image from NXTCam on EV3Dev. At the moment, with the API, it is possible to take the picture but do you know if it is possible to store to manage by a third party program or store in the EV3Dev file system or something similar.

Many thanks in advance.

Juan Antonio

dlech, 2017-07-31:
Like this?

[image]

jabrena, 2017-08-01:
Ehhh, it is magic!!

Any example in some place?

dwalton76, 2017-08-01:
I do not have one of these cameras, can you get to it via `/dev/video0`?  If so I can send you a little python program that runs a web server and serves the video from the camera.

jabrena, 2017-08-01:
Hi @dwalton76 that idea is amazing. 

But we need the technical details to tinker!!! :p

dwalton76, 2017-08-01:
It is a little buggy (I only use it for lining up my camera every once in a while) but give this a try

```
#!/usr/bin/env python2

"""
Author: Igor Maculan - n3wtron@gmail.com
A Simple mjpg stream http server

https://gist.github.com/n3wtron/4624820
"""

# sudo apt-get install python-opencv
import cv2

# sudo apt-get install python-pil
from PIL import Image

import logging
import threading
from BaseHTTPServer import BaseHTTPRequestHandler,HTTPServer
from SocketServer import ThreadingMixIn
import StringIO
import time

log = logging.getLogger(__name__)

capture = None
shutdown = False

class CamHandler(BaseHTTPRequestHandler):

    def do_GET(self):
        if self.path.endswith('.mjpg'):
            self.send_response(200)
            self.send_header('Content-type','multipart/x-mixed-replace; boundary=--jpgboundary')
            self.end_headers()

            while not shutdown:
                try:
                    rc,img = capture.read()

                    if not rc:
                        log.info("Failed to capture image")
                        continue

                    imgRGB=cv2.cvtColor(img,cv2.COLOR_BGR2RGB)
                    jpg = Image.fromarray(imgRGB)
                    tmpFile = StringIO.StringIO()
                    jpg.save(tmpFile,'JPEG')
                    self.wfile.write("--jpgboundary")
                    self.send_header('Content-type','image/jpeg')
                    self.send_header('Content-length',str(tmpFile.len))
                    self.end_headers()
                    jpg.save(self.wfile,'JPEG')
                    time.sleep(0.05)
                except Exception as e:
                    log.exception(e)
                    break
        else:
            self.send_response(200)
            self.send_header('Content-type','text/html')
            self.end_headers()
            self.wfile.write('<html><head></head><body>')
            self.wfile.write('<a href="/cam.mjpg">camera</a>')
            self.wfile.write('</body></html>')

class ThreadedHTTPServer(ThreadingMixIn, HTTPServer):
    """Handle requests in a separate thread."""

def main():
    global capture
    capture = cv2.VideoCapture(0)
    capture.set(cv2.cv.CV_CAP_PROP_FRAME_WIDTH, 320); 
    capture.set(cv2.cv.CV_CAP_PROP_FRAME_HEIGHT, 240);
    capture.set(cv2.cv.CV_CAP_PROP_SATURATION,0.2);
    global img
    server = None

    try:
        PORT = 8000
        server = ThreadedHTTPServer(('0.0.0.0', PORT), CamHandler)
        log.info("server started on port %d" % PORT)
        server.serve_forever()
    except:
        global shutdown
        log.info("Caught Exception...shutting down")

        shutdown = True
        if capture:
            capture.release()

        if server:
            server.shutdown
            server.socket.close()

if __name__ == '__main__':
    logging.basicConfig(level=logging.INFO,
                    format='%(asctime)s %(filename)12s %(levelname)8s: %(message)s')
    log = logging.getLogger(__name__)

    # Color the errors and warnings in red
    logging.addLevelName(logging.ERROR, "\033[91m   %s\033[0m" % logging.getLevelName(logging.ERROR))
    logging.addLevelName(logging.WARNING, "\033[91m %s\033[0m" % logging.getLevelName(logging.WARNING))

    main()
```

jabrena, 2017-08-01:
I will try today on a EV3 Brick.

dwalton76, 2017-08-01:
cut-n-paste the code above to a file, say camera.py, then
```
chmod 755 camera.py
./camera.py
```

It will listen on port 8000 so in your web browser go to `http://x.x.x.x:8000/` where x.x.x.x is the IP address of your ev3.

jabrena, 2017-08-01:
Oki, I will try.

dlech, 2017-08-01:
This is not a webcam, so there is no `/dev/video*`, so @dwalton76's code won't do anything for you.

dlech, 2017-08-01:
The code I used for the video is [here](https://gist.github.com/dlech/6877e0d3f96ba563b2ca5dfded651d48). But as I mention in the gist, it won't actually work unless your compile GRX for yourself.

jabrena, 2017-08-01:
Hi @dlech,

I read the GIST but using the custom `main.py` on `NXTCam` is possible to transmit a frame so later, we could store on disk and in this case, it is not necessary to compile `GRX`, isn't it?

closed 2020-08-17 by dlech