ev3dev/ev3dev #930: How to detect Jessie or Stretch distro?

jabrena, 2017-08-05:
Hi,

I would like to know how to detect by `terminal`, if the brick are using a `Jessie Distro` or a `Stretch distro`. Any idea?

Juan Antonio

JorgePe, 2017-08-05:
```
cat /etc/os-release
```

I'm not near my EV3 but a few weeks ago this was the output:

stretch:
```
PRETTY_NAME="ev3dev-stretch"
NAME="ev3dev-stretch"
ID=ev3dev
ID_LIKE=debian
HOME_URL="http://www.ev3dev.org"
SUPPORT_URL="http://www.ev3dev.org/support"
BUG_REPORT_URL="https://github.com/ev3dev/ev3dev/issues"
```
jessie:
```
PRETTY_NAME="Debian GNU/Linux 8 (jessie)"
NAME="Debian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=ev3dev
ID_LIKE=debian
HOME_URL="http://www.ev3dev.org/"
SUPPORT_URL="http://www.ev3dev.org/support"
BUG_REPORT_URL="https://github.com/ev3dev/ev3dev/issues"
```

JorgePe, 2017-08-05:
Or perhapsh
```
cat /etc/debian_version
8.5
```

jabrena, 2017-08-05:
Hi @JorgePe,

Many thanks, I will test later with a `Bash script`.

Maybe the second option is better:
https://www.debian.org/releases/index.en.html

```
Debian 9 (stretch) — current stable release
Debian 8 (jessie) — obsolete stable release
``` 

At the moment, all my development was tested for Jessie, but I will have to test Stretch in some moment. In your case, did you migrate to Stretch?

jabrena, 2017-08-05:
Yes, using the second option, it is possible to check.

Later, I find some ways to get the integer part:
https://unix.stackexchange.com/questions/109958/bash-script-to-find-debian-release-number-from-etc-debian-version

```
robot@brick1:/dev$ cat /etc/debian_version
8.5
robot@brick1:/dev$ VERSION=$(sed 's/\..*//' /etc/debian_version)
robot@brick1:/dev$ echo $VERSION
8
```

Cheers

Juan Antonio

JorgePe, 2017-08-05:
`At the moment, all my development was tested for Jessie, but I will have to test Stretch in some moment. In your case, did you migrate to Stretch?`

Not yet. Had some problems with internal Bluetooth, sometimes it doesn't exist at startup so my BT audio doesn't work. Had one or two other minor issues (like one package I use gone from the universe) so I'm waiting.

But some things I use (lircd, bluez, kernel support, pybluez+gattlib) are already much newer on stretch so I keep returning to it once on a while to recheck. I have a python script for reading BT BLE proximity that works fine in Ubuntu and doesn't work at all in jessie, had to use an ugly bash workaround and for some reason I still doesn't understand doesn't worked in stretch last time I check.

Also I'm waiting for a BOOST next week... once I get my gatt scritps working on Ubuntu with it will try the same with ev3dev stretch.

With the new Visual Studio Code IDE I wlll surely check mor often.

closed 2017-08-05 by jabrena