ev3dev/ev3dev #931: Request for Color Sensor Support (Lego 9694)

logwet, 2017-08-06:
### What are you trying to do?
I am trying to use two lego color sensors in my project (id# 9694).

A a image of the [sensor](https://shop.lego.com/en-AU/Color-Sensor-9694):
[image]

### What did you expect to happen?
When using the default Mindstorms software the sensors are recognised and I'm able to use them, so I expected the sensors to automatically show up in ev3dev and brickman.

### What actually happened?
I tried to use the sensors using the Python3 REPL but no color sensors were recognised by the system, even when I explicitly stated the port. In addition, the sensors are also inaccessible from within a script and the sensors do not show up in Brickman under the connected sensors menu.

### What hardware and software are you using (including version numbers)?
EV3 Brick
ev3dev-jessie-2017-02-11

dlech, 2017-08-06:
We don't currently have a driver for this sensor. I am surprised that it has taken 4 years for someone to ask about it. 😄

logwet, 2017-08-07:
Heh heh, the [wiki page on sensors)(http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensors.html) recommended to open a Github Issue if a sensor wasn't supported or listed. I figured that because the sensor was not explicitly listed as unsupported nobody had asked for implementation in the past. I did however notice that *"The NXT Color Sensor is not usable at this point in time. __We can detect it with the auto-detect__, but we don’t have a driver for it yet."* - As listed on the wiki page. But because this sensor *wasn't* detected (so maybe it was different from the one listed in the wiki I figured that I might as well try. 😉 

So is this a sensor that can easily be supported (by writing a driver) or should I try something else?

Thanks,
logwet

dlech, 2017-08-07:
It should be detected. You can check with `cat /sys/class/lego-port/port*/state`.

For the driver, we basically just need to clean up the [official LEGO driver](https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-stretch/lms2012/d_analog.c#L801) and make it work with the ev3dev lego-sensor class.

logwet, 2017-08-07:
That's awesome! I don't want to be rude or presumptous, but is there a timeframe for this? The reason is because I have a deadline for this project (it's a entry for RoboCup Junior Aus) and if adapting the driver is difficult I may have to consider something else.

I really, really apologise for any inconvenience or rudeness; I just want a brief indication of what action I should take, and I understand if this isn't something that is high up on your priority.

dlech, 2017-08-07:
I don't have any definite plans.

WasabiFan, 2017-08-07:
Am I correct that the EV3 color sensor (not NXT) works already? I believe it does; if so, @logwet you could use those instead of the older NXT ones for your project.

logwet, 2017-08-07:
@WasabiFan I have since made use of two HiTechnic NXT color v2 sensors. They are recognised and there is a registered driver. I am having issues with associating the device with the color sensor class in ev3dev python though. The sensor is recognised but I receive several error messages that state that a color sensor is not connected...
I'm very confused.

WasabiFan, 2017-08-07:
@logwet if you want to investigate that, please open an issue at the ev3dev-lang-python repository and we can look into it over there. I don't want to clutter this thread with the separate problem.

logwet, 2017-08-08:
I have opened a [new thread](https://github.com/rhempel/ev3dev-lang-python/issues/333) on the ev3dev python repo.

dlech, 2017-08-08:
We'll leave this open until the LEGO NXT Color sensor driver is actually implemented.

jabrena, 2018-02-14:
Hi @dlech, I received a request to add support for this sensor from Java side.

Does exist a guide or some technical documentation to add a Driver in EV3Dev project?

Many thanks in advance.

Juan Antonio

dlech, 2018-02-14:
http://www.ev3dev.org/contribute/#write-a-driver

https://github.com/mindboards/ev3sources/blob/master/lms2012/d_analog/Linuxmod_AM1808/d_analog.c