ev3dev/ev3dev #935: Detect that a EV3 Brick is charging

jabrena, 2017-08-14:
Hi,

Does exist some property in the file system to detect that an EV3 Brick is recharging with the power supply?

`/sys/class/power_supply/legoev3-battery/`

https://www.amazon.co.uk/LEGO-NXT-EV3-Power-Supply-x/dp/B00901ZT5Q

Juan Antonio

jabrena, 2017-08-14:
In the docs, I don't any property:
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/ev3.html#battery

dlech, 2017-08-14:
This is not possible. The battery does not send any signals to the EV3.

Perhaps you could use a color sensor or light sensor to detect the external LEDs on the battery?

jabrena, 2017-08-14:
Yes, this the POC which I am building but I made the question if you know in order to save one sensor. Maybe, this week, I will publish something.

Many thanks @dlech

Juan Antonio

JorgePe, 2017-08-15:
Tracking "voltage_now" wouldn't allow to know that? When it increases (and if it is not a glitch) then it is charging.

dlech, 2017-08-15:
Maybe. But voltage will also increase when you stop the motors or unplug a USB device, so how do you tell the difference?

JorgePe, 2017-08-15:
You'll have to measure always in same "state", like all motors off and most sensors idle.
Unplugging a USB device is a non-usual scenario in robot operation but if it expected you should also check dmesg/lsusb.
I think it will work for a docking station / recharging base scenario. The problem is time... how many seconds/minutes are necessary to wait to be sure that it is indeed charging?

jabrena, 2017-08-16:
Hi,

Reading the battery with a Thread is possible but you know, that the values need a time to increase. The best way that i know to know that the brick is charging is using the color sensor to detect the green color. Using this detection is inmediate and it is useful to stop another behavior like aproximation.

I am working in a Prototype. I will show soon.

Cheers

JorgePe, 2017-08-16:
If you don't want to "waste" a sensor you could make a smart charging base that sends through bluetooth or wi-fi the acknowledgement.
Yes, more complex than just a passive version but that could allow extra features like powering it off when not in use or when the robot already full charged but still engaged.

Or just forget it and add a sensor mux to your robot. Color sensors work fine with it, I already got 8 working in just one EV3.

jabrena, 2017-08-16:
Hi Jorge, yes exist multiple alternatives. :)

Give me a chance for the solution for a EV3 Color Sensor. I will inform with the results. We will research more alternatives. :)

In few hours, I will continue with it.

dwalton76, 2017-08-16:
I like @JorgePe 's idea....maybe use something like this and have the ev3 query it via wifi to ask how much power it is currently drawing

https://www.amazon.com/TP-Link-Monitoring-Required-Anywhere-HS110/dp/B0178IC5ZY/ref=sr_1_5?ie=UTF8&qid=1502886211&sr=8-5&keywords=monitor+power+outlet

jabrena, 2017-08-16:
@dwalton76, @JorgePe now I understand the idea from Jorge. Yes, it is a good indirection. 
But maybe the http pooling require more power consumption than the usage of a EV3 Sensor, but anyway we should test and play. 

I will try to test with the sensor and later, I will test with the product that you say @dwalton76, but yes, exist more alternatives.

I reopen the issue, it is not closed :p

Juan Antonio

dwalton76, 2017-08-16:
Disclaimer I haven't used that product...just googled and found it

JorgePe, 2017-08-17:
I've used simillar, can't remember the brand on this video but I think it is an Orvibo S20, quite cheap and a big sucess in Domotics at least here in Portugal:
https://www.youtube.com/watch?v=3sZCVXsysx4

I also used at least one BT BLE power plug, the "Awox smartplug".

But never used their monitoring features, just the ON/OFF.

jabrena, 2017-08-20:
I can certificate that using EV3 Color sensor, is possible to read the status of the EV3 battery:

**Idea:**
[image]

[image]

[image]

[image]

[image]

[image]

jabrena, 2017-08-20:
The feature to read:

* Energy Monitoring – Analyze a device’s real-time and historical power consumption.

http://www.tp-link.com/us/products/details/cat-5516_HS110.html
https://www.youtube.com/watch?v=6VjXQs3D7B4
http://static.tp-link.com/res/down/doc/HS110(US)-1.0.pdf
https://github.com/computerlyrik/openhab2-addon-hs110/tree/master/src/main/java/org/openhab/binding/hs110

jabrena, 2017-09-05:
Today I tested the device (HS110) with the following library: https://github.com/intrbiz/hs110
Device: http://uk.tp-link.com/products/details/cat-5258_HS110.html

[image]

Well, I have to build something for multiple Bricks to Park and recharge:

[image]

closed 2020-08-17 by dlech