ev3dev/ev3dev #936: Ev3dev crashes during boot.

Darwin-II, 2017-08-15:
### What are you trying to do?
I have recently placed Ev3dev on a 16GB Kingston SD card. The problem is however that when I attempt to boot the Ev3 with the SD card it crashes. On the SD card the are the files:

.Spotlight-V100

.Trashes

_.Trashes

_boot.scr

_ev3dev-jessie-ev3-generic-2017-06-09.img

_ulmage

boot.scr

ev3dev-jessie-ev3-generic-2017-06-09.img

ulmage

I have a Linux machine running Debian 7. I have also tried on a Windows machine and a Mac. I would prefer it if all help is possible on Linux.

dlech, 2017-08-15:
It looks like you did not use a image writing utility to write the SD card image. We recommend using Etcher. See http://www.ev3dev.org/docs/getting-started/.

Darwin-II, 2017-08-29:
I use an ARM based computer and etcher is thus not compatible. Is there any ARM compatible software that I could Use?

mob41, 2017-08-29:
Is your ARM based computer running Linux? You can use a dd utility.

If you don't have it, you can install it by (in Debian based systems)

    sudo apt-get update
    sudo apt-get install dd

Plug in your SD card to the computer and type "dmesg" to find your drive location

Replace "ev3devImage" with the image file name and "/dev/sdX" with the drive location in the following command

    dd bs=4M if=ev3devImage.img of=/dev/sdX conv=fsync

Darwin-II, 2017-09-08:
I have tried to use the dd tool and yet it still crashes during boot time after about a second of attempting to boot.

dlech, 2017-09-08:
What is the *exact* command you used to flash the SD card?

Darwin-II, 2017-09-08:
I used the following:

`sudo dd bs=1M if=ev3dev-jessie-ev3-generic-2017-06-09.img of=/dev/sda
`

dlech, 2017-09-08:
I hope that `/dev/sda` is not your main hard drive!

With the SD card plugged into the computer, what is the output of the command `df -h`?

Darwin-II, 2017-09-08:
The command produces as follows:

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        29G  5.8G   22G  22% /
devtmpfs        426M     0  426M   0% /dev
tmpfs           430M   27M  404M   7% /dev/shm
tmpfs           430M   12M  419M   3% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           430M     0  430M   0% /sys/fs/cgroup
/dev/mmcblk0p6   68M   21M   47M  31% /boot
tmpfs            86M  8.0K   86M   1% /run/user/1000
/dev/mmcblk0p5   30M  1.6M   27M   6% /media/pi/SETTINGS
**/dev/sda1        48M  2.9M   45M   7% /media/pi/EV3DEV_BOOT**

dlech, 2017-09-08:
The boot partition looks right, so your `dd` command is correct. I would expect to see a `EV3DEV_ROOTFS` partition too though (on `/dev/sda2`).

Is the SD card at least 4GB or bigger?

Did you do something to safely eject the SD card after `dd` was complete (e.g. run the `sync` command)?

 Do you have another SD card you could try?

Darwin-II, 2017-09-09:
I have attempted to boot it now on another SD card. This time a Sandisk ultra 16GB but it yields the same response.

dlech, 2017-09-25:
> Did you do something to safely eject the SD card after dd was complete (e.g. run the sync command)?

Other than that, it sounds like you are doing everything right. I'm not sure why it wouldn't be booting for you.