ev3dev/ev3dev #937: micropython

dlech, 2017-08-18:
Are you tired of your python program taking 8 seconds to start?

Well... I have some great news for you! I've just uploaded a new package to ev3dev-stretch called `micropython`. [It is a super-trimmed down version of Python 3](https://micropython.org/) designed for embedded systems way less capable than the EV3. 

Want to try it out? Install the latest stretch snapshot, then...

```
sudo apt update
sudo apt install micropython
micropython
```

Or try this:

```
echo -e '#!/usr/bin/env micropython\n\nprint("Hello World!")' > hello.py
chmod +x hello.py
./hello.py
```

But, what is really interesting....

```
$ time ./hello.py 
Hello World!

real	0m0.082s
user	0m0.030s
sys	0m0.030s
```

That is a far cry from...

```
$ time python3 hello.py 
Hello World!

real	0m2.082s
user	0m1.680s
sys	0m0.100s
```

Now, as you may guess, `micropython` has some limitations compared to full `python3`, but from what I have seen, it seems to be quite fully featured. You can get the details [here](http://docs.micropython.org/en/latest/pyboard/genrst/index.html).

Try it out and share what you learn.

WasabiFan, 2017-08-18:
Nice! I tried this project out a bit in its infancy but I couldn't persuade it to run on the EV3 with my limited attention span :laughing: Have you done much testing with ev3dev-lang-python?

ddemidov, 2017-08-18:
I did not try this on the EV3, but here is an experiment on Arch linux:

```
$ cat microhi.py
import ev3dev.ev3 as ev3
m = ev3.LargeMotor()
print(m.connected)
```
```
$ python microhi.py
False
```
```
$ micropython microhi.py
Traceback (most recent call last):
  File "microhi.py", line 1, in <module>
  File "/home/demidov/work/ev3/ev3dev-lang-python/ev3dev/ev3.py", line 29, in <module>
  File "/home/demidov/work/ev3/ev3dev-lang-python/ev3dev/core.py", line 40, in <module>
ImportError: no module named 'os'
```

ddemidov, 2017-08-18:
https://pypi.python.org/pypi/micropython-os/0.6

ddemidov, 2017-08-18:
So `os` module seems to be available after installing `micropython-lib`, but now it can not find `mmap`, `ctypes`, and `subprocess.Popen`. I guess it is possible to come up with a cut-down version of `ev3dev-lang-python` that would not use these, but we will need a volunteer for that :).

WasabiFan, 2017-08-18:
I think it's worth the effort to implement workarounds which handle these differences; I'll probably procrastinate enough over the next few days to make room for this 😁

ddemidov, 2017-08-18:
With https://github.com/rhempel/ev3dev-lang-python/commit/692dc6ae6e2c59f9370ed7e4c969495b0ebc91de micropython is able to run the `microhi.py` test above.

dlech, 2017-08-18:
IMHO, I would rather see a new `micropython-ev3dev` library instead of breaking ev3dev-lang-python to try to make it work with micropython.

ddemidov, 2017-08-18:
I agree. I was just trying to see what needs to be changed/dropped to be able to run with micropython. Would be interesting to compare timings for the test script with the motor.

ddemidov, 2017-08-18:
```
robot@ev3dev:~$ sudo apt-get install micropython 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package micropython
```

Looks like its time to update to ev3dev-stretch. Not today though :(.

WasabiFan, 2017-08-18:
I think we should avoid fracturing the Python libraries at all costs. The great thing about Python is that it's an interpreted language -- if the interfaces aren't available at runtime, you can make the explicit decision to avoid them! We should do the same thing we already do for OS platforms and omit special functionality if it isn't supported. Any behavior like that needs documentation, though.

dlech, 2017-08-18:
I added a new package `micropython-lib` that includes all (well most) of the core python libraries.

dlech, 2017-08-19:
I've started a new library: https://github.com/ev3dev/micropython-ev3dev

It has basic motors and sensor operations working as well as LEDs and buttons.

WasabiFan, 2017-08-19:
@dlech Please don't publish that library. It might be useful as a test, but fracturing the Python ecosystem will only serve to confuse users -- _there is no reason to do this_. With twenty minutes of work the existing library will work with both Python runtimes, which is plainly the ideal case.

dlech, 2017-08-19:
Don't worry, I don't have any plans to. Right now, it is just something to play with.

dlech, 2017-09-23:
As nice as it would be to get ev3dev-lang-python running in both micropython and python3, I'm finding that is going to be a huge challenge. I just don't think it would be a practical use of everyone's time and would result in breaking changes and compromises to make it work. On the other hand, I am finding that implementing things in micropython easy, quick and fun. The fact is that micropython is, by design, a rather small subset of python when it comes to the standard libraries and there are a number of core micropython libraries that are used to replace some of that functionality, but these don't work on python3 (e.g. micropython/micropython-lib#18). This issue of micropython/cpython compatibility is something that has just not been solve yet in general and I don't feel inspired to be a pioneer in this area.

So, what I would like to propose is to continue work on a stand-alone micropython-ev3dev library. This will be a very minimal library (partly because that is just the nature of micropython). By keeping it to a minimum, it will also be a good entry point for beginners wanting to learn python or programming in general without being overwhelmed by too much information. It could also serve as a reference implementation for anyone wanting to add a new programming language to ev3dev.

With regards to ev3dev-lang-python compatibility, I like the suggestion from @ddemidov in https://github.com/rhempel/ev3dev-lang-python/issues/370#issuecomment-331653783 that there should be an `ev3dev.ev3g` module in ev3dev-lang-python. This module should be API compatible with the micropython-ev3dev library so that people can switch freely between the two implementations by just modifying the shebang and the import statement.

jabrena, 2017-09-23:
Hi,

I think that `micropython` has good speed features in compare to `python 3` but it doesn`t have a complete implementation of python.
https://github.com/micropython/micropython/wiki/Differences

If `ev3dev-lang-python` has compatibility for both interpreter is great, but python3 should be the first interpreter for the library. This a humble opinion.

Juan Antonio

laurensvalk, 2017-09-23:
Besides the startup time, another useful benchmark would be file i/o. I.e. how fast it can read a sensor valueX file or write a duty_cycle_sp file.

laurensvalk, 2017-09-23:
I'll have to do some more reading on this, but here is one idea for micropython-ev3dev. It looks like @dlech might already be doing this, though: 

Rather than duplicating ev3dev-lang-python, it would be useful to have a small library that essentially just simplifies accessing the hardware files. 

Documentation would not really be needed; it's the same as using the file system, just less inconvenient. Function calls would use the same names as the files (get_value0, set_command, .... ) as much as possible.

So, that would make it easy to use sensors by port name (A, B, ... ), set modes, read valueX files, set motor speed files, but not do much more.

That's perhaps only suited for more advanced users, but beginning users might be better off using the regular Python with ev3dev-lang-python anyway, I think.

dlech, 2017-09-23:
Actually, I am approaching it the other way around. I have found that libraries that expose the driver implementation details are very fragile and hard for newbies to understand. So, I'm using the API from the EV3-G software as a reference to provide an API that is familiar to EV3 users already. The micropython library will be advertised as for beginners and the full python library for advanced users.

laurensvalk, 2017-09-23:
OK, understood. I look forward to trying it out.

jabrena, 2017-09-23:
Hi @dlech,

in your opinion what is the main motivation to introduce `Micropython` in the project? 
I think that the performance is a good argument, but I think that EV3 is based on a Linux distro + Complete programming languages. In NXT for example (Based on a microcontroller) the idea to use `Micropython` could have more sense. With EV3, you get some seconds but maybe, the project introduce possible confusion for the final user with different behaviours or possibilities between `Python 3x` or `Micropython` in the middle term.

Imagine this scenario:
One student creates a program for Mindstorms Bricks and it uses OOP approach for example. For EV3, you get some seconds with `Micropython` but for Raspi3 it is not necessary. Executing the example discover some problem and the source of the problem is the different behaviour. I think that it is better to have common behaviour and avoid this kind of problems:

http://docs.micropython.org/en/latest/wipy/genrst/core_language.html#classes

Note: I try to offer a constructive comment, you know that I develop with `Java` :) but I try to add to add support in some global way.

Juan Antonio

dwalton76, 2017-09-23:
> So, what I would like to propose is to continue work on a stand-alone micropython-ev3dev library

Thoughts on having a micropython compatible base class in ev3dev-lang-python that the cpython compatible classes inherit from?  That would avoid splitting the libraries and would avoid some duplication of work.

> With regards to ev3dev-lang-python compatibility, I like the suggestion from @ddemidov in rhempel/ev3dev-lang-python#370 (comment) that there should be an ev3dev.ev3g module in ev3dev-lang-python

Of the EV3-G APIs that I've worked on so far this week, our Motors class as of a week ago makes a clean base for a more enhanced EV3-G child Motor class with all of the `on_for_XYZ` calls.  Many of the others are not as clear though
- TouchSensor - what we had was pretty comparable to EV3-G...I added a couple of things so it is now a superset of EV3-G.  I doubt we would need a TouchSensor class in an ev3g.py
- ColorSensor - ours was pretty comparable with the exception that we don't support calibrate.  I doubt we would need a ColorSensor class in an ev3g.py unless we add support for calibrate.
- Infraredsensor - I feel like what we have is cleaner than the EV3-G blocks, we have InfraredSensor, BeaconSeeker and RemoteControl.  EV3-G has those combined all in one block.  So our ev3g.py would have a InfraredSensor class that under the covers uses the three classes we have today.

I'm not saying I'm against it just that it might look a little strange to a new user. I could picture someone saying "why didn't they just put all of this in core.py and keep the API in one place". My 2c

jabrena, 2017-09-23:
In my opinion, the problem is not the library or the EV3 Brick, the problem is the Language support. If the project admits 2 different interpreters, the user will have to learn the "tricks" about both languages.

In terms of performance is very nice to run everything quick but you are opening the door to learn one language with different behaviours.

My external comment is: continue with normal `Python` and offer support for `Micropython` but not loose the focus on the library or the new API design. In the past, the project jump from Python 2x to 3x but now the possible problem is the introduction of a dialect and it could be worse.

**Note:** If LEGO could release a EV3 1.5 with better technical specs, the problem could be solved. :)

Good night.

Juan Antonio

WasabiFan, 2017-09-24:
The discussion here is quite complicated and responses equally nuanced, so I want to make sure we are clear and thorough when conversing. The most important part of this post is below the fold, but I wanted to make sure I addressed others' thoughts so we can try to reach a consensus.

---

@dlech:

> As nice as it would be to get ev3dev-lang-python running in both micropython and python3, I'm finding that is going to be a huge challenge. I just don't think it would be a practical use of everyone's time and would result in breaking changes and compromises to make it work

Could you provide some examples here? As of now, I'm using motors and sensors with only minor changes locally.

> So, what I would like to propose is to continue work on a stand-alone micropython-ev3dev library. This will be a very minimal library (partly because that is just the nature of micropython).

I'm not seeing what would make this library _minimal_ compared to ev3dev-lang-python. We've added auxiliary classes in the past year or two, but motor and sensor interfaces are solely file accessors with some performance improvements and methods to set them pleasantly. What is it that would be omitted or different? As of now, the `micropython-ev3dev` implementation is the same as ev3dev-lang-python from a "minimal implementation" standpoint.

> By keeping it to a minimum, it will also be a good entry point for beginners wanting to learn python or programming in general without being overwhelmed by too much information.

I belive we should design for our single API model to be comprehendible and friendly rather than singling out a demographic. As I see it, that "friendly" interface is the one we have begun with the EV3-G methods.

The distinction between a "beginner" and an "advanced user" is arbitrary; if we tell "beginners" to start using a particular SKU of our API, they will always use that SKU, even once they graduate to the "advanced" label. There's no forcing function to make people transition, which signals a fundamental analytical or design error. One API isn't an extension of the other, and neither carries with it an advantage nor warranted change in complexity. The only difference between the two API surfaces is in naming and unit choice; and frankly, those changes should be adopted library-wide regardless of who we expect to be using them.

> With regards to ev3dev-lang-python compatibility, I like the suggestion from @ddemidov in rhempel/ev3dev-lang-python#370 (comment) that there should be an ev3dev.ev3g module in ev3dev-lang-python.

_Why?_ What is the value added from one API set to the other? That one was better-designed? That seems like a call for an intentional decision to deprecate an old API space and adopt one we are willing to support. To borrow a metaphor from [one of my favorite essays on programming](http://www.stilldrinking.org/programming-sucks): leaning a Picasso over the melting snowflakes won't hide the cat urine. And, frankly, no-one wants to deal with the resultant puddle.

> This module should be API compatible with the micropython-ev3dev library so that people can switch freely between the two implementations by just modifying the shebang and the import statement.

I fully and wholeheartedly support this.

---

I've said a lot about the existing proposals, but this would be meaningless without a suggested solution. **I would like to propose the following:**

- We declare a core area of APIs which we support on all Python platforms. This is the motor and sensor interface at a minimum, following the EV3-G design, other options pending.
- We produce an implementation of that core which works on both CPython and MicroPython. This could be the existing ev3dev-lang-python implementation, an adaptation of David's micropython library, or a new/modified version.
- We build platform-specific additions to that base. Our fancy Pillow implementation of the screen would be a CPython-only interface, and we might produce interfaces that take advantage of the low-level APIs that Micropython has for that platform.
- We ship the "common" base with the platform-specific additions for their respective Python implementations. Each is its own package but code using motors and sensors is entirely portable.

This has the advantages that:
- There is no risk of the APIs diverging, and hence no potential for confusing users with differing SKUs of the same stuff.
- We can fully take advantage of the unique abilities of each platform.
- It gives a discovery path so someone can evaluate the choice between the two platforms. Each class in out documentation would have a "platform" note, telling you which platform it was supported on. If you find an API that you want to use but isn't supported on your platform, you can make an educated decision to switch.

In short, it has all the benefits of creating an entirely separate module except we don't run the risk of confusing or frustrating people.

The remaining questions for that option are:
- How do we handle/support the existing API? I think its design is a bit suboptimal, so IMHO I think we should deprecate it. Certainly we should keep it around for some time regardless for compatibility.
- How does packaging work? I haven't spend any time with the packaging structure of this library.

For reference, my goals with this proposal are:
- Choose a single, agreed-upon API surface which we want to support until the end of time
- Prevent overlapping APIs or implementations to limit the confusion we cause
- Take the opportunity to make any necessary breaking changes now, for Stretch, and then commit to the product from here forward.

ddemidov, 2017-09-24:
> Why? What is the value added from one API set to the other? That one was better-designed? 

The thing I don't like about the new API is that it is unnecessary restrictive. For example, there is
```py
def on_for_seconds(self, speed_pct, seconds, brake=True, block=True):
```
First, the functionality may be replicated in the old API (with some extra properties)
```py
def run_timed(self, **kwargs):
```
with `run_timed(seconds_sp=S, speed_pct_sp=P, stop_action='brake')`. The blocking call would need to be implemented as a separate call to `wait_while('running')` or, even better, we can add an extra parameter to `run_timed`. And second, there are things that you can do with the old API that are impossible with the new one. An example is `run_timed(seconds_sp=3, ramp_up_sp=300, ramp_down_sp=500)`.
(Also, I don't like the name `on_for_seconds`: `on_for_time` would be better, which could work as `on_for_time(minutes_sp=1)` or `on_for_time(milliseconds_sp=100)`).

In other words, I see replacing old API with new one as destructive. That is why I think moving the new API into its own module is a better idea. Even better idea would be to compromise and have the best of both worlds in a single API. As in
```py
def run_timed(block=True, **kwargs):
```

dlech, 2017-09-24:
I think we are getting way ahead of our selves here. Let's take a moment to step back and ask ourselves what exactly are we trying to accomplish here (with regards to micropython).

I'll go first...

1. I want to have fun. For me, the VSCode extension + the fast startup time of micropython has made the difference between tedious and fun when it comes to writing programs that run on an EV3.
2. I would like to verify that micropython is a viable runtime on the EV3. (It seems very promising, but it is not a super-mature project, so there could be some surprises we haven't discovered yet)
3. I would like to try an experiment and create a library that makes it trivial to translate an EV3-G program (#668) and see what happens.
4. I don't want the goals above to break or otherwise make a mess out of ev3dev-lang-python.

---

And to answer @WasabiFan's questions...

> Could you provide some examples here? As of now, I'm using motors and sensors with only minor changes locally.

As you have seen, the screen/framebuffer is really the main blocker and I was looking at this from an all or nothing point of view. (The fact that motors and sensors work is because we designed them to with *any* programming language - good job, us.)

> What is it that would be omitted or different?

Omitted would be anything public that is a driver implementation detail, like methods/properties that correspond to sysfs attributes. Also omitted would be extras like the gyro balancer class.

> Why? What is the value added from one API set to the other? That one was better-designed?

From my point of view, the API's have clear separate goals. This is why we need to take a step back and say what exactly are we trying to accomplish here.

WasabiFan, 2017-09-25:
**The most crucial part of this post is at the end!**

> Let's take a moment to step back and ask ourselves what exactly are we trying to accomplish here (with regards to micropython).

I agree with all the goals you laid out. In addition:
- While we are considering this new API area, I am also looking at our existing APIs. I'll explain that in a bit.
- I want to make sure that we support the entire community of Python users well. That means that we should have high-quality documentation for any/all Python platforms, as well as put energy into software design for existing users.

---

> As you have seen, the screen/framebuffer is really the main blocker and I was looking at this from an all or nothing point of view. (The fact that motors and sensors work is because we designed them to with any programming language - good job, us.)

I agree that the framebuffer won't be a good candidate for cross-platform APIs. In line with my proposal above, I think we should implement the screen APIs on a per-platform basis, although we may want to consider a functionality that we support on both to make super simple things work. The vast majority of the times I've used the display are just showing text and circles/rectangles/lines.

> Omitted would be anything public that is a driver implementation detail, like methods/properties that correspond to sysfs attributes. Also omitted would be extras like the gyro balancer class. Different would be that it is modeled after EV3-G rather than ev3dev drivers.

As I mentioned in our call, I think we should demote the internal sysfs API details to either a nested object or make them "python private" (leading underscore). If we produce a well-thought-out API, there will be no reason for accessing those properties outside of exceptional circumstances. That doesn't mean they should never be used, of course, which is why I think they should be available.

As for the gyro balancer, I think it would make sense to separate those "helpers" into their own package or -- following the model I described above -- simply omit them from the micropython version. This isn't a core library feature as I see it.

> Different would be that it is modeled after EV3-G rather than ev3dev drivers.

Why should we believe that Micropython users inherently want an EV3-G API, and CPython users inherently want a different style of interface? As I said in my previous post and in our call, I think this would be baseless. It feels to me like "Micropython" and "EV3-G" are being (in)advertently conflated. As I'll explain below, I don't view EV3-G APIs as a compatibility layer, but instead as a welcome usability improvement for everyone. As you said, it makes writing ev3dev Python programs much more enjoyable.

> From my point of view, the API's have clear separate goals. This is why we need to take a step back and say what exactly are we trying to accomplish here.

This is where I'll go into detail on my broader vision. Our current APIs can be categorized into two (and a half) buckets:
1. A set of properties which supply direct access to the ev3dev drivers.
2. A set of functions which supply direct access to the ev3dev drivers _and then_ set a command.
3. Some getters to access particular sensor values.

The first is clearly not a friendly interface for most programming.

The second is better, but:
- Provides no opportunity for autocomplete
- Provides no "hint" as to what properties you need just by looking at the signature
- Leaves you exposed to accidental bugs because a property retains a value in a way that you didn't expect it

The third is only applicable to sensors, but I think it's quite a nice experience. I do think there's some confusion around the use of the `value` method (we don't make it clear that these properties are the ones you want to use, rather than `value`), but I digress.

This gap in buckets one and two, where there isn't a clear way to perform common tasks, is where I think the EV3-G APIs come in. **These new EV3-G-style APIs aren't just a compatibility layer, they fill a crucial gap in our library's interfaces.** I was already planning to take a critical look at our design to implement some sort of an "easy" API, but this concept of supporting the interfaces as LEGO does fills the use case quite nicely. I have no interest in reinventing the wheel when there's already a paradigm applicable to us which our users understand. 

Additionally, after a bit more thought, I _don't_ think we should deprecate the APIs for bucket two. Instead, I think we should consider them "intermediate-level" and rethink their naming to make them fit better into this world of easy-to-use interfaces.

dwalton76, 2017-09-25:
My biggest concern is that we end up with two ev3dev-python libraries, one for cpython and one for micropython, I would really really really like to avoid that if we can.  Re-posting this question since I think it got lost in the chatter above.

> Thoughts on having a micropython compatible base class in ev3dev-lang-python that the cpython compatible classes inherit from? That would avoid splitting the libraries and would avoid some duplication of work.

So in a nutshell
- code that works with micropython is moved from `core.py` to `ev3dev/coremicro.py`
- code that doesn't work with micropython remains in `core.py`. For classes where some of the functionality works in micropython and some doesn't, there would be a base class in coremicro.py and a child class in core.py that inherits from the coremicro.py base class.
- core.py imports * from coremicro.py
- if the user has a foo.py robot that they want to use with micropython they import coremicro.py
- if the user has a bar.py robot that they want to use with cpython they import core.py (no change from today)

I think all of the motor and sensor classes could be moved to coremicro.py, that would allow a lot of robots to run via micropython without having two repos and duplicating tons of code.

ddemidov, 2017-09-25:
> code that works with micropython is moved from core.py to ev3dev/coremicro.py

This sounds like a nice plan, only I think the only functionality that is not supported out of the box by micropython is screen interaction. So may be its better to keep core.py as is, but move screen to its own module. This has the added bonus that users who are not planning to use screen will not import PIL module and save some initialization time.

We will also need to either have different implementations of `Sound` for cpython and micropython or, better, implement is in a portable way.

WasabiFan, 2017-09-25:
I agree with @dwalton76 on all points. However, I'd like to suggest a slightly different structure. Navigating our code base is quite tough right now because everything is in a monolithic "core" file. I think we should separate it into files for motors, sensors, screen, sound, etc. That way, we could have a structure that looks something like this (just an example):

- `motors.py`
- `sensors.py`
- `cpython/`
  - `advanced_screen.py`
- `micropython/`
  - `simple_screen.py`

We'd only include the relevant files when packaging for each platform. That's just one example of the file structure.

In doing it this way, we would require that all the root-level files are compatible with both platforms.

ddemidov, 2017-09-25:
How about `micro/screen.py` and `macro/screen.py`? :)

dwalton76, 2017-09-25:
yep core.py is pretty big, I think it would be good to break it up.

dlech, 2017-09-25:
> Thoughts on having a micropython compatible base class in ev3dev-lang-python that the cpython compatible classes inherit from? That would avoid splitting the libraries and would avoid some duplication of work.

This question didn't get lost. It is part of getting ahead of ourselves. We don't even know that micropython is a stable platform to build on yet. And based on my experience so far, I am starting to finding some of those "surprises" I mentioned, like [PEP 475](https://www.python.org/dev/peps/pep-0475/) is not implemented and getting random segfaults when using threads.

So, before we go messing too much with ev3dev-lang-python trying to get it to work with micropython (or even just having lengthy discussions about it), I think we need to spend some time figuring out if micropython is going to work at all.

As such, I am still in inclined to keep working on the current micropython-ev3dev project I have been playing around with to see if we can solve these problems easily. I understand the concerns about code duplication, but in this case, it will be much easier to throw out micropython-ev3dev entirely if it doesn't work out (and we are only talking about ~1000 lines of duplication, so it's really not that much).

And, since micropython-ev3dev is going to be totally experimental, I think it is a good opportunity to implement #668 in it's most literal form. I don't think that such a literal translation would be good for a general purpose library, but I do think it has the practical benefit of giving us a definite design to work towards. More specifically, discussions about what we should name a thing and what we should include or exclude are kept to a minimum because we have a reference design (EV3-G) that we are working against. And it will force us to try out the full range of EV3 capabilities instead of just limiting ourselves to the things we are interested it.

If this experiment works out and we find that micropython is everything we hoped for, then we can start discussing how to make ev3dev-lang-python work with micropython and how to take the parts of the experiment we like and merge them into a common code base. But until then, I think we should focus on getting to know the quirks of microython and working with upstream to fix these bugs to make sure we have something that is going to be viable in the long term.

dlech, 2017-09-25:
Another thing worth mentioning is that upstream micropython is very resistant to adding anything new. No, they are not grumpy. They just have the goal to keep the **micro** in micropython as small as possible. So you have to have a very good reason for any changes that you suggest.

WasabiFan, 2017-09-25:
> So, before we go messing too much with ev3dev-lang-python trying to get it to work with micropython (or even just having lengthy discussions about it), I think we need to spend some time figuring out if micropython is going to work at all.

Separating out the classes into separate files and cleaning up other related API structures is something I expect to do regardless of whether or not we support Micropython. That said, including support for Micropython on top is solely a matter of packaging configuration. With such a miniscule investment required, I see no reason that we would want or need an entirely separate implementation from the ground up.

> As such, I am still in inclined to keep working on the current micropython-ev3dev project I have been playing around with to see if we can solve these problems easily. I understand the concerns [...]

It's even easier to _not throw anything out_. This micropython-ev3dev library is just a re-implementation of what we've already done with ev3dev-lang-python. Implementing it separately only splits work and breaks down the clear path forward for users.

If this library _really is_ only a testbed for us to experiment with the platform and interfaces, it should be called `micropython-test` and should have a README which solely tells people to visit `ev3dev-lang-python`, eliminating confusion.

> And, since micropython-ev3dev is going to be totally experimental, I think it is a good opportunity to implement #668 in it's most literal form. 

What is the use case for a utility that ingests an EV3-G program and spits out a Python program which is incompatible with the ev3dev Python library? If the goal with the utility is to act as a runtime for users' programs, we have lms2012-compat already to cover the use case. If the goal is to give people a starting point to learn ev3dev, generating a program which doesn't mimic the methodology we suggest for normal users learning our APIs is pointless. I simply don't see a relation here between a translation utility and this micropython library; they're two entirely independent things which serve different goals.

> I don't think that such a literal translation would be good for a general purpose library, but I do think it has the practical benefit of giving us a definite design to work towards. More [...]

Are you saying that writing this micropython library is good because it is an initial implementation that can be learned from? How is that different from the work we're doing in ev3dev-lang-python preparing for an initial Stretch release down the road? And again, this has nothing to do with micropython.

dlech, 2017-09-26:
OK, forget it. This is too much like work. I just wanted to have a little fun without having to have a good reason for it or thinking about it too much.

closed 2017-09-26 by dlech