ev3dev/ev3dev #938: Wiring, Adafruit 1,8" display in EV3 Brick

oknystad, 2017-08-18:
Hey there, this is my very first message on GitHub (though I've been following this repository for a while).

I've got ahold of an Adafruit 1,8" TFT display (ST7735) + 0,5 mm pitch flat flex cable + FPC breakout board, and would like to attempt to fit it into the EV3, as shown in @rhempel 's video: 
[https://www.youtube.com/watch?v=gPNJC5Uz9HY ](url)

However, I can't find information about the exact wiring between the pins in the display socket (1-16) of the EV3 and the pins on the Adafruit display (LITE, MISO, SCK, MOSI, TFT_CS, CARD_CS, D/C, RESET, VCC, Gnd).

Does anyone know a working wiring setup? 

EV3 kernel version: 4.4.78-21-ev3dev-ev3

dlech, 2017-08-18:
I need to do a blog about this. I get asked every once in a while. Stay tuned...

dlech, 2017-08-18:
Here is the short version...

```
LEGO FPC          |     AdaFruit
------------------+-------------------------
CSDIS     6       |   1 BackLight
N/C               |   2 MISO
SCKDISR   2       |   3 SCK
MOSIDISR  3       |   4 MOSI
GND      10,11    |   5 TFT_CS
N/C               |   6 CARD_CS
MISODIS   4       |   7 D/C
RSTDIS    5       |   8 RESET
VCC3V3    8,12    |   9 VCC
GND      10,11    |  10 GND 
```

oknystad, 2017-08-19:
Thanks for the info!
I've tried to solder the pieces together accordingly, but no luck, got no reaction in the display on startup. (Previously, the display worked fine when connected to an Arduino). Perhaps something got messed up due to too much soldering/de-soldering.. Anyway, when checking for 0 ohms with a voltmeter, it seems to be proper contact between pins on the end of the flat flex cable and the corresponding soldering points on the AdaFruit.

Do I understand it correctly, that pins 10+11 (Lego FPC) and 5+10 (AdaFruit) are all wired directly together, and the same with pins 8+12 (Lego FPC) and 9 (AdaFruit)?

dlech, 2017-08-19:
Sorry to hear it is not working. Check for solder bridges. (And that the FPC is not backwards.)

And yes, it sounds like you are understanding the pinout correctly.

oknystad, 2017-08-20:
Yes, I had the Lego FPC pins backwards.. (found 3,3 V from VCC and GND on the opposite side of the flex from where I expected.) After reversing the wiring on the FPC breakout board, the screen now is bright white upon startup, but the buttons become unresponsive, the EV3 lamps are off, and the EV3 is unreachable via ssh. I need to re-insert the battery to make the buttons responsive again. 
I found no solder bridges upon inspection. Could it be driver related?

dlech, 2017-08-20:
The LEDs should turn red when you power on the EV3. This comes from an EEPROM in the EV3. If this is not happening, then you must have a short or something that is preventing the EV3 from powering up correctly.

oknystad, 2017-09-03:
I started from scratch with a new 1,8" display, but still can't get it to work properly.
At startup, the LEDs are red for about 4 seconds before they go dark and the booting stops, while the display remains white. The same happened on my last attempt with another display card. 
If I remove the 1,8" display again and restart the EV3, it boots completely and I can access it via SSH.

Furthermore, if I insert the flex cable into the EV3 sometime _during_ booting, it boots up completely (though the 1,8" display is still white, flickering somewhat, for example on button presses).
Trying a new SD card with ev3dev-jessie-ev3-generic-2017-06-09, doesn't make any difference.
If I remove the SD card and start the EV3 with 1,8" display attached, it boots up normally into the original LEGO program and gives the sound signal when it's ready for use. The display doesn't show anything useful (only white), as expected.
Continuity testing indicates no bridges, and otherwise OK connections. Wiring:
```
From AdaFruit | To flex pin | Cable color
--------------+-------------+---------------
LITE          | 6           | Brown
SCK           | 2           | Orange
MOSI          | 3           | White
TFT_CS + Gnd  | 10 + 11     | Black
D/C           | 4           | Green
RESET         | 5           | Blue
VCC           | 8 + 12      | Red
```
My test setup (on the upper picture, 25 on the FPC breakout board corresponds to pin 1, 26 to pin 2 and so on):
[image]
[image]

dlech, 2017-09-03:
I pulled out my display to see what is going on. It looks like we have a kernel crash that is causing the EV3 to no boot up. I'll see what I can do about this and in the mean time, you can try an older ev3dev image. Maybe from last year some time. https://github.com/ev3dev/ev3dev/releases Just don't upgrade the kernel. 

```
[<c02ce5b0>] (davinci_spi_chipselect) from [<c02cd1b8>] (spi_bitbang_set_cs+0x58
/0x74)
 r6:00000001 r5:c061ab78 r4:c39f7400 r3:c02ce5b0
[<c02cd160>] (spi_bitbang_set_cs) from [<c02cbb0c>] (spi_set_cs+0x64/0x6c)
 r8:00000000 r7:c0605768 r6:c39d3000 r5:00000000 r4:00000000 r3:c02cd160
[<c02cbaa8>] (spi_set_cs) from [<c02cbc1c>] (spi_setup+0x108/0x12c)
 r4:c39f7400 r3:fef0e000
[<c02cbb14>] (spi_setup) from [<c02cc530>] (spi_add_device+0xb0/0x130)
 r6:c39d3000 r5:00000000 r4:c39f7400 r3:00000000
[<c02cc480>] (spi_add_device) from [<c02cc6c8>] (spi_new_device+0x8c/0xb4)
 r7:00000000 r6:c39d3000 r5:00000000 r4:c39f7400
[<c02cc63c>] (spi_new_device) from [<c02cc71c>] (spi_match_master_to_boardinfo+0
x2c/0x4c)
 r5:c39d3000 r4:c3804fc8
[<c02cc6f0>] (spi_match_master_to_boardinfo) from [<c02cca84>] (spi_register_master+0x348/0x6d4)
 r5:c3804fc0 r4:c39d3000
[<c02cc73c>] (spi_register_master) from [<c02cd2cc>] (spi_bitbang_start+0xe0/0x134)
 r10:00000000 r8:c0605768 r7:c0605768 r6:c39d3000 r5:c39d3278 r4:c39d3000
[<c02cd1ec>] (spi_bitbang_start) from [<c02cdc14>] (davinci_spi_probe+0x3f4/0x4f0)
 r5:00000000 r4:c39d3278
[<c02cd820>] (davinci_spi_probe) from [<c028bedc>] (platform_drv_probe+0x44/0x8c)
 r10:00000000 r8:00000000 r7:c0621d20 r6:c0621d34 r5:c0605768 r4:c02cd820
[<c028be98>] (platform_drv_probe) from [<c028a618>] (driver_probe_device+0x110/0x298)
 r7:c0621d34 r6:c064b0b4 r5:c060579c r4:c0605768
[<c028a508>] (driver_probe_device) from [<c028a808>] (__driver_attach+0x68/0x8c)
 r8:00000000 r7:00000000 r6:c0621d34 r5:c060579c r4:c0605768
[<c028a7a0>] (__driver_attach) from [<c0288e1c>] (bus_for_each_dev+0x58/0x90)
 r6:00000000 r5:c028a7a0 r4:c0621d34 r3:00000000
[<c0288dc4>] (bus_for_each_dev) from [<c028a0f4>] (driver_attach+0x20/0x28)
 r6:c061f100 r5:c39fa0c0 r4:c0621d34
[<c028a0d4>] (driver_attach) from [<c0289d2c>] (bus_add_driver+0xd4/0x1f0)
[<c0289c58>] (bus_add_driver) from [<c028aef8>] (driver_register+0xa4/0xe8)
 r7:c05e3938 r6:c05ee82c r5:c05ffbf8 r4:c0621d34
[<c028ae54>] (driver_register) from [<c028bc78>] (__platform_driver_register+0x38/0x4c)
 r5:c05ffbf8 r4:c39f81e0
[<c028bc40>] (__platform_driver_register) from [<c05e3950>] (davinci_spi_driver_init+0x18/0x20)
[<c05e3938>] (davinci_spi_driver_init) from [<c0009670>] (do_one_initcall+0x118/0x1dc)
[<c0009558>] (do_one_initcall) from [<c05cee00>] (kernel_init_freeable+0x100/0x1cc)
 r8:c05fb9d4 r7:000000ae r6:c05ee82c r5:c05ee84c r4:00000006
[<c05ced00>] (kernel_init_freeable) from [<c047b6a8>] (kernel_init+0x10/0xec)
 r8:00000000 r7:00000000 r6:00000000 r5:c047b698 r4:00000000
[<c047b698>] (kernel_init) from [<c000a310>] (ret_from_fork+0x14/0x24)
 r4:00000000 r3:c3838000
Code: e5d02140 e5900138 e590504c e593014c (e5d04000)
---[ end trace ef491c95ece3bde7 ]---
```

oknystad, 2017-09-03:
Thank you for looking into this, @dlech. Brickman does indeed show up on my 1,8" display with an older image :smiley:. Tested images:
```
ev3-ev3dev-jessie-2015-12-30	          1,8" works fine
ev3dev-jessie-ev3-generic-2016-10-17	  EV3 fails to boot
ev3dev-jessie-ev3-generic-2016-12-21      EV3 fails to boot
```
So I'll just wait for the next kernel update and see if the latest fix does the trick.

dlech, 2017-09-09:
Please try the 2017-09-09 ev3dev-jessie snapshot.

https://oss.jfrog.org/artifactory/list/oss-snapshot-local/org/ev3dev/brickstrap/

oknystad, 2017-09-17:
I finally had the time to test the newest release now.

Test of `snapshot-ev3dev-jessie-ev3-generic-2017-09-14`:
Boots normally, uses ca 1 min to reach green lights. Brickman shows up, but the orientation is now upside down compared to `ev3-ev3dev-jessie-2015-12-30`. This surely was done in ev3dev/ev3dev#673.
I could try to turn my display around, but it wouldn't be a very elegant solution at this point. Is there any other (simple) way to change the rotation setting?

Test of `snapshot-ev3dev-stretch-ev3-generic-2017-09-14`:
Seems to boot completely, uses ca 3 mins to reach green lights. But the display is all white, blinking black during bootup, stays white when finished. (Same result with two SD cards that otherwise work fine - and display shows brickman normally again when returning to the jessie image, so it's not a hardware problem.)

dlech, 2017-09-17:
> Is there any other (simple) way to change the rotation setting?

The only way to do this is to compile your own kernel.

> Test of snapshot-ev3dev-stretch-ev3-generic-2017-09-14

The color display is not working in ev3dev-stretch yet. I haven't figured out what is the best way to handle it yet. Booting works a bit different with the newer kernel because we are using device tree.

oknystad, 2017-09-18:
Okay, thanks.
I'll tinker some more with the display, but it clearly works as expected, so I'm closing this issue.
I haven't compiled kernels before, but would be interesting in trying. Any tips/pointers to get started with the basics (using Windows 10) would be welcome. :wink:

closed 2017-09-18 by oknystad

dlech, 2017-09-18:
Have a look at <https://github.com/ev3dev/ev3dev-buildscripts/tree/ev3dev-jessie>.

You will have to install Ubuntu in a virtual machine if you only have Windows.

WasabiFan, 2017-09-19:
> You will have to install Ubuntu in a virtual machine if you only have Windows.

In the past, I've used the Windows Subsystem for Linux (built-in to recent-ish Windows 10 versions) to try building the kernel. I just tested it out (it's been a few months) and it seems to still work, at least to the point of compiling and linking.

_(image creation failed with ""mkimage" command not found - U-Boot images will not be built", I can open an issue for that if you'd like.)_

oknystad, 2017-09-19:
> (image creation failed with ""mkimage" command not found - U-Boot images will not be built", I can open an issue for that if you'd like.)

@WasabiFan: Please do, sounds like a useful thing to resolve. I was unaware of the Windows Subsystem for Linux (!) - will try it out.

dlech, 2017-09-19:
`mkimage` froms from the `u-boot-tools`, so you probably just missed installing that package.

oknystad, 2017-10-02:
Building custom kernel was solved in #969.

dkapur17, 2023-05-29:
Hi. I'm trying this mod because the ribbon connector for my stock display has broken. I'm trying to get the display off the carriage, but it seems to be stuck with some kind of adhesive and taking it off would break the screen entirely. Is this expected or am I doing something wrong?

dlech, 2023-05-29:
On the stock display, the ribbon cable is permanently connected. You just disconnect the cable from the main circuit board of the EV3.

dkapur17, 2023-05-30:
@dlech sorry for the unclear wording. By carriage, I meant the black plastic holder (on the right in the image from Ralph's video).

[image]

It seems to me that the stock display is glued onto that holder, and to host the color LCD onto that holder, I would have to somehow detach the stock display from it first.

dlech, 2023-05-30:
Ah yes, I did break one screen trying to pry it off. Heat would probably help loosen up the adhesive.

I didn't want to cut up the black part on mine, so I just taped the new display to the outside case of the EV3 and left the black part out. Probably not the best if the EV3 moves around a lot though.

If you are into 3D printing, you could print a replacement black part that actually fits the new display.

dkapur17, 2023-06-03:
Hi @dlech. I got the required parts for this hack. Unfortunately, due to availability at my location, I couldn't get the Adafruit display and instead had to settle for the [Fermion 1.8" 128x160 IPS TFT LCD](https://www.dfrobot.com/product-2580.html), which for all intents and purposes seems a suitable replacement for the Adafruit display. There is a difference in the pin numbering for the display, which is shown below (from the [schematics](https://dfimg.dfrobot.com/nobody/wiki/240100b3cb38cdd21f36792a94d03843.PDF)).

[image]

Based on this, and your wiring for the Adafruit display, I came up with the required wiring, which is as follows:

| Fermion Display | Lego FPC        |
|-----------------|-----------------|
| 1 VCC           | VCC3V3    8,12  |
| 2 GND           | GND       10,11 |
| 3 M_SCK         | SCKDISR   2     |
| 4 M_MOSI        | MOSIDISR  3     |
| 5 M_MISO        | N/C             |
| 6 M_CS_LCD      | GND       10,11 |
| 7 M_RESET       | RSTDIS    5     |
| 8 M_D/C         | MISODIS   4     |
| 9 BLK           | CSDIS     6     |
| 10 M_CS_SD      | N/C             |

I don't have much of an electronics background so this is an educated guess at best.

Next, I used the `ev3dev-stretch-ev3-generic-2020-04-10` release, and uncommented the line that says `fdtfile=da850-lego-ev3-adafruit18.dtb` in the `uEnv.txt`. I also tried `fdtfile=da850-lego-ev3-color.dtb`. In all cases the EV3 boots up, but the display remains white. 

[image]

I can't seem to figure out which step it is that I am messing up at (hoping the physical wire connections I've made are correct🤞). Any help here would be appreciated. TIA!

dlech, 2023-06-03:
The wiring diagram look right. Do you have a logic analyzer? That is really the only way to see what is actually going on.

dkapur17, 2023-06-04:
I don't. But I think I can probably create a makeshift one using an ESP32? What kind of analysis do you want me to run?

dkapur17, 2023-06-05:
@dlech going through the product descriptions I found that the Adafruit display uses the ST7735R driver chip while the Fermion display uses the ST7735S driver chip. Could that possibly causing some issues?

dlech, 2023-06-05:
Glad to hear you got it working.

> I'm assuming for the EV3 this would require a custom kernel with the required offsets

Yes, this would require a custom kernel. I think you would need to add CASET and RASET commands to the init sequence [here](https://github.com/ev3dev/ev3-kernel/blob/b8e704fe9ed45fcc34a6e5658078bfc5a923fb8d/drivers/gpu/drm/tinydrm/st7735r.c#L38).

BobGaemes, 2024-06-23:
I know that this issue was like 6 years ago but I also broke my ev3 lcd and I have this [lcd ](https://www.amazon.se/-/en/dp/B07FYCKYP1/)from Amazon (EU only?). So I tried using it and I think that I made the correct wiring but this lcd does not have a RST pin. Do the ev3 need this pin? I also have an Adafruit 1.8" display on the way so I will test if that works too.

LeFFaQ, 2026-10-03:
There is any way to wire up this display with original firmware? Or maybe step to do it. One of my bricks in the classroom has died, and I’d like to restore it. We work with both the original firmware and EV3dev.

dlech, 2026-10-03:
The original firmware would need to be modified to add a driver for the different display. The wiring, initialization and pixel format are all different, so the original firmware can't be used as-is.