ev3dev/ev3dev #939: Empty Lego-Ports, correct config.txt:q

Triarier, 2017-08-22:
### What are you trying to do?
Lego-Ports is empty, even though the lines:
dtoverlay=brickpi3
dtoverlay=pi3-disable-bt

### What did you expect to happen?
Lego-Ports should have symlinks to the actual 8 ports if I am correct. 

### What actually happened?
It is empty. 

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2017-08-11
Kernel version:     4.9.41-ev3dev-1.4.0-rpi2
/usr/bin/ev3dev-sysinfo: line 68: warning: command substitution: ignored null byte in input
Board:              Raspberry Pi 3 Model B Rev 1.2
Revision:           a02082
Brickman:           0.9.0
ev3devKit:          0.5.1

dlech, 2017-08-22:
Check the output of `dmesg` for messages related to `brickpi3`.

Triarier, 2017-08-22:
Good answer. 
Firmware outdated. I guess this will fix it in the morning. 

Thank you very muchl

closed 2017-08-22 by Triarier