ev3dev/ev3dev #941: Is it possible to use the built-in speakers with an iPhone?

marlonalkan, 2017-08-26:
### What are you trying to do?

I'm trying to use the built-in speakers with an iPhone.

dlech, 2017-08-26:
Can you give some more details on what exactly you are trying to do? Do you want to play audio from the iPhone on the speaker in the EV3?

marlonalkan, 2017-08-26:
Yes, exactly I want to play music from my iPhone on my EV3 using the built-in speakers of the EV3.

dlech, 2017-08-26:
There is some information on how to do this [here](https://github.com/ev3dev/ev3dev/issues/198#issuecomment-62339000) and maybe [here](https://github.com/ev3dev/ev3dev.github.io/pull/24/files).

marlonalkan, 2017-08-27:
I'll try it. 

	
Marlon Alkan
about.me/marlonalkan

marlonalkan, 2017-08-27:
Nope it isn't working. I can connect via Bluetooth but I can't Play  Music on the Ev3

closed 2020-08-17 by dlech