ev3dev/ev3dev #942: Unable to connect via usb with network manager in Ubuntu 17.04

GKnirps, 2017-08-26:
### What are you trying to do?
Hi! I'm trying to establish a network connection to my ev3 via USB. When I tried it last year, everything worked fine. When I tried it this morning, the network connection was established, I could find the device though its IP-Address (though not through a host name), but I was unable establish an ssh connection.

Since the ev3dev version I installed was an old one (I forgot which one), I installed a new version on the SD-Card, booted it and connected it with USB to my Computer. The Interface was configured as described in the tutorial.

### What did you expect to happen?
I expected network-manager to start up the interface an be able to connect to the ev3 

### What actually happened?
network-manager displayed an error message ("Aktivierung der Verbindung fehlgeschlagen: (2) Die aktive Verbindung wurde entfernt, bevor sie initialisiert wurde", for some reason, I am unable to get english error messages, this means something like "Activating the connection failed: (2) The active connection was removed before it could be initialized").

### What hardware and software are you using (including version numbers)?
I use an ev3 with ev3dev-jessie-ev3-generic-2017-06-09 and Ubuntu 17.04 with network-manager on my computer.

/var/log/syslog logs the following lines:
`Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7451] device (enp1s0u1): Activation: starting connection 'ev3dev' (fcd693b9-6cbe-3528-9cab-73376cbd5e09)
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7452] audit: op="connection-activate" uuid="fcd693b9-6cbe-3528-9cab-73376cbd5e09" name="ev3dev" pid=2175 uid=1000 result="success"
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7453] device (enp1s0u1): state change: disconnected -> prepare (reason 'none') [30 40 0]
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7456] device (enp1s0u1): state change: prepare -> config (reason 'none') [40 50 0]
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7461] device (enp1s0u1): state change: config -> ip-config (reason 'none') [50 70 0]
Aug 26 17:59:54 tuls kernel: [ 3128.037326] cdc_ether 3-1:1.0 enp1s0u1: kevent 12 may have been dropped
Aug 26 17:59:54 tuls kernel: [ 3128.037335] cdc_ether 3-1:1.0 enp1s0u1: kevent 12 may have been dropped
Aug 26 17:59:54 tuls kernel: [ 3128.037344] cdc_ether 3-1:1.0 enp1s0u1: kevent 12 may have been dropped
Aug 26 17:59:54 tuls avahi-daemon[1152]: Joining mDNS multicast group on interface enp1s0u1.IPv4 with address 10.42.0.1.
Aug 26 17:59:54 tuls avahi-daemon[1152]: New relevant interface enp1s0u1.IPv4 for mDNS.
Aug 26 17:59:54 tuls avahi-daemon[1152]: Registering new address record for 10.42.0.1 on enp1s0u1.IPv4.
Aug 26 17:59:54 tuls kernel: [ 3128.039308] cdc_ether 3-1:1.0 enp1s0u1: kevent 12 may have been dropped
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7593] Executing: /sbin/iptables --table filter --insert INPUT --in-interface enp1s0u1 --protocol tcp --destination-port 53 --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7611] Executing: /sbin/iptables --table filter --insert INPUT --in-interface enp1s0u1 --protocol udp --destination-port 53 --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7630] Executing: /sbin/iptables --table filter --insert INPUT --in-interface enp1s0u1 --protocol tcp --destination-port 67 --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7646] Executing: /sbin/iptables --table filter --insert INPUT --in-interface enp1s0u1 --protocol udp --destination-port 67 --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7660] Executing: /sbin/iptables --table filter --insert FORWARD --in-interface enp1s0u1 --jump REJECT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7673] Executing: /sbin/iptables --table filter --insert FORWARD --out-interface enp1s0u1 --jump REJECT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7685] Executing: /sbin/iptables --table filter --insert FORWARD --in-interface enp1s0u1 --out-interface enp1s0u1 --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7697] Executing: /sbin/iptables --table filter --insert FORWARD --source 10.42.0.0/255.255.255.0 --in-interface enp1s0u1 --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7709] Executing: /sbin/iptables --table filter --insert FORWARD --destination 10.42.0.0/255.255.255.0 --out-interface enp1s0u1 --match state --state ESTABLISHED,RELATED --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7722] Executing: /sbin/iptables --table nat --insert POSTROUTING --source 10.42.0.0/255.255.255.0 ! --destination 10.42.0.0/255.255.255.0 --jump MASQUERADE
Aug 26 17:59:54 tuls NetworkManager[1162]: <error> [1503763194.7738] device (enp1s0u1): share: (enp1s0u1) failed to start dnsmasq: Binärdatei »dnsmasq« konnte nicht gefunden werden
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7738] Executing: /sbin/iptables --table nat --delete POSTROUTING --source 10.42.0.0/255.255.255.0 ! --destination 10.42.0.0/255.255.255.0 --jump MASQUERADE
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7761] Executing: /sbin/iptables --table filter --delete FORWARD --destination 10.42.0.0/255.255.255.0 --out-interface enp1s0u1 --match state --state ESTABLISHED,RELATED --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7783] Executing: /sbin/iptables --table filter --delete FORWARD --source 10.42.0.0/255.255.255.0 --in-interface enp1s0u1 --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7797] Executing: /sbin/iptables --table filter --delete FORWARD --in-interface enp1s0u1 --out-interface enp1s0u1 --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7810] Executing: /sbin/iptables --table filter --delete FORWARD --out-interface enp1s0u1 --jump REJECT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7822] Executing: /sbin/iptables --table filter --delete FORWARD --in-interface enp1s0u1 --jump REJECT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7835] Executing: /sbin/iptables --table filter --delete INPUT --in-interface enp1s0u1 --protocol udp --destination-port 67 --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7850] Executing: /sbin/iptables --table filter --delete INPUT --in-interface enp1s0u1 --protocol tcp --destination-port 67 --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7864] Executing: /sbin/iptables --table filter --delete INPUT --in-interface enp1s0u1 --protocol udp --destination-port 53 --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7879] Executing: /sbin/iptables --table filter --delete INPUT --in-interface enp1s0u1 --protocol tcp --destination-port 53 --jump ACCEPT
Aug 26 17:59:54 tuls NetworkManager[1162]: <warn>  [1503763194.7893] device (enp1s0u1): Activation: Stage 5 of 5 (IPv4 Commit) start sharing failed.
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7893] device (enp1s0u1): state change: ip-config -> failed (reason 'sharing-start-failed') [70 120 18]
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7896] policy: disabling autoconnect for connection 'ev3dev'.
Aug 26 17:59:54 tuls NetworkManager[1162]: <warn>  [1503763194.7897] device (enp1s0u1): Activation: failed for connection 'ev3dev'
Aug 26 17:59:54 tuls NetworkManager[1162]: <info>  [1503763194.7906] device (enp1s0u1): state change: failed -> disconnected (reason 'none') [120 30 0]
Aug 26 17:59:54 tuls avahi-daemon[1152]: Withdrawing address record for 10.42.0.1 on enp1s0u1.
Aug 26 17:59:54 tuls avahi-daemon[1152]: Leaving mDNS multicast group on interface enp1s0u1.IPv4 with address 10.42.0.1.
Aug 26 17:59:54 tuls kernel: [ 3128.081887] cdc_ether 3-1:1.0 enp1s0u1: kevent 12 may have been dropped
Aug 26 17:59:54 tuls avahi-daemon[1152]: Interface enp1s0u1.IPv4 no longer relevant for mDNS.
Aug 26 17:59:54 tuls nm-applet[2175]: Aktivierung der Verbindung fehlgeschlagen: (2) Die aktive Verbindung wurde entfernt, bevor sie initialisiert wurde
`

dlech, 2017-08-26:
Can you post some screenshots of how you have configured the connection in NetworkManager?

I know there were some issues with systemd setting predictable network names and NetworkManager, but I don't know which versions are a problem.

GKnirps, 2017-08-26:
Thanks for the fast reply. Here are my settings (in German though, as mentioned above).
[image]
[image]

dlech, 2017-08-26:
The configuration looks correct.

I was able to resolve the problem by installing the `dnsmasq` package and restarting NetworkManager, then reconnecting the EV3.

```
sudo apt update
sudo apt install dnsmasq
sudo systemctl restart NetworkManager
```

GKnirps, 2017-08-26:
Thanks, installing dnsmasq worked. Maybe this should be added to the tutorial.

closed 2017-08-26 by GKnirps

dlech, 2017-08-26:
I already did. :wink:

GKnirps, 2017-08-26:
oh.