ev3dev/ev3dev #944: Slow write speed when setting duty_cycle_sp with BrickPi 2.8 shield.

cjeppe, 2017-08-28:
### Problem description
I observed a difference in performance of my developed controller algorithm with three identically built lego cars all with Raspberry Pi 3 and BrickPi shields. Two of the cars has a BrickPi 2.8 shield while the last has a BrickPi3 shield. The cars with the version 2.8 are having worse performance. I tracked down the issue to be due to long write times when the duty_cycle_sp is changed. Therefore I decided to write a simple program which only changes the duty_cycle_sp and store the individual write times in a log. The results are shown in the attached picture. Here it should be noted that the cars with BrickPi 2.8 are having a mean write time of 30 [ms] while the car with BrickPi 3 shield is having an average write time of 0.28 [ms]. But even worse is the long sporadic spikes of hundreds of [ms] which are degrading the control algorithm significantly.
Is this to be expected with the older BrickPi 2.8 shield? Or is it an actual bug which can be fixed?
[image]

### Hardware and software
There happens to be a small difference in the image and kernel version. Has the kernel been improved between these versions?

BrickPi3 Unit 3
Image file:         ev3dev-jessie-rpi2-generic-2017-04-14
Kernel version:     4.9.22-ev3dev-1.1.0-rpi2
Board:              Raspberry Pi 3 Model B Rev 1.2
Revision:           a02082
Brickman:           0.8.1
ev3devKit:          0.4.3

BrickPi 2.8 Unit 1 and Unit 2
Image file:         ev3dev-jessie-rpi2-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-rpi2
Board:              Raspberry Pi 3 Model B Rev 1.2
Revision:           a02082
Brickman:           0.8.1
ev3devKit:          0.4.3

dlech, 2017-08-28:
Nice analysis. :smile: (I wish more people would take a little time like this to analyze a problem. It really helps make ev3dev better.)

So yes, this is somewhat expected. BrickPi3 is totally redesigned hardware. It has one CPU and uses a duplex SPI bus for communications at 500kHz. The response from the BrickPi3 is read at the same time as we are sending the message from the RPi.

BrickPi/BrickPi+, on the other hand, have 2 CPUs and use UART for communication at 500kbaud. Technically, this could be duplex, but in practice it works more simplex. The RPi sends a message to the BrickPi and then has to wait around until the BrickPi sends a message back.

Since both have about the same data rate, I would not expect there to be a 100x difference in speed though. I have a feeling that the fact we are using work queues for scheduling is what is slowing things down. (e.g. <https://github.com/ev3dev/lego-linux-drivers/blob/v1.4.0/brickpi/brickpi_ld.c#L401>)

So, yeah, let's call this a bug and see what we can do about it.

dlech, 2018-04-22:
@cjeppe, do you have some test code to reproduce your results? I'm not seeing this on ev3dev-stretch

```
Image file:         ev3dev-stretch-rpi-generic-2017-09-01
Kernel version:     4.14.33-ev3dev-1.6.1-rpi+
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=000D
BOARD_INFO_MODEL=Raspberry Pi Model B Rev 2
BOARD_INFO_SERIAL_NUM=00000000dc468384
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=2
BOARD_INFO_MODEL=Dexter Industries BrickPi
BOARD_INFO_TYPE=aux
```

using the following program

```python
from timeit import timeit

with open('/sys/class/tacho-motor/motor0/command', 'w') as f:
    f.write('run-direct')
with open('log.txt', 'w') as log:
    with open('/sys/class/tacho-motor/motor0/duty_cycle_sp', 'w') as f:
        for i in range(2500):
            t = timeit('f.write(str(0)); f.flush()', number=1, globals=globals())
            log.write('{}\n'.format(t))
        f.write(str(0))
```

I get [these results](https://github.com/ev3dev/ev3dev/files/1935165/log.txt). On average, it is not quite as good as BrickPi3, but below 2ms most of the time and the median value is 0.24ms, which is actually slightly faster than the numbers you have for BrickPi3.

I have recently made changes to the brickpi driver that could have inadvertently fixed this issue.

dlech, 2019-03-25:
Assuming this is fixed

closed 2019-03-25 by dlech