ev3dev/ev3dev #946: Trying to connect to Wi-Fi using username/password get error GDBus.Error

CaptnMo, 2017-08-30:
I am trying to connect to our company network using Wi-Fi. The company network is asking for a username and password. On the brick I get the following error:
**GDBus.Error:net.connman.Error.InvalidArguments: Invalid Arguments**

Working with a personal hotspot using my IPhone works fine. We also added a 
sudo nano /etc/wpa_supplicant/wpa_supplicant.conf
config file but that would not help.

dlech, 2017-08-30:
Brickman *should* be able to handle this, but I don't have any such network to test with, so there it probably a bug.

As a workaround, you can connected via USB or Bluetooth to get an SSH terminal and then use the `connmanctl` command to connect to the Wi-Fi network,

http://www.ev3dev.org/docs/tutorials/setting-up-wifi-using-the-command-line/

Technonati, 2019-11-06:
My team's school's wifi password is ABOMINABLY LONG - 35 characters - and the brick says the connection times out.

So, the solution above you mention is what I should use?

Thanks in advance for your advice.

dlech, 2019-11-06:
yes and yes

Technonati, 2019-11-06:
Awesome.

Thank you.

My team thanks you.