ev3dev/ev3dev #947: Bluetooth becomes unavailable when 2 Large Motors attached

phantom-8, 2017-08-31:
### What are you trying to do?
Establish bluetooth connection with OSX Sierra

### What did you expect to happen?
Bluetooth available after bootup every single time.

### What actually happened?
With 2 EV3 Large Motors attached, EV3 bluetooth became unreliable after bootup.  There maybe around 30% chance of working.  Most of the time, the bluetooth became unavailable upon bootup.  Reboot didn't always fix it.  It seemed to be random whether it worked or not.

I'm aware of previous reported bluetooth bootup issues.  However, most of those reported cases related to sensors being plugged in.  During my testing, the bluetooth problem only occurred with 2 EV3 large motors attached and no other sensors attached.  I had also tried to connect 4 Color Sensors and no motors, the bluetooth booted up fine every single time.  Thus, I'm not sure if this is related to previous reported bluetooth issues.

Besides the latest release 2017-06-09, I've tested release 2017-02-11 and it had the same problem.  However, with release dated 2016-12-21, the bluetooth booted up perfectly fine every single time with 2 EV3 Large Motors.  Thus, the problem seemed to be introduced in release 2017-02-11 and afterwards.

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-06-09
Kernel version:     4.4.68-20-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

hciconfig -a output
```
● ev3-bluetooth.service - LEGO MINDSTORMS EV3 built-in Bluetooth
   Loaded: loaded (/lib/systemd/system/ev3-bluetooth.service; static)
   Active: failed (Result: exit-code) since Thu 2017-08-31 15:32:32 UTC; 3min 36s ago
  Process: 263 ExecStart=/usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 flow nosleep (code=exited, status=1/FAILURE)

Aug 31 15:32:30 ev3dev hciattach[263]: Found a Texas Instruments' chip!
Aug 31 15:32:30 ev3dev hciattach[263]: Firmware file : /lib/firmware/ti-connectivity/TIInit_6.6.15.bts
Aug 31 15:32:30 ev3dev hciattach[263]: Loaded BTS script version 1
Aug 31 15:32:30 ev3dev hciattach[263]: texas: changing baud rate to 2000000, flow control to 1
Aug 31 15:32:30 ev3dev hciattach[263]: Cannot send hci command to socket: Connection timed out
Aug 31 15:32:30 ev3dev hciattach[263]: Can't initialize device: Connection timed out
Aug 31 15:32:32 ev3dev systemd[1]: ev3-bluetooth.service: control process exited, code=exited status=1
Aug 31 15:32:32 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 built-in Bluetooth.
Aug 31 15:32:32 ev3dev systemd[1]: Unit ev3-bluetooth.service entered failed state.
```

journalctl -b | grep -i bluetooth
```
Aug 31 15:32:08 ev3dev kernel: Bluetooth: Core ver 2.21
Aug 31 15:32:08 ev3dev kernel: Bluetooth: HCI device and connection manager initialized
Aug 31 15:32:08 ev3dev kernel: Bluetooth: HCI socket layer initialized
Aug 31 15:32:08 ev3dev kernel: Bluetooth: L2CAP socket layer initialized
Aug 31 15:32:08 ev3dev kernel: Bluetooth: SCO socket layer initialized
Aug 31 15:32:08 ev3dev kernel: Bluetooth: HCI UART driver ver 2.3
Aug 31 15:32:08 ev3dev kernel: Bluetooth: HCI UART protocol H4 registered
Aug 31 15:32:08 ev3dev kernel: Bluetooth: HCI UART protocol BCSP registered
Aug 31 15:32:08 ev3dev kernel: Bluetooth: HCI UART protocol LL registered
Aug 31 15:32:08 ev3dev kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Aug 31 15:32:08 ev3dev kernel: Bluetooth: BNEP socket layer initialized
Aug 31 15:32:32 ev3dev systemd[1]: ev3-bluetooth.service: control process exited, code=exited status=1
Aug 31 15:32:32 ev3dev systemd[1]: Failed to start LEGO MINDSTORMS EV3 built-in Bluetooth.
Aug 31 15:32:32 ev3dev systemd[1]: Unit ev3-bluetooth.service entered failed state.
Aug 31 15:32:34 ev3dev systemd[1]: Starting Bluetooth service...
Aug 31 15:32:35 ev3dev bluetoothd[339]: Bluetooth daemon 5.23
Aug 31 15:32:35 ev3dev bluetoothd[339]: Starting SDP server
Aug 31 15:32:35 ev3dev bluetoothd[339]: Bluetooth management interface 1.10 initialized
Aug 31 15:32:36 ev3dev systemd[1]: Started Bluetooth service.
Aug 31 15:32:37 ev3dev systemd[1]: Starting Bluetooth.
Aug 31 15:32:37 ev3dev systemd[1]: Reached target Bluetooth.
Aug 31 15:32:37 ev3dev systemd[1]: Service bluetooth.target is not needed anymore. Stopping.
Aug 31 15:32:37 ev3dev systemd[1]: Stopping Bluetooth.
Aug 31 15:32:37 ev3dev systemd[1]: Stopped target Bluetooth.
Aug 31 15:35:56 ev3dev sudo[514]: robot : TTY=pts/0 ; PWD=/home/robot ; USER=root ; COMMAND=/bin/systemctl -l status ev3-bluetooth.service
Aug 31 15:36:08 ev3dev sudo[522]: robot : TTY=pts/0 ; PWD=/home/robot ; USER=root ; COMMAND=/bin/systemctl -l status ev3-bluetooth.service
```

phantom-8, 2017-09-01:
I've added a sleep command before hciattach and that fixed the bluetooth unavailable problem.  The issue seems to be certain system resource wasn't ready (UART?) when calling the hciattach.  Here is my updated file "ev3-bluetooth.service" with the added sleep command.  Currently, it slept for 5 seconds and you may be able to tune it down a bit.  Sleep for 2 seconds also seems to work fine for me.

```
[Unit]
Description=LEGO MINDSTORMS EV3 built-in Bluetooth
# TODO: This binding does not actually do anything useful. What we really want
# to do is kill hciattach when the enable attribute of this device is set to 0.
BindsTo=sys-devices-platform-legoev3\x2dbluetooth.device
# If any of these starts before hciattach is complete, hciattach could fail.
Before=bluetooth.service connman.service systemd-rfkill@rfkill0.service

[Service]
Type=forking
# This can fail due to timeout during boot because the system is so busy.
# We give it a long timeout and use nice to try to keep if from failing.
ExecStart=/bin/sh -c '/bin/sleep 5; \
        /usr/bin/hciattach -t 30 /dev/ttyS2 texas 2000000 flow nosleep'
Nice=-10
```

dlech, 2017-09-01:
Thanks for the detailed information here.

I was not able to reproduce the issue with just motors attached, but I am able to reproduce by also attaching 4 UART sensors and shaking them around during boot so that they produce lots of data.

It is interesting that delaying `hciattach` fixes the problem, but it would be better to understand the root cause of the problem. I had to extend the time to 10 seconds to reliably start Bluetooth using my worst-case 4 sensors.

The observation about the older image working could be a good lead though. I am glad you tested it. Here are some changes between those two versions that could effect what is going on here.

* https://github.com/ev3dev/ev3-kernel/commit/efa7fa546a9e2e6526c84ab56a0fe61d85f3a792
* https://github.com/ev3dev/lego-linux-drivers/commit/03011a571922cfee6a245b310be5930813488eb6

I'll trying digging into that to see if we can come up with a better solution. It's tricky because the publicly available LEGO schematics are not accurate when it comes to Bluetooth.

dlech, 2017-09-02:
I've added your workaround Can you see if the 2017-09-01 ev3dev-jessie snapshot works for you?

https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2017-09-02/

phantom-8, 2017-09-02:
Just tested your new snapshot.  It works fine for me.  Thanks for taking the time to confirm the issue.  I hope you can find the root cause eventually.

dlech, 2017-09-02:
I think we'll call this workaround good enough for ev3dev-jessie, so closing. Tracking ev3dev-stretch in #925.

closed 2017-09-02 by dlech