ev3dev/ev3dev #949: Doubt about Gyro Filter command on AbsoluteIMU support

jabrena, 2017-09-03:
Hi,

I was checking the technical documentation for the Mindsensors IMU:
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensor_data.html#ms-absolute-imu

But I discovered that using the API, it is not possible to use the feature to set a filter for gyro data:
http://www.mindsensors.com/ev3-and-nxt/15-gyro-multisensitivity-accelerometer-and-compass-for-nxt-or-ev3
http://www.mindsensors.com/pdfs/AbsoluteIMU-User-Guide.pdf

```
protected static final int  GYRO_FILTER         = 0x5a;
```

In my Brick, I have connected the sensor on Port 1 and the brick detects the I2C device connected:

```
robot@brick1:~$ ls /dev/i2c-in*
/dev/i2c-in1  /dev/i2c-in2
robot@brick1:~$ udevadm info -q path -n /dev/i2c-in1
/devices/platform/legoev3-ports/lego-port/port0/i2c-legoev3.3/i2c-3/i2c-dev/i2c-3
``` 

How to write the register 0x5a to set a value from 0-7?

Many thanks in advance

Juan Antonio

dlech, 2017-09-03:
> How to write the register 0x5a to set a value from 0-7?

See http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/i2c.html#direct-reading-and-writing-of-the-sensor

jabrena, 2017-09-03:
Many thanks @dlech,

``` 
robot@brick1:/sys/class/lego-sensor$ ls
sensor1
robot@brick1:/sys/class/lego-sensor$ cd sensor1
robot@brick1:/sys/class/lego-sensor/sensor1$ ls
address   bin_data_format  commands  device  driver_name  mode   num_values  power      text_value  units   value1  value3  value5  value7
bin_data  command          decimals  direct  fw_version   modes  poll_ms     subsystem  uevent      value0  value2  value4  value6
robot@brick1:/sys/class/lego-sensor/sensor1$ echo -e -n "4" | dd bs=1 of=direct seek=$(( 0x5a ))
1+0 records in
1+0 records out
1 byte (1 B) copied, 0.0097268 s, 0.1 kB/s
robot@brick1:/sys/class/lego-sensor/sensor1$ 
```

dlech, 2017-09-03:
Which programming language do you want to use?

jabrena, 2017-09-03:
`Java`, in my case.

dlech, 2017-09-03:
does it have low level read, write and seek functions for files?

jabrena, 2017-09-03:
Hi @dlech,

Are you talking about this one?
https://www.tutorialspoint.com/java/io/randomaccessfile_seek.htm

For the statement to set the value on the register, I have some problem:

```
echo -e -n "4" | dd bs=1 of=direct seek=$(( 0x5a ))
``` 

In a dynamic way, the Java program has to write the value on direct:

```
echo -e -n "5" | dd bs=1 of=/sys/class/lego-sensor/sensor1/direct seek=$(( 0x5a ))
``` 

Does exist other way in the second part of the statement to put the value (5)?

Juan Antonio

dlech, 2017-09-03:
OK, so maybe something like this:

```java
RandomAccessFile direct = new RandomAccessFile("/sys/class/lego-sensor/sensor1/direct", "rw");

// writing
// equivalent of echo -e -n "\x04" | dd bs=1 of=direct seek=$(( 0x5a ))
direct.seek(0x5a);
direct.writeByte(4);

// reading
// equivalent of hd -s $(( 0x5a )) -n 8 direct
direct.seek(0x05a);
byte data = direct.readByte();
```

jabrena, 2017-09-03:
Hi @dlech,

using the idea that you say, it is possible to write in the file but the value doesn`t write the same part that manually:

``` 
RandomAccessFile direct = new RandomAccessFile("/sys/class/lego-sensor/sensor1/direct", "rw");

// writing
// equivalent of echo -e -n "\x04" | dd bs=1 of=direct seek=$(( 0x5a ))
direct.seek(0x5a);
direct.writeByte((byte)2);
```

Output:

```
robot@brick1:/sys/class/lego-sensor/sensor2$ hd -s $(( 0x5a )) -n 8 direct
0000005a  02 00 00 00 00 00 00 00                           |........|
00000062

``` 

If I execute manually:

``` 
robot@brick1:/sys/class/lego-sensor/sensor2$  echo -e -n "5" | dd bs=1 of=/sys/class/lego-sensor/sensor2/direct seek=$(( 0x5a ))
1+0 records in
1+0 records out
1 byte (1 B) copied, 0.00999186 s, 0.1 kB/s
robot@brick1:/sys/class/lego-sensor/sensor2$ hd -s $(( 0x5a )) -n 8 direct
0000005a  35 00 00 00 00 00 00 00                           |5.......|
00000062
```

jabrena, 2017-09-03:
I found a solution:

If value = 1:

``` java
        final String i2c_command2 = "echo \"" + value + "\" | dd bs=1 of=" + this.PATH_DEVICE + "/direct seek=$(( 0x5a ))";
        String[] cmd = {
                "/bin/sh",
                "-c",
                i2c_command2
        };
        String response = Shell.execute(cmd);
``` 

Review by command line:

```
robot@brick1:/sys/class/lego-sensor/sensor2$ hd -s $(( 0x5a )) -n 8 direct
0000005a  31 00 00 00 00 00 00 00                           |1.......|
00000062
robot@brick1:/sys/class/lego-sensor/sensor2$ 
```

I think that you have reason, I will research about seek features with Java

dlech, 2017-09-03:
> using the idea that you say, it is possible to write in the file but the value doesn`t write the same part that manually:

It looks the same to me. In java, you write `(byte)2` and `hd` shows hex `02` as the first byte.

In bash, you write ASCII "5", which is 0x35 and `hd` shows hex `35` as the first byte.

jabrena, 2017-09-03:
Tomorrow, I will review again the notes that you say.

dlech, 2017-10-03:
Can this be closed?

jabrena, 2017-10-03:
I will review today

closed 2020-08-17 by dlech