ev3dev/ev3dev #950: How to read a Boolean value from Grove Adapter?

jabrena, 2017-09-03:
Hi, I am following the article from @JorgePe and the Documentation from the sensor which I am using for the Tests:
http://wiki.seeed.cc/Grove-PIR_Motion_Sensor/
http://ofalcao.pt/blog/2017/using-grove-devices-to-the-ev3

And I have a doubt about how to read a Digital value from the sensor.

dlech, 2017-09-03:
`i2set` will write to the device. If you want to read data, you would need to use `i2cget`.

jabrena, 2017-09-03:
Many thanks @dlech,

I will read the documentation:
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/i2c.html

and some technical docs about i2c:
https://www.cs.sfu.ca/CourseCentral/433/bfraser/other/I2CGuide.pdf

Juan Antonio

dlech, 2017-10-03:
Can this be closed?

closed 2020-08-17 by dlech