ev3dev/ev3dev #952: new getting started experience for ev3dev-stretch

dlech, 2017-09-05:
I've started a rough draft of what I think the new getting started page for ev3dev-stretch should look like based on feedback we've received over the years.

https://github.com/ev3dev/ev3dev/blob/ev3dev-stretch/docs/getting-started-ev3.rst

Comments, suggestions, pull requests, etc., please.

WasabiFan, 2017-09-05:
Question: Do you intend to put this on ev3dev.org or RTD? I would vote for ev3dev.org because we have formatting control for download flow and visually splitting tracks, but I'm open to other options. My thinking is that technical docs should be on RTD and UX-focused ones should be on ev3dev.org.

WasabiFan, 2017-09-05:
> TODO: make a list of supported Wi-Fi dongles somewhere

http://www.ev3dev.org/docs/networking/#with-a-wi-fi-dongl

Otherwise, looks good -- It'd be great if we could actually make set-up that simple :laughing: I'm interested to see what you can do there.

jabrena, 2017-09-05:
Today, I will test the extension for Java:
https://marketplace.visualstudio.com/items?itemName=georgewfraser.vscode-javac

jabrena, 2017-09-05:
Hi @dlech, @WasabiFan ,

When in the plugin appear this file:

``` 
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Download and Run",
            "type": "ev3devBrowser",
            "request": "launch",
            "program": "/home/robot/${workspaceRootFolderName}/hello.py"
        }
    ]
}
``` 

When the extesion use the attribute `"request": "launch",` is it using the new command-line tool: `brickrun`?
https://github.com/ev3dev/ev3dev/issues/790

Juan Antonio

JorgePe, 2017-09-05:
That part of installing ev3dev with Etcher... is it because Etcher is available for all major OS's?
I just use gnome-disk-utility ("Disks" in Ubuntu).

dlech, 2017-09-05:
> Do you intend to put this on ev3dev.org or RTD?

I'm leaning towards having most docs on RTD, mainly because of the versioning capability. The download page will stay on ev3dev.org for sure though.

dlech, 2017-09-05:
> When the extesion use the attribute "request": "launch", is it using the new command-line tool: brickrun?

Yes, it is.

dlech, 2017-09-05:
> That part of installing ev3dev with Etcher... is it because Etcher is available for all major OS's?

Yes. It lets us have one set of instructions that works on all OS's. And also, it uses bmaps, which speeds up flashing significantly by skipping all of the parts of the disk image that are just 0s.

dwalton76, 2017-09-05:
> MicroSDHC memory card (cards > 32GB will not work

What's the scoop on the 32GB limit? I have a 128GB card in my EVB, everything with the filesystem seems to be ok.

dlech, 2017-09-05:
The page is for EV3 only. The EV3 hardware is only capable of [SDHC](https://en.wikipedia.org/wiki/Secure_Digital#SDHC), which means 32GB max. Based on feedback from ev3dev users, cards larger than that may seem to sort of work, but have issues with random data corruption.

I will add additional pages for Raspberry Pi and BeagleBone, which may have different requirements.

dlech, 2017-09-05:
According to [this](https://groups.google.com/forum/#!topic/beagleboard/rK3y2ih1kig) BeagleBone only officially supports SDHC as well, but SDXC will work albeit much slower than it should. So, our official docs will still say 32GB max for BeagleBone as well.

dwalton76, 2017-09-06:
Gotcha.  I have a 128G in an EV3 too but it is still running jessie.  I haven't noticed issues with either but I haven't done tons with the EVB lately.

dlech, 2017-09-06:
Pretty version with additional platforms: http://docs.ev3dev.org/en/ev3dev-stretch/index.html

JorgePe, 2017-09-06:
Nice style sheet.
I like the way you are structuring the information.
And the tab tip is a good idea, it gives a fresh look... I started using tabs 2 years ago and now even write on it to know what is inside the SD card.
Perhaps people have more tips? Like using a wall charger and a LiPO battery (or fake batteries) as an UPS for long sessions.

dwalton76, 2017-09-07:
Looks nice on my iPhone

BertLindeman, 2017-09-11:
Looks good.

Is it helpful to "edit on github" for `http://docs.ev3dev.org/en/ev3dev-stretch/index.html`?
Some small typo's like "Rapsberry" and the lego site is not https so the link issues an error message.

dlech, 2017-09-11:
Yes, that would be very helpful.

BertLindeman, 2017-09-11:
Done ;-)

dlech, 2017-10-28:
Stretch docs have been updated with more detailed platform-specific information: http://docs.ev3dev.org/en/ev3dev-stretch/platforms/index.html

dlech, 2017-10-28:
Additional ideas: #734, #566, #492

JorgePe, 2017-10-29:
I can't edit the Hardware Platforms so I add my suggestion here:
- Raspberry Pi Zero W also works with ev3dev and you can add it also to note 19 (built-in BT BLE) and 21, built-in Wi-Fi 
I didn't tried it yet with BrickPi3, just ev3dev by its own. To bad it's so difficult to buy here in Portugal because I like it very much, can take it with me everywhere.
I see people sing Arduino as a gateway to connect sensors to EV3 (I use the NodeMCU for the same reasons because it's small and has Wi-Fi) but I see the RpiZeroW as a much better "companion".

dlech, 2017-10-29:
FYI: https://github.com/rtfd/readthedocs.org/issues/3203

If you delete "origin" from the link, you should be able to edit pages.

dlech, 2017-10-29:
> Raspberry Pi Zero W also works with ev3dev and you can add it also to note 19 (built-in BT BLE) and 21, built-in Wi-Fi

I added Zero W to the table, but it looks like it was already mentioned in notes 19 and 21.

JorgePe, 2017-10-29:
I tried to edit the table but the preview showed nothing so I canceled. Do you edit the table "by hand" or do you use some editor? The table in markup is quite a mess for someone without experience (i.e. me).

dlech, 2017-10-29:
By hand. It helps to have a text editor with fixed width font and multiple cursors. I'm sure it is nearly impossible in the GitHub online editor.