ev3dev/ev3dev #955: Wi-Fi wiki page

dlech, 2017-09-06:
I started a Wi-Fi wiki page. https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles

If you use Wi-Fi with ev3dev, please take a minute to add your device to the list.

JorgePe, 2017-09-07:
Added ASUS USB-N10 Nano to the known working dongles. Not quite the same chipset as the Edimax but it works fine, also with LeJOS.

BertLindeman, 2017-09-07:
Updated:

Typo "and" should be "add" on the first line.

Added 
`and the Raspberry Pi Zero W`
to
`Built-in Wi-Fi on the Raspberry Pi 3  is also supported`

bixnel, 2017-09-10:
Added `D-Link DWL-G132`. Works great.

superPershing, 2017-10-17:
Are there any dongle support 5G-hz WIFI?

dlech, 2017-10-17:
I suspect that all of the supported dongles are capable of 5GHz since they all are capable of 802.11n.

superPershing, 2017-10-18:
@dlech I use the officially supported dongle: "WNA1100 Wireless-N 150`, it says that it's capable of 802.11n, but it only support 2.4000~2.4835GHz from its packaging introduction, and I can only see 2.4GHz wifi from EV3's screen.

dlech, 2017-10-24:
Thanks to everyone who edited the page.

closed 2017-10-24 by dlech