ev3dev/ev3dev #956: Recent stretch snapshot images do not boot on rpi-zero-w

BertLindeman, 2017-09-07:
### What are you trying to do?
Boot Raspberry pi zero with ev3dev snapshot at stretch level

### What did you expect to happen?
;-) boot

### What actually happened?
Only two blinks of the green onboard led, futher NOTHING.
I went back over the older stretch snapshots.
They acted the same until I used
`ev3dev-stretch-rpi-generic-2017-07-26`
This image is the NEWEST stretch image that DOES boot.

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi-generic-2017-07-26
Kernel version:     4.9.39-ev3dev-1.3.0-rpi
/usr/bin/ev3dev-sysinfo: line 68: warning: command substitution: ignored null byte in input
Board:              Raspberry Pi Zero W Rev 1.1
Revision:           9000c1
Brickman:           0.9.0
ev3devKit:          0.5.1
```

dlech, 2017-09-07:
I probably updated the Raspberry Pi bootloader package around that time. I'll try updating it again.

Also, `Raspberry Pi Zero W Rev 1.1` needs to be added to the flash-kernel package.

BertLindeman, 2017-09-09:
Not sure if this proves this **is** fixed but I did the following on an older stretch snapshot image 
`Linux rpi0 4.9.39-ev3dev-1.3.0-rpi #1 Tue Jul 25 15:50:03 CDT 2017 armv6l GNU/Linux`:
```
sudo apt-get update && sudo apt-get -V upgrade
sudo flash-kernel
```
This updated flash-kernel to ` 3.75ev3dev5`
Output of flash-kernel:
```
Using DTB: bcm2708-rpi-0-w.dtb
Installing /usr/lib/linux-image-4.9.41-ev3dev-1.4.0-rpi/bcm2708-rpi-0-w.dtb into /boot/dtbs/4.9.41-ev3dev-1.4.0-rpi/bcm2708-rpi-0-w.dtb
Installing new bcm2708-rpi-0-w.dtb.
flash-kernel: installing version 4.9.41-ev3dev-1.4.0-rpi
Will use /dev/mmcblk0p1 as boot device.
Taking backup of kernel.img.
Installing new kernel.img.
Installing new bcm2708-rpi-0-w.dtb.
```
Then `sudo reboot` 
Fingers crossed....

BOOT OK
```#uname -a
Linux rpi0 4.9.41-ev3dev-1.4.0-rpi #1 Fri Aug 11 12:38:04 CDT 2017 armv6l GNU/Linux
```
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi-generic-2017-07-26
Kernel version:     4.9.41-ev3dev-1.4.0-rpi
/usr/bin/ev3dev-sysinfo: line 68: warning: command substitution: ignored null byte in input
Board:              Raspberry Pi Zero W Rev 1.1
Revision:           9000c1
Brickman:           0.10.0
ev3devKit:          0.5.2
```

So closing.
Thank you @dlech

closed 2017-09-09 by BertLindeman

dlech, 2017-09-09:
When the next image snapshot is released, it should have the `bcm2708-rpi-0-w.dtb` installed. This is what was preventing booting.

dlech, 2017-09-09:
Please try the 2017-09-09 ev3dev-stretch snapshot.

https://oss.jfrog.org/artifactory/list/oss-snapshot-local/org/ev3dev/brickstrap/

BertLindeman, 2017-09-10:
Dropped `snapshot-ev3dev-stretch-rpi-generic-2017-09-09.img.xz`  onto an sd-card.
rpi-zero-w boots OK. So verified fix **is** working.

Had no display on hdmi on these images before. 
Just to be sure I now had a display on the rpi0. So Brickman comes up on that huge screen with that small font. 
One word: **FUN**. 

Thanks all for EV3DEV!

dlech, 2017-09-10:
I recently added 32bpp support for Brickman, so I was wondering if it would work on the HDMI display now. Nice to know that it does.