ev3dev/ev3dev #957: [stretch] Left brick button is not working in brickman

ddemidov, 2017-09-10:
I've flashed the latest ([snapshot-ev3dev-stretch-ev3-generic-2017-09-09.img.xz](https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2017-09-09/snapshot-ev3dev-stretch-ev3-generic-2017-09-09.img.xz)) stretch image and noticed that brickman interface ignores the left brick button. This makes things like entering wi-fi password difficult. This is not a hardware issue: when I remove the SD card and boot into the stock system, the button works.

ddemidov, 2017-09-10:
I tried to test this from python. I had to apply this patch to [ev3dev/ev3.py](https://github.com/rhempel/ev3dev-lang-python/blob/develop/ev3dev/ev3.py) for the `Button` class to work at all:
```diff
diff --git a/ev3dev/ev3.py b/ev3dev/ev3.py
index 725f713..ca295ed 100644
--- a/ev3dev/ev3.py
+++ b/ev3dev/ev3.py
@@ -165,12 +165,12 @@ class Button(ButtonEVIO):
 
 
     _buttons = {
-            'up': {'name': '/dev/input/by-path/platform-gpio-keys.0-event', 'value': 103},
-            'down': {'name': '/dev/input/by-path/platform-gpio-keys.0-event', 'value': 108},
-            'left': {'name': '/dev/input/by-path/platform-gpio-keys.0-event', 'value': 105},
-            'right': {'name': '/dev/input/by-path/platform-gpio-keys.0-event', 'value': 106},
-            'enter': {'name': '/dev/input/by-path/platform-gpio-keys.0-event', 'value': 28},
-            'backspace': {'name': '/dev/input/by-path/platform-gpio-keys.0-event', 'value': 14},
+            'up': {'name': '/dev/input/by-path/platform-gpio_keys-event', 'value': 103},
+            'down': {'name': '/dev/input/by-path/platform-gpio_keys-event', 'value': 108},
+            'left': {'name': '/dev/input/by-path/platform-gpio_keys-event', 'value': 105},
+            'right': {'name': '/dev/input/by-path/platform-gpio_keys-event', 'value': 106},
+            'enter': {'name': '/dev/input/by-path/platform-gpio_keys-event', 'value': 28},
+            'backspace': {'name': '/dev/input/by-path/platform-gpio_keys-event', 'value': 14},
         }
 
     @property
```
After that, all buttons except for the 'left' one work as intended. 'Left' button is always reported to be pressed. This script was used:
```py
>>> import ev3dev.ev3 as ev3
>>> b = ev3.Button()
>>> b.left # while not pressing the button
True
>>> b.left # while pressing the button
True
>>> b.right # while not pressing the button
False
>>> b.right # while pressing the button
True

```

dwalton76, 2017-09-10:
Using @ddemidov patch all of the buttons work for me...I'm not seeing `left` as stuck.  Tested with
```
robot@ev3alt[ev3dev-lang-python]# cat utils/button.py 
#!/usr/bin/env python3

from ev3dev.auto import Button
import logging
import sys
from time import sleep

# logging
logging.basicConfig(level=logging.INFO,
                    format="%(asctime)s %(levelname)5s: %(message)s")
log = logging.getLogger(__name__)

btn = Button()

while True:
    print("left %s, right %s, up %s, down %s, backspace %s, enter %s" % (btn.left, btn.right, btn.up, btn.down, btn.backspace, btn.enter))
    sleep(1)
robot@ev3alt[ev3dev-lang-python]# 
```

dlech, 2017-09-10:
You probably haven't upgraded your kernel. This bug is a regression in the `ev3dev-`1.5.0` kernel released yesterday.

dlech, 2017-09-14:
Fix released in the `ev3dev-1.5.1` kernel

closed 2017-09-14 by dlech