ev3dev/ev3dev #958: [stretch] Built-in wifi on rpi-zero-w driver problem

BertLindeman, 2017-09-10:
### What are you trying to do?
Activate built-in wifi on rpi-zero-w running stretch image

### What did you expect to happen?
Auto setup for driver brcmfmac

### What actually happened?
last output of `dmesg` after `modprobe brcmfmac`
```
[Sun Sep 10 09:59:04 2017] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[Sun Sep 10 09:59:04 2017] usbcore: registered new interface driver brcmfmac
[Sun Sep 10 09:59:04 2017] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.txt failed with error -2
[Sun Sep 10 09:59:05 2017] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[Sun Sep 10 09:59:06 2017] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi-generic-2017-07-26
Kernel version:     4.9.48-ev3dev-1.5.0-rpi
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=9000C1
BOARD_INFO_MODEL=Raspberry Pi Zero W Rev 1.1
BOARD_INFO_SERIAL_NUM=0000000055cbb493
```

Seen `bin` but no `txt` file there:
```
 ll /lib/firmware/brcm/brcmfmac4343*
-rw-r--r-- 1 root root 369577 Jun  6 01:56 /lib/firmware/brcm/brcmfmac43430-sdio.bin
```
Package `firmware-brcm80211` delivers `/lib/firmware/brcm/brcmfmac43430-sdio.bin`
the package is at version `20161130-3`

No high priority issue for me, as I use an eth-cable at the moment.

Maybe helps: On Raspbian I saw:
```
$ dmesg|grep -i wl
[    7.555042] brcmfmac: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[   10.483233] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   13.281245] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

dlech, 2017-09-14:
Can you check something on the official Rasbian? 

    dpkg -S /lib/firmware/BCM43430A1.hcd

This is the Bluetooth firmware for RPi3.

dlech, 2017-09-14:
I think I figured it out. `bluez-firmware`.

Please try the ev3dev-stretch 2017-09-14 rpi image (in about an hour).

BertLindeman, 2017-09-14:
A bit of confusion, @dlech in the 20170914 directory the rpi image is named 20170909
`https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2017-09-14/snapshot-ev3dev-stretch-rpi-generic-2017-09-09.img.xz`
Will use this image anyway as the filedate is "today" and the "packed size" is different

dlech, 2017-09-14:
Travis CI is having some issues today. I might have to retry that build *again*.

dlech, 2017-09-14:
I don't expect the rebuilt 2017-09-09 to work. It will be identical to the previous image.

BertLindeman, 2017-09-14:
Confirmed ;-)

dlech, 2017-09-14:
Correct image is now available.

BertLindeman, 2017-09-14:
Doawnloaded rpi  image 2017-09-14- booted rpi-zero-w
```
grep -i brcm
[    9.002459] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    9.043673] usbcore: registered new interface driver brcmfmac
[    9.678830] brcmfmac: Firmware version = wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
```
So success!

For convenience used connmanctl via ssh to setup the connection.

So with THANKS, closing.

closed 2017-09-14 by BertLindeman