ev3dev/ev3dev #959: Wireless USB Adapter for 5GHz

cho934, 2017-09-11:
Hi,

I want to connect the EV3dev to a new wifi network using 5Ghz only.

What are the compatible Wireless USB Adapters  I can buy?

dlech, 2017-09-11:
We have a wiki page for this. https://github.com/ev3dev/ev3dev/wiki/USB-Wi-Fi-Dongles

hglee0128, 2017-09-21:
ID 148f: 7601 Ralink Technology

Are these products not supported?

Are only the dongles on the wiki page supported?

dlech, 2017-09-21:
Only the dongles listed as "supported" on the wiki page are "supported". "Supported" means I own them and if they don't work, I will try to fix it. Many other dongles work just fine, including, but not limited to those listed on the wiki. But you will have to fix it yourself if it does not work or stops working.

dlech, 2017-09-22:
If you have this Ralink and find that it works, please add it to the wiki. I know some others have used Ralink successfully in the past, but I don't know which model exactly.

closed 2020-08-17 by dlech