ev3dev/ev3dev #961: How to use GRX?

jabrena, 2017-09-16:
Hi @dlech,

I read some weeks ago that GRX is the solution to manage display.

Does exist some documentation to use it?
https://github.com/ev3dev/grx

many thanks in advance.

Juan Antonio

WasabiFan, 2017-09-17:
http://docs.ev3dev.org/projects/grx/en/latest/ ? (linked to in the README)

jabrena, 2017-09-17:
Hi @WasabiFan, many thanks for the link.
I notice that it is necessary to integrate with this concept:
https://wiki.gnome.org/Projects/GObjectIntrospection

I will review with this `Java` library in order to run `GRX` using `GObject Introspection`
https://wiki.gnome.org/Projects/JGIR

Juan Antonio

dlech, 2017-10-03:
Can this be closed?

JakubVanek, 2018-11-18:
This is not necessary anymore, because we have [pure Java support for display](https://github.com/ev3dev-lang-java/ev3dev-lang-java/blob/master/docs/adr/adr-display-api.md) now.

closed 2018-11-19 by jabrena