ev3dev/ev3dev #962: EV3 color screen does not work in stretch

dlech, 2017-09-18:
From #938, known issue. Since we changed to device tree, we need to figure out how to hand the color display again.

dlech, 2018-06-14:
This is fixed in kernel `4.14.49-ev3dev-2.0.2-ev3` and `flash-kernel 3.75ev3dev9 `.

It also requires an ev3dev-stretch image 2018-05-24 or newer with `u-boot.bin`.

The file `uEnv.txt` must be modified to select the `da850-lego-ev3-color.dtb` device tree file.

closed 2018-06-14 by dlech

oknystad, 2018-06-23:
I have the Adafruit "18 screen attached to my EV3, and it works fine with an older jessie kernel.
I have now tested the image `snapshot-ev3dev-stretch-ev3-generic-2018-06-13.img.xz`, and uncommented the line `# fdtfile=da850-lego-ev3-adafruit18.dtb` in `uEnv.txt`. When pressing the ON button, the EV3 doesn't boot, it just shows a white screen and constant red LEDs.
Any suggestions to why it doesn't boot?
I've tested with two SD cards which otherwise work fine, and with charger attached. When re-inserting the `#`, booting works fine (normal LEDs and button behaviour, though only pixel noise on the screen.)

oknystad, 2018-06-23:
Thanks, that worked!
Just for the record, my setup makes the screen upside-down - it would be nice if it could be rotated easily, but it's not critical for my use.

dlech, 2018-06-23:
sorry, I haven't figured out device tree overlays yet, or it would be easy. if you are not afraid of a little C coding, you might be able to hack u-boot to change the value in the device tree based on an environment variable. 

https://github.com/ev3dev/u-boot/blob/ev3dev-stretch/board/lego/ev3/legoev3.c

feel free to open a new issue for this and someone may get around to it eventually