ev3dev/ev3dev #965: How to tune parameters in a Python program

laurensvalk, 2017-09-23:
This is more of a generic Python question, but quite relevant for EV3 robots:

What is a good way to efficiently tune "constant" parameters in a program?

As you know, starting a Python program on the EV3 takes quite some time. I'd like to be able to adapt a few tuning parameters (numeric values) without restarting the Python interpreter.

Right now, I place the whole program in a loop. At the start of the loop, the program reads the tuning parameters from a text file. I edit this text file in a separate session. I use the Touch Sensor to stop and start the program after editing the file.

This feels a bit hacky. Is there a better way?

ddemidov, 2017-09-23:
Show the parameter values on screen, use up/down brick buttons to select a parameter, left/right to change its value?

WasabiFan, 2017-09-23:
What about running it from an SSH session and reading from stdin? In each loop you could ask for changes or something of that nature.

dlech, 2017-09-23:
* Create a python program that runs your other python program, like OpenRoberta: https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/roberta/lab.py#L278

* Use micropython (#937)

laurensvalk, 2017-09-23:
Good suggestions so far, thanks! I'll look into those ideas. Here are some more specifics to clarify a bit more. 

Perhaps not surprisingly, this relates to the [segway](https://github.com/laurensvalk/segway), but it applies to any robot where small parameter changes drastically affect performance, like fast line-followers. I usually change this bit of code:

```
gainGyroAngle                  = 1156
gainGyroRate                   = 146
gainMotorAngle                 = 7
gainMotorAngularSpeed          = 9
gainMotorAngleErrorAccumulated = 3
```

I usually change one constant at a time (I'd prefer not to loop through them all each time). The amount of change varies a lot as well. In the meantime, I hold the robot upright, making the screen and buttons less accessible.

ddemidov, 2017-09-24:
> I usually change one constant at a time (I'd prefer not to loop through them all each time). The amount of change varies a lot as well. In the meantime, I hold the robot upright, making the screen and buttons less accessible.

Use color sensor instead of brick buttons. Red color - increase value, blue color - decrease value. Use Lego bricks for colors :).

laurensvalk, 2017-09-24:
Well, this does the trick in a less hacky, more pythonic way, I think:

```
# Reload parameters class before restarting the balancing loop
importlib.reload(parameters)
```

parameters.py is a file with a few classes in it, containing the tuning parameters that can readily be changed.

closed 2017-09-26 by laurensvalk