ev3dev/ev3dev #966: Possible problem with debian-jessie-armel.dockerfile

jabrena, 2017-09-24:
I was testing some stuff with the `ev3dev docker images` but I think that the armel image for jessie is broken.

_Affected file:_
https://github.com/ev3dev-lang-java/docker-cross/blob/master/ev3dev-jessie/debian-jessie-armel.dockerfile

In my local tests I receive the following error:

```
  ev3dev-jessie git:(master) ✗ docker build -t ev3dev-armel -f debian-jessie-armel.dockerfile .

Sending build context to Docker daemon  20.99kB
Step 1/11 : FROM ev3dev/debian-jessie-armel-qemu-minbase
 ---> 342e80581d45
Step 2/11 : COPY apt.sources.list.debian /etc/apt/sources.list
 ---> Using cache
 ---> c42559e14819
Step 3/11 : RUN dpkg --add-architecture i386 &&     apt-key adv --keyserver pgp.mit.edu --recv-keys D57D95AF93178A7C &&     apt-get update &&     DEBIAN_FRONTEND=noninteractive apt-get install --yes xz-utils         libc6:i386 libstdc++6:i386 libz1:i386 libncurses5:i386 libpython2.7:i386         build-essential cmake valac man-db bash-completion sudo vim nano less tree wget
 ---> Running in 9e2966c6888d
Executing: gpg --ignore-time-conflict --no-options --no-default-keyring --homedir /tmp/tmp.8CyvKXhwQs --no-auto-check-trustdb --trust-model always --primary-keyring /etc/apt/trusted.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-jessie-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-jessie-security-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-jessie-stable.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-stretch-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-stretch-security-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-stretch-stable.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-wheezy-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-wheezy-stable.gpg --keyserver pgp.mit.edu --recv-keys D57D95AF93178A7C
gpg: requesting key 93178A7C from hkp server pgp.mit.edu
gpg: key 93178A7C: public key "ppa.lechnology.com - Archive Signing Key <david@lechnology.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
Ign http://ftp.debian.org jessie InRelease
Get:1 http://security.debian.org jessie/updates InRelease [63.1 kB]
Get:2 http://ftp.debian.org jessie Release.gpg [2373 B]
Get:3 http://archive.ev3dev.org jessie InRelease [8418 B]
Get:4 http://emdebian.org jessie InRelease [30.7 kB]
Get:5 http://ftp.debian.org jessie Release [148 kB]
Get:6 http://security.debian.org jessie/updates/main Sources [259 kB]
Get:7 http://security.debian.org jessie/updates/contrib Sources [1298 B]
Get:8 http://security.debian.org jessie/updates/non-free Sources [20 B]
Get:9 http://security.debian.org jessie/updates/main armel Packages [537 kB]
Get:10 http://archive.ev3dev.org jessie/main Sources [21.2 kB]
Get:11 http://security.debian.org jessie/updates/contrib armel Packages [994 B]
Get:12 http://security.debian.org jessie/updates/non-free armel Packages [20 B]
Get:13 http://security.debian.org jessie/updates/main i386 Packages [546 kB]
Get:14 http://archive.ev3dev.org jessie/main armel Packages [48.5 kB]
Get:15 http://security.debian.org jessie/updates/contrib i386 Packages [2369 B]
Get:16 http://security.debian.org jessie/updates/non-free i386 Packages [20 B]
Get:17 http://archive.ev3dev.org jessie/main i386 Packages [15.4 kB]
Ign http://emdebian.org jessie InRelease
Get:18 http://emdebian.org jessie/main Sources [6458 B]
Get:19 http://emdebian.org jessie/main armel Packages [8215 B]
Get:20 http://emdebian.org jessie/main i386 Packages [14.6 kB]
Get:21 http://ftp.debian.org jessie/main Sources [9166 kB]
Get:22 http://ftp.debian.org jessie/contrib Sources [58.9 kB]
Get:23 http://ftp.debian.org jessie/non-free Sources [119 kB]
Get:24 http://ftp.debian.org jessie/main armel Packages [8868 kB]
Get:25 http://ftp.debian.org jessie/contrib armel Packages [42.9 kB]
Get:26 http://ftp.debian.org jessie/non-free armel Packages [71.1 kB]
Get:27 http://ftp.debian.org jessie/main i386 Packages [9065 kB]
Get:28 http://ftp.debian.org jessie/contrib i386 Packages [58.1 kB]
Get:29 http://ftp.debian.org jessie/non-free i386 Packages [97.1 kB]
Fetched 29.3 MB in 31s (920 kB/s)
Reading package lists...
W: GPG error: http://emdebian.org jessie InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7DE089671804772E
Reading package lists...
Building dependency tree...
The following extra packages will be installed:
  binutils bsdmainutils bzip2 ca-certificates cmake-data cpp cpp-4.9 dpkg-dev
  fakeroot file g++ g++-4.9 gcc gcc-4.9 gcc-4.9-base:i386 groff-base ifupdown
  iproute2 isc-dhcp-client isc-dhcp-common krb5-locales libalgorithm-c3-perl
  libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl
  libarchive-extract-perl libarchive13 libasan1 libatm1 libatomic1
  libbz2-1.0:i386 libc-dev-bin libc6-dev libc6-i686:i386 libcgi-fast-perl
  libcgi-pm-perl libclass-c3-perl libclass-c3-xs-perl libcloog-isl4
  libcpan-meta-perl libcurl3 libdata-optlist-perl libdata-section-perl
  libdb5.3:i386 libdns-export100 libdpkg-perl libelfg0 libexpat1
  libexpat1:i386 libfakeroot libfcgi-perl libffi6 libffi6:i386
[... 79 lines cut ...]
WARNING: The following packages cannot be authenticated!
  pkg-config-bin pkg-config
E: There are problems and -y was used without --force-yes
The command '/bin/sh -c dpkg --add-architecture i386 &&     apt-key adv --keyserver pgp.mit.edu --recv-keys D57D95AF93178A7C &&     apt-get update &&     DEBIAN_FRONTEND=noninteractive apt-get install --yes xz-utils         libc6:i386 libstdc++6:i386 libz1:i386 libncurses5:i386 libpython2.7:i386         build-essential cmake valac man-db bash-completion sudo vim nano less tree wget' returned a non-zero code: 100
➜  ev3dev-jessie git:(master) ✗ docker run -t ev3dev-armel 
Unable to find image 'ev3dev-armel:latest' locally
docker: Error response from daemon: repository ev3dev-armel not found: does not exist or no pull access.
See 'docker run --help'.
➜  ev3dev-jessie git:(master) ✗
```

The same image for `stretch` run nice.

```
docker build -t ev3dev-armel -f debian-stretch-armel.dockerfile .

debconf: delaying package configuration, since apt-utils is not installed

Successfully built 3980a87ca37d
Successfully tagged ev3dev-armel:latest
``` 

Maybe it could be interesting to add in the image the package `apt-utils`

Juan Antonio

dlech, 2017-09-24:
Why do you not just use the pre-built image from docker hub?

    docker pull ev3dev/ev3dev-jessie-cross

or 

    docker pull ev3dev/ev3dev-jessie-armel-cross

jabrena, 2017-09-24:
Hi @dlech, 

because I tried to emulated the same steps from here:
https://www.eclipse.org/openj9/oj9_build.html

in a preliminar phase and later create a new `.dockerfile` for the required dependencies for `Open J9`
https://es.slideshare.net/CharlieGracie/fosdem-2017-open-j9-the-next-free-java-vm

but building the image, I get that error. Later using stretch version, I could continue with the process. This is the reason.

I suppose that you say to create a `.dockerfile` using as base `ev3dev/ev3dev-jessie-armel-cross`, isn´t it?

``` dockerfile
FROM ev3dev/ev3dev-jessie-armel-cross

#REST OF THE FILE
```

Juan Antonio

dlech, 2017-09-24:
That sounds like a good idea.

jabrena, 2017-09-24:
Next week, I will try it.

Good night

dlech, 2017-10-24:
Can this be closed?

jabrena, 2017-10-25:
Yes

closed 2017-10-25 by jabrena