ev3dev/ev3dev #967: lego rebot with EV3DEV don't show its ip address on the Screen.

superPershing, 2017-09-25:
I write EV3DEV to my Lego mindstroms EV3. Lego robot show its ip address after the first boot. Then I connect to the robot's IP address by SSH. After the first boot, neither I reboot or start my lego from shutdown,  the screen do not show its ip address. When I enter the `Wireless and Networks`, it always keep `Loading...`. I still can connect the lego robot because the lego has its ip address the same as before. The light of wireless network card still keep flashing. It will be terrible when I move my lego robot to another wifi environment that I cannot know the ip address so I may cannot connect it my SSH.

jabrena, 2017-09-25:
Hi @superPershing, did you try to disconnect the wifi dongle and restart the brick?

superPershing, 2017-09-25:
@jabrena I can't enter the `Wireless and Networks`, it keep show `Loading...`. It seems no use that I try unpluging the wireless network card and pluging it again to the robot.

jabrena, 2017-09-25:
Try to repeat this action:

- Shutdown the brick from the menu.
- Unplug the wifi dongle
- Init the brick
- After starting Brickman, plug the wifi dongle

superPershing, 2017-09-25:
@jabrena No use. It works badly as before.

jabrena, 2017-09-25:
mmmm,

in that case, if you are using EV3, maybe you should format again your sd again with ev3dev. Maybe you have some configuration corrupt from a previous session.

With a Raspi board with BrickPi/PiStorms, you could connect using a wired connection and check what happen. But for EV3Dev, I suspect that some config file is  corrupt now.

The problem is that you can get access to network features so you are not able to use another alternative to connect.

Can you burn again ev3dev and test with your wifi dongle?
What dongle do you use?

Juan antonio

superPershing, 2017-09-25:
@jabrena I try writing the image to the SD card and boot from SD card about 2 to 3 times. It always perform like before. Last time I try forgetting the WIFI in the `Wireless and Networks`, then I reboot my rebot, enter `Wireless and Networks` then connect my WIFI again. This time I see ev3's ip address again. I do not know why? Need I have to dis connect WIFI before I shutdown the robot everytime?

superPershing, 2017-09-25:
@jabrena I use the `EW-7811Un` dongle.

jabrena, 2017-09-25:
The dongle that you are using should not give you any problem:
http://us.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/us/wireless_adapters_n150/ew-7811un/

I have to check but one of my wifi dongles is a similar version of your Dongle. Another wifi dongle running without any problem is this one:
https://www.netgear.com/home/products/networking/wifi-adapters/WNA1100.aspx

In my case, once the wifi dongle is configured I don't have any network issue. Do you connect the dongle in an USB hub and you change the position of the dongle? or you only plug the dongle in the unique usb port in the EV3?

One behaviour that I detected is when the brick is blocked for some weird reason and you have to remove the rechargeable batteries, in a weird case, I lost the configuration.

Juan Antonio

dlech, 2017-09-25:
In SSH terminal on EV3...

what is output of `ev3dev-sysinfo -m`?

what is output of `sudo systemctl status brickman`?

what is output of `sudo systemctl status connman`?

If re-flashing the SD card fixed the problem, then you probably have a defective SD card. Do you have another SD card you can use?

superPershing, 2017-09-29:
I change my SD card. All problem has disappeared. Thank you very much!

closed 2017-09-29 by superPershing